esp8266/Arduino #2924: Esp8266 W5100 ethernet not working

littleoz2, 2017-02-03:
Ethernet.begin returns 0 after a few minutes, doesn't get an IP address

littleoz2, 2017-02-03:
I'm using a NodeMCU board with an ESP-12E on it

capolan, 2017-02-14:
Hi, I'm also have this problem, with nodemcu v3 wemos with ethershield w5100. There is no way get IP, nether fixed IP. It stucks on Ethernet.Begin(mac) like littleoz2;

The link below shows differecences between NodeMCU versions.
http://frightanic.com/iot/comparison-of-esp8266-nodemcu-development-boards/

The connetions is as follows
  SPI on nodemcu                ethershield ISCP
D5 SCK GPIO14      -->           pin 3
D6 MISO GPIO 12  -->           pin 1
D7 MOSI GPIO13 -->             pin 4
D8 SS GPGIO 15    -->            digital pin 10 on shield 
VV                       -->              pin 2 VCC 5v ISCP
GND                    -->              pin 8 GND     ISCP

VV = power from USB, there is no external power for shield (it draws only 150mA).

The shield works fine on arduino mega2560, nut not on nodeMCU. On nodeMCU the RX/TX stay off, and the other keep ok. The ethernet connector ligths on.

What am i doing wrong ?

MarcT2app, 2017-02-24:
I'm using a NodeMCU v1 and TOP Mini W5100 LAN Ethernet Shield Network Module board for Arduino Best...
And received IP from DHCP server. (I connect nss to D8, CK to D5, MI to D6, MO to D7).

My problem is with UDPSendReceivedString, 
If test with example in 8888 port work fine (Send data and received acknowledged text)
But if only change port 8888 to other not work.

StaffMark, 2017-02-24:
Hi,

I test with Nodemmcu v.1 and TOP Mini W5100 LAN Ethernet Shield.

DHCP work fine: "My IP address: 192.168.100.39" (tested witch DhcpAddressPrinter example)

I have a problem with UDPSendReceivedString... If test with example port 8888 work fine, but if change port to 3009 not work.

Is possible EthernetUdp have an error?

JamesGKent, 2017-02-24:
not using ethernet, but using the ESP wifi i found i could not get UDP to work with a port number lower of 4096 or below, only 4097 and higher worked, maybe try that?

StaffMark, 2017-02-24:
Hi James, yes i tested with UDP port 8887 and not work with ethernet class...
I use ESPwifi and UDP ports 3008,3009,3010,5011 UDP and work fine.

devyte, 2017-10-27:
Closing due to age and lack of info. If there is still a problem, please open a new issue and fill in the requested fields.

closed 2017-10-27 by devyte