esp8266/Arduino #2925: New Simple Easily Reproducable Catastrophic TCP Stack Failure Results in Low Heap Crash shown with WiFiServer,  Latency > 100ms, and Larger Send Bytes

CircuitSerialKiller, 2017-02-04:
### Basic Infos

#### Hardware
Hardware:			ESP-12E, WeMos D1 Mini
Core Version:      	Latest github version

### Description

I have to use the newer core on github for the 
MEMP_NUM_TCP_PCB_TIME_WAIT = 5
to prevent crashing with frequent HTTP connections. (Issue 2767). 

After switching to the latest github version of the ESP8266 core from version 2.3.0, I found that when accessing my HTTP server from VPN, or via the outside work with higher latency connections, HTTP responses from the ESP8266 will not make it to the client, and the ESP8266 will eventually run out of free heap until it fails. I have found this occurs only when the response is beyond a certain number of bytes in length as well, though I haven't determined the exact number of bytes.  > 2048 bytes definitely demonstrates this issue. I tested tested via telnet direct to port 80 as well, and issuing HTTP commands. This does not occur with the 2.3.0 release. 

I've modified the WiFiWebServer.ino file to easily demonstrate this issue. Just setup a NAT to your ESP8266 IP address with the appropriate ports, and from a remote site, cell phone, etc, try accessing the following URL repeatedly while watching the serial console. You'll see the free heap steadily decrease and never recover!
**http://a.b.c.d/gpio/1**
Where a.b.c.d is your outside IP address.

### Settings in IDE

Module:  			NodeMCU 0.9
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp

/*
 *  This sketch demonstrates how to set up a simple HTTP-like server.
 *  The server will set a GPIO pin depending on the request
 *    http://server_ip/gpio/0 will set the GPIO2 low,
 *    http://server_ip/gpio/1 will set the GPIO2 high
 *  server_ip is the IP address of the ESP8266 module, will be 
 *  printed to Serial when the module is connected.
 */

//WiFiWebServer.ino
#include <ESP8266WiFi.h>

const char* ssid = "GoogCorp";
const char* password = "345SpearSt";

//********************Added for Crash Test********************
unsigned long connectionTimeout = 4000;
extern "C" {
#include "user_interface.h"
  //needed for free heap check
}
unsigned long LastHeapReport;
//************************************************************

// Create an instance of the server
// specify the port to listen on as an argument
WiFiServer server(80);

void setup() {
  Serial.begin(115200);
  delay(10);

  // prepare GPIO2
  pinMode(2, OUTPUT);
  digitalWrite(2, 0);
  
  // Connect to WiFi network
  Serial.println();
[... 566 lines cut ...]
Free heap:25824
Free heap:25824
Free heap:25824
Free heap:25824
Free heap:25824
Free heap:25824
Free heap:25824
Free heap:25824
Free heap:25824
Free heap:25824
Free heap:25824
Free heap:25824
Free heap:25824
Free heap:25824
Free heap:25824
Free heap:25824
Free heap:25824
Free heap:25824
Free heap:25824
Free heap:25824
Free heap:25824
Free heap:25824
Free heap:25824
{d
```

me-no-dev, 2017-02-07:
I see only [this](https://github.com/esp8266/Arduino/commit/61787b23afee9bb68e32df39f948529584022387) commit to have relevant changes. try rolling them back and see if that changes your results

igrr, 2017-02-08:
I have tried reproducing this last night, but seemingly my latency was not big enough. @CircuitSerialKiller any chance you can help narrowing down on the offending change using `git bisect`?

CircuitSerialKiller, 2017-02-08:
Yes I can. I'm also going to grab a Wireshark capture. I should have a few hours this weekend to dedicate to this. 
Thanks @igrr

CircuitSerialKiller, 2017-03-11:
Hi @igrr and @joelucid 
The commit that introduced this issue is:
[joelucid committed with igrr ClientConnection uses too much heap when streaming files #2871 (#2874)](https://github.com/esp8266/Arduino/tree/61787b23afee9bb68e32df39f948529584022387)

The commit before this,
[ ivankravets committed with igrr Update @PlatformIO links](https://github.com/esp8266/Arduino/tree/90729ea50c5f98c7f63fc6d92acd8bf4b667415f)
does not have this issue.

Sorry for the long delay before I could dedicate a few hours on a weekend to commit to tracking this down.

d-a-v, 2017-05-22:
Hello,

I did try this with [lwip2](https://github.com/esp8266/Arduino/pull/3206): it works with no bug.

With lwip1.4 it does not. I suspect that the bissected commit makes things work better, so lwip1.4 has too much data to deal with and it just does not work with too much stuffed buffers. This is the reason why I wanted to update lwip in the first place.

Here is the log of the sketch above, with a display every seconds, and available heap size.
Watch time:
```
WiFi connected
Server started
Free heap:42456  was:42456  diff:0  clients=0  time=1
[...]
Free heap:42456  was:42456  diff:0  clients=0  time=7
new client
GET /gpio/3 HTTP/1.1
invalid request
Free heap:42264  was:42264  diff:0  clients=1  time=8
pm open,type:2 0
Free heap:41504  was:41504  diff:0  clients=1  time=9
Free heap:41504  was:41504  diff:0  clients=1  time=10
Free heap:41504  was:41504  diff:0  clients=1  time=11
Free heap:41504  was:41504  diff:0  clients=1  time=12
Free heap:41504  was:41504  diff:0  clients=1  time=13   
new client
GET /gpio/1 HTTP/1.1
Client disonnected
Free heap:41336  was:41504  diff:-168  clients=2  time=14
[...]
Free heap:41336  was:41504  diff:-168  clients=2  time=24
new client
GET /gpio/1 HTTP/1.1
Client disonnected
[...]
```
lots of requests from now
```
Free heap:4376  was:41504  diff:-37128  clients=214  time=35
[...]
Client disonnected  
Free heap:2864  was:41504  diff:-38640  clients=232  time=38
Free heap:2864  was:41504  diff:-38640  clients=232  time=39
Free heap:2864  was:41504  diff:-38640  clients=232  time=40
Free heap:2864  was:41504  diff:-38640  clients=232  time=41
[...]
```
lwip starts to timeout something
```
Free heap:2864  was:41504  diff:-38640  clients=232  time=121
Free heap:3032  was:41504  diff:-38472  clients=232  time=122
[...]
Free heap:3200  was:41504  diff:-38304  clients=232  time=128
[...]
Free heap:19328  was:41504  diff:-22176  clients=232  time=143
[...]
Free heap:41336  was:41504  diff:-168  clients=232  time=150
Free heap:41504  was:41504  diff:0  clients=232  time=151
```
so about ~2min after burst requests, memory is released back by lwip timers.

I am not pushing to lwip2 in v2.4.0, I still think this is too early.
But it is worth saying that this very bug will not be easily solved using lwip1.4.

devyte, 2017-09-09:
I witnessed this same issue with the webserver, and was forced to move to the asyncwebserver. I wish I had seen this before, I would have tested lwip2 instead.

devyte, 2017-09-09:
@d-a-v I'm going a bit crazy with the cleanup, but I think I saw a PR with lwip2. Or was it a branch? Anyways, is that up to date? I may just try testing it anyways.

igrr, 2017-09-12:
@devyte Yes, there is a PR for lwip2: https://github.com/esp8266/Arduino/pull/3206.
As it has been agreed with @d-a-v, it will be merged once 2.4.0 is released.

d-a-v, 2017-09-29:
@devyte: #3362 is updated so you can test. It was not at the time of my previous [message](https://github.com/esp8266/Arduino/issues/2925#issuecomment-328880599).

devyte, 2017-10-03:
@d-a-v thanks, I will get to this right after I finish my current pendings for this repo.

CircuitSerialKiller, 2017-10-31:
Hi, this problem wasn't introduced with an LWIP change. This problem still exists in the current Arduino Core with LWIP v2.1.0-14-g33f234f.
The problem was introduced in the January 31st commit mentioned above with the chunking changes to ClientContext.h. If I set _write_chunk_size to 2920, the problem is gone. 
Without a deep dive into the ESP8266 core and following tcp_write, I can't tell you exactly why. However, I believe chunking should be left out of the picture by default at the moment. @joelucid in #2871 mentions throughput as a justification for this change being acceptable, but latency and waiting for acknowledgements has a major effect on throughput when sending small chunks. I suspect reducing TCP writes to 256 byte blocks are creating a problem elsewhere in the code with latency since cumulative acknowledgments to all those small blocks will take quite some time.

d-a-v, 2017-10-31:
@CircuitSerialKiller Yes it still does not work as stated [above](https://github.com/esp8266/Arduino/issues/2925#issuecomment-303100724). It says:
- the commit on january was efficient (it is not an lwip change, but a WiFi* buffer management improvement)
- it was too efficient for lwip1.4
- current implementation of lwip1.4 does not suffer beeing shaked too much
- updating to lwip2 seems to solve the problem

However once - let's be imaginative enough - lwip2 is merged, I guess the chunk size will be reconsidered because it will have no impact on lwip stability. So it could become configurable for a bandwidth purpose (and TCP_MSS too). There will be a tradeoff between bandwidth and free HEAP.

devyte, 2018-01-03:
@CircuitSerialKiller @d-a-v how does release 2.4.0 work out? I would think no trouble with lwip MSS=536.

devyte, 2018-10-02:
What is the status here?

earlephilhower, 2018-11-29:
Closing due to age and lack of response.

closed 2018-11-29 by earlephilhower