esp8266/Arduino #2926: convert IPAddress to bytes

el3ctrician, 2017-02-04:
hello, i am trying to make an ESP-network for many esps connected to same local network. I am using UDP multicast for discovery but i got one issue : when a device got a discovery packet it should answer with it's IP address in the response packet payload. I looked to the class IPAddress and seems that the only way to do is to use toString function. i want to keep things efficeint so i need only for bytes to send the IP. is there is any way to get ther ip as 4 bytes address ? 
would it be helpful to implement such a thing ?
### Basic Infos

#### Hardware
Hardware:ESP-12
Core Version:2.3.0

### Description

Feature Request 

### Settings in IDE

Module:Generic ESP8266 Module
Flash Size: 1MB
CPU Frequency:80Mhz
Flash Mode:dio
Flash Frequency:40Mhz
Upload Using:SERIAL
Reset Method:ck

CircuitSerialKiller, 2017-02-04:
Are you just talking about the IPAddress object? Why not just access the IPAddress object as any plain old byte array?

el3ctrician, 2017-02-04:
@CircuitSerialKiller can you explain more please.. i see that the byte array is private so i can't acess it ..

el3ctrician, 2017-02-04:
so i continued to dig more into the code and i found two possibile solution

1. Using a buffer of uint32_t 
`IPAddress myIP(127,0,0,1);
uint32_t ipBuffer = myIP;`

2. Using a buffer if uint8_t 
 `IPAddress myIP(127,0,0,1);
uint8_t ipBuffer[4] = {myIP[0],myIP[1],myIP[2],myIP[3]};`

closed 2017-02-04 by el3ctrician