esp8266/Arduino #2928: udp recived packets number

el3ctrician, 2017-02-04:
hello,
i would like to know if there is a way to get the number of packets in udp buffer and also to set the packet len according to the payload ?

thanks in advance

wuwx, 2017-02-06:
did you mean this:

int packetSize = Udp.parsePacket();

el3ctrician, 2017-02-07:
nope, according to the header file [WiFiUdp.h](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiUdp.h) this wuold return the size of the packet in bytes. I think it's refearing to the first packet. So my question is how can i get the total number (and not size) of packets recived in the Udp buffer ?

el3ctrician, 2017-02-10:
@igrr if it's not implemented i can try to do if you can kindly guide me from where should i start ?

igrr, 2017-02-10:
Take a look at UdpContext.h. There is a pbuf chain there, you can add a method to count the number of pbufs in the chain.

devyte, 2017-09-09:
@el3ctrician did you get around to implementing this? do you want to make a PR?

el3ctrician, 2018-03-29:
@devyte  i didn't have enough time to do it since the project that i was working on got cancelled, but i would like to try to implement it so maby i'll fork the code for a try this weekend ...

devyte, 2018-03-29:
Ok, post results or ask questions here.

el3ctrician, 2018-03-29:
well actually i just saw that the pbuf in the udpContext is a  udp_pcb struct type, so it is a linked list right ? i need to count by checking the  udp_pcb.next right?

d-a-v, 2018-03-30:
The pbuf api is [lwip's](http://www.nongnu.org/lwip/2_0_x/group__pbuf.html).

WiFiUdp through UdpContext makes use of it: ```_recv()``` is called when a packet is received, filling ```_rx_buf ``` which is a pbuf, using ```pbuf_cat()```. It makes the pbuf grow and it *does not* chain (in the linked list way) the new incoming data. So ```pbuf->len == pbuf->tot_len``` and next is always null.

The notion of packet is then wiped out showing udp's incoming data to the user as a stream, which is what it really is (```class UDP: public Stream```) but not compatible with the "datagram" way of thinking with udp.
Is that your initial concern ?

If this had to be changed, ```pbuf_chain()``` would have to be used instead of ```pbuf_cat()``` along with the linked list management. That would put malloc under less stress and restore the notion of packet if this had to be needed.

devyte, 2020-05-14:
I am closing this due to age and lack of feedback.
If there is still an idea to be contributed regarding this topic, please open a new issue, follow the template instructions, and explain the idea in detail.

closed 2020-05-14 by devyte