esp8266/Arduino #2930: nodeMCU ESP doesn't wakeup after deepSleep

ductruong253, 2017-02-05:
### Basic Infos

#### Hardware
Hardware:			nodeMCU ESP-12E V1.0
Core Version:      	<I don't really know what this is>

### Description

DeepSleep Problem

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E)
Flash Size: 		4MB (3M SPIFFS)
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, LOW);   // Turn the LED on (Note that LOW is the voltage level
                                    // but actually the LED is on; this is because 
                                    // it is acive low on the ESP-01)
  delay(5000);                      // Wait for a second
  digitalWrite(LED_BUILTIN, HIGH);  // Turn the LED off by making the voltage HIGH
  ESP.deepSleep(5000000,WAKE_RF_DEFAULT); //go to sleep for 5 second
  delay(500);
}

messages here
```
The broblem is that the ESP didn't wake up properly. (The GPIO16 is connected to RST after code upload). The LEDs (both blue and red) keep flashing and printout (baudrate = 74880):

ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld

I don't know what happen. Please help me

igrr, 2017-02-05:
Your built-in led is connected to gpio16, so you are resetting their chip when you are configuring gpio16 as output. Try using another pin for an led.

closed 2017-02-05 by igrr

ductruong253, 2017-02-05:
Hi, igrr

I did what you said, and I think it's working, but I've changed my code to see if it was running good.

`void setup() {
  pinMode(D4, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
}

// the loop function runs over and over again forever
void loop() {
  Serial.println("AWAKE");
  digitalWrite(D4, LOW);   // Turn the LED on (Note that LOW is the voltage level
                                    // but actually the LED is on; this is because 
                                    // it is acive low on the ESP-01)
  delay(5000);                      // Wait for a second
  digitalWrite(D4, HIGH);
  Serial.println("Sleep");// Turn the LED off by making the voltage HIGH
  ESP.deepSleep(5000000,WAKE_RF_DEFAULT); //go to sleep for 5 second
  delay(500);
}`

And the Serial Monitor printed (78800):
`
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
`

There is no "AWAKE" or "Sleep" printed out.

hn-nguyen, 2017-12-13:
I was in the same situation. I went out by linking GPIO_0 to RST pin.

padcom, 2018-09-02:
@hn-nguyen You have no idea how long I was looking for a solution to this problem! Linking GPIO_0 to RST and pulling it up with a resistor solved the problem! Thanks!!!

padcom, 2018-09-02:
It is worth to note that although this works great with the breakout board it is not a solution for the Wemos D1 mini. If anyone knows a solution for that please share.

padcom, 2018-09-03:
Correction: one in 3 units (D1 mini R1) works as expected. I think it is time to upgrade...

liorhal1, 2019-05-12:
> I was in the same situation. I went out by linking GPIO_0 to RST pin.

with the D0 too , or  only   the D3 to RST ?

5chufti, 2019-05-12:
There are problems with certain esp-12 modules, especially the E and F version, S seems to be ok so far

liorhal1, 2019-05-12:
how can i know whats modul i got , i orderd 20 pcs

5chufti, 2019-05-12:
usually it is stamped on the lid

liorhal1, 2019-05-13:
i didnt find

I looked at the difference between the products that wake up after a deep sleep and those who do not wake up and found that those who wake up have an AI-TINKER stamp and the others do not, there is only a symbol of half circles

5chufti, 2019-05-13:
sorry, I have just quickly looked at one module I had nearby.
[image]

So nowadays they seem to only print it on the back ...
[image]

liorhal1, 2019-05-13:
i notice on the serial that it write  "ets Jan  8 2013,rst cause:2, boot mode:(3,6)" 

and sometimes after that write also "ets_main.c "

and then stuck

liorhal1, 2019-05-13:
> There are problems with certain esp-12 modules, especially the E and F version, S seems to be ok so far

isnt away to fix it, if it will not work i need to buy 20 nodemcu again