esp8266/Arduino #2932: ESP8266 - COAP with DTLS support required

jyotirajsharma, 2017-02-05:
I wrote following code using ESP8266 ESP-01. I have used CoAP protocol which runs over UDP.
I now want to use CoAP with DTLS. Is there any support in ESP, please let me know ?

#include <SPI.h>
#include <ESP8266WiFi.h>          //http server library
#include <WiFiUDP.h>              //udp library
#include <coap.h>
const char* ssid          = "AndroidAP";
const char* password      = "abcde12345";
byte mac[] = { 0x00, 0xAA, 0xBB, 0xCC, 0xDE, 0x02 };
IPAddress ip(192, 168, 43, 7);
// CoAP client response callback
void callback_response(CoapPacket &packet, IPAddress ip, int port);

// UDP and CoAP class
WiFiUDP Udp;
Coap coap(Udp);

// CoAP client response callback
void callback_response(CoapPacket &packet, IPAddress ip, int port) {
  Serial.println("[Coap Response got]");

  char p[packet.payloadlen + 1];
  memcpy(p, packet.payload, packet.payloadlen);
  p[packet.payloadlen] = NULL;

  Serial.println(p);
}

void setup() {
  Serial.begin(115200);

    if (WiFi.status() != WL_CONNECTED) 
    {
        delay(1);
        // Connect to WiFi network
        Serial.println();
        delay(10);
        Serial.println();
        delay(10);
        Serial.print("Connecting to ");
        delay(10);
        Serial.println(ssid);
        delay(10); 
        WiFi.begin(ssid, password);
        
        while (WiFi.status() != WL_CONNECTED) {
          Serial.print(".");
          delay(500);
        }
        Serial.println("");
        Serial.println("WiFi connected");
        
        // Print the IP addres 
        Serial.print("ESP8266 IP: ");
        Serial.println(WiFi.localIP());
      
        //Serial.print("ESP8266 WebServer Port: ");
        //Serial.println(SVRPORT);
        delay(300);
    }

  // client response callback.
  // this endpoint is single callback.
  Serial.println("Setup Response Callback");
  coap.response(callback_response);

  // start coap server/client
  coap.start();
}

void loop() {
  // send GET or PUT coap request to CoAP server.
  // To test, use libcoap, microcoap server...etc
   int msgid = coap.put(IPAddress(192,168,43,7), 5683, "sensordata","temp 23 humid 35");
  Serial.println("Send Request");
 // int msgid1 = coap.get(IPAddress(192, 168, 43, 7), 5683, "time");

  delay(1000);
  coap.loop();
}

jyotirajsharma, 2017-02-08:
Is there any support of COAP with DTLS in ESP8266 WiFiUDP ? If WiFIUDP does not support DTLS, if anyone knows the support of COAP with DTLS in EtherentUDP ??

praveshbansal, 2017-03-03:
HI Jyoti , 

Are you able to resolve this issue for the Esp module . 
I am struggling with the same problem .
Any help will be appreciated .

Pravesh Bansal

jyotirajsharma, 2017-04-03:
Sorry for late response. I am still not able to find the solution for this.

devyte, 2017-09-09:
@jyotirajsharma is this issue still valid with latest?

jyotirajsharma, 2017-09-09:
I have not used with latest. Is DTLS supported in latest esp build?

beleyj, 2017-09-27:
Has anyone been able to implement DTLS with CoAP?

malachib, 2017-10-16:
I have not seen one yet.  However I have gotten it to work in a very controlled environment - nothing publishable.  Lots of misinformation that ESP8266 doesn't have enough RAM to do so - but that is misinformation, don't believe it.

beleyj, 2017-10-20:
Thank you so much sharing! I'll keep you updated if I make any improvements.

malachib, 2018-05-31:
I'm still plugging away.  It's a hard nut to crack.  Getting close to an alpha, but pretty far from an Arduino-like interface: https://github.com/moducom/mc-coap 

Working on the DTLS components now, starting with ESP32 planning to backport to ESP8266

malachib, 2018-07-03:
For the record, DTLS has been an (expletive deleted).  I get every indication that it is the go-to mechanism for securing IoT devices, so it's pretty frustrating the apparent lack of interest in it.  I'm fighting the same bugs and misbehaviors from a year ago.  mbedtls, axTLS, WolfTLS all have let me down.  Hopefully a reader has had better luck and can share their secret

devyte, 2018-07-03:
We have experimental bearssl support. No idea if it will work, but it's been pretty stable so far. Give that a try.
Also, no idea how to get your coap lib to use it, you'll likely have to tweak something in there. There are examples on bearssl usage available to study.

earlephilhower, 2018-07-03:
DTLS is a different protocol from TLS since it's UDP based, not connection oriented.

Neither BearSSL nor the other SSL libs mentioned support it, AFAIK.  [Wikipedia](https://en.wikipedia.org/wiki/Comparison_of_TLS_implementations)

You should be able to re-use most TLS lower-level functions, the protocol layer and packet formats are going to be custom for this.  BearSSL's state machines are written in a custom Forth dialect which would either make this way harder or way easier, depending on how friendly you are with RPN.

ARM's mbedTLS supports DTLS per that Wikipedia entry, but requires too much memory (RX and TX buffers are an identical, fixed size at compile time, meaning 32KB required just for them, ignoring all the other memory needs of TLS!) for the ESP8266 Arduino core from what I can see.

malachib, 2018-07-03:
@earlephilhower this seems spot on.

I am under the impression `CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN` option will, with some caveats, allow you to adjust down that size.

@devyte mating the CoAP lib to DTLS won't be too difficult really.  Had a long time to prepare :)

devyte, 2018-07-04:
I was speaking at abstract level, since I have no idea of the lower level workings. I'm glad I wasn't far off the mark :)
If you get DTLS to work it'd be a big win. I've seen discussions and requests in several places.

earlephilhower, 2018-07-06:
Wow, @malachib, that PR in mbedtls escalated quickly!  Ouch.

As the first post said, though, the `#define` change to limit the size breaks the internet (but is legal if you control both sides of the pipe like the ARM team member said or have server supporting MFLN which is still very rare).

For BearSSL, @pornin lets you dynamically choose receive and send buffer size independently.  We use that here to use a 16K receive buffer (so even w/o MFLN it works everywhere) and a 512b send buffer (also 100% legal, we just send more fragments) by default.

On reexamining the wiki article, I see that wolfSSL does support DTLS.  I've seen the author and his booth at several security shows, but my (very possibly incorrect) impression was it's still too big for ESP8266 and more focused on embedded x86-type setups.

malachib, 2018-07-06:
@earlephilhower I would definitely be curious to see how BearSSL performs if one were to fill in the missing DTLS implementation pieces.  Not sure I am up to that task personally,  though my knowledge is getting there slowly.

As it happens, there is limited wolfSSL supported on ESP8266 (https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/components/ssl) but, wait for it.... it's a precompiled blob without DTLS activated 🙄 

I tried to manually compile it for the platform, but no luck.  However, did not attempt to do so with https://github.com/espressif/esp8266-wolfssl yet, which almost assuredly would be easier

pornin, 2018-07-06:
DTLS support in BearSSL is on my todo-list. I don't have a scheduled date (it depends on how much time I get) but it will definitely be done at one point. DTLS reuses most of the mechanics of TLS (e.g. encryption of records or encoding of handshake messages); however, it must tolerate loss of records, and, during the handshake, there is a complicated process of fragmentation and retransmit. The following areas will be challenging:

 - BearSSL strives to be a computational library with no hardware/system dependency. Thus, any timer management (waiting/retransmit) has to be done externally, and designing the proper API for that is not easy (generally speaking, the hardest part of developing a library like BearSSL is getting the API right).

 - Theoretically, a DTLS implementation should buffer handshake message fragments (in fact RFC 6347, section 4.2.3, makes it a "MUST"). This requires having a memory buffer large enough to receive the complete message; this will use up some RAM... BearSSL normally streams all processing, and can thus handle (for instance) a 100 kB "Certificate" message without allocating anything. It is unclear whether such streaming can be applied to DTLS, especially when the RFC explicitly prohibits it. I have some ideas about how to do that heuristically, but more thinking is required.

malachib, 2018-07-06:
@pornin I'm noticing that though there are many MUST statements in that section, only this one

> DTLS implementations MUST be able to handle overlapping fragment ranges.
   This allows senders to retransmit handshake messages with smaller
   fragment sizes if the PMTU estimate changes.

Seems to expressly mandate a complete intact buffer.  

If I read this right, it is to protect against the contingency where the MTU changes from 1500 (or other MTU size, if that's truly your environment).  If that's the case, I'd be willing to take that risk and deviate from spec :)  I don't expect MTU to be changing ever in my own use cases

Also as a side note, though it's definitely long in the tooth, LwIP's PBUF approach is pretty cool for this kinda stuff

devyte, 2019-11-06:
@pornin @malachib has anything been done about DTLS in the last year?

malachib, 2019-11-07:
I gave up on it, but I did see a fair bit of activity on DTLS in general trickling into the `esp-idf` repo.  Whether it addressed this specific issue I couldn't say... hopefully?

Friedmonkey, 2025-07-24:
is there any way to send a simple coaps request with dstl psk and identity yet?
i want to request to my ikea tradfri gateway which uses coap with dtls 
dtsl coap clients are easy to find for windows or linux
but i want to run it on my esp32 with ardunio ide but i cannot for the life of me get it to work

its been years there should be something that works by now does anyone know?