esp8266/Arduino #2934: ESP8266WebServer - serveStatic, json.gz bug

renat2985, 2017-02-05:
I use caching so
```
***
HTTP.serveStatic("/css/", SPIFFS, "/css/", "max-age=31536000"); // кеширование на 1 год
HTTP.serveStatic("/js/", SPIFFS, "/js/", "max-age=31536000"); // кеширование на 1 год
HTTP.serveStatic("/lang/", SPIFFS, "/lang/", "max-age=31536000"); // кеширование на 1 год
***
```
files in folders css, js work well
/css/style.css.gz
/js/display.js.gz

But json files if they are compressed (/lang/ru.json.gz) in the lang folder, I see an error
Why?

[image]

me-no-dev, 2017-02-05:
what is the content type that the browser sees? it's on the Заголовки tab

renat2985, 2017-02-05:
> /lang/ru.json.gz
```
Access-Control-Allow-Orig...	
*
Cache-Control	max-age=31536000
Content-Length	2335
Content-Type	application/octet-stream

показать исходный код
Accept	text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
Accept-Encoding	gzip, deflate
Accept-Language	en-US,en;q=0.5
Host	192.168.1.166
Referer	http://192.168.1.166/
User-Agent	Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:45.0) Gecko/20100101 Firefox/45.0
```

me-no-dev, 2017-02-06:
Content-Type should be text/json and Content-Encoding should be gzip. Interesting why it's not

me-no-dev, 2017-02-06:
oh maybe it's not the whole file? if part of it is missing it might not be able to decode it. is the size correct? can you download the file directly?

me-no-dev, 2017-02-06:
aaahhhh json is missing from the content types
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/src/detail/RequestHandlersImpl.h#L118

renat2985, 2017-02-06:
You can do fix this?

closed 2018-01-02 by igrr