esp8266/Arduino #2935: Communication between two ESP8266 using Arduino IDE

Manje-Gowda-K-R, 2017-02-06:
Hello Everyone..... Firstly, I'll just give you a brief introduction about our project... its "Speech controlled Home Automation System" where we are controlling the Home appliances by the voice or speech commands. We're successfully done with the voice recognition module. But our problem is with achieving the communication between the transmitter {where the user gives the voice command} and receiver {where the home appliances are present} through WiFi i.e. we have got 2 ESP8266 12E modules interfaced with Arduino both at the transmitter and receiver side and these two WiFi modules need to be communicated(To interchange some data sent by Arduinos).So please let us know which Arduino code need to be used at transmitter side wifi module and at Receiver side wifi module and also about the header files or additional softwares(if any)to be used.

el3ctrician, 2017-02-06:
Hello, 
First of all i think such a post should go to the comunity forum. and not github. Also it should be very common things that esps are used to and there are a ton on projects out there that you can see. Anywayi would say that you can use UDP to achive that. You should define some application level protocol to associate certian packets to certain events. note also that UDP by default dosen't use ACK back so if you need to be sure that the message is recived you should add an ACK system to the application level. You can check this example [https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/doc/esp8266wifi/udp-examples.md](UDP-Example).
I would consider also using a multicast to avoid searching for esps 
[https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/doc/esp8266wifi/udp-class.md](MultiCast)

devyte, 2017-09-09:
@Manje-Gowda-K-R this is not the right place to ask such questions. This is an issue tracker, meant for tracking issues in the core. Assuming you still need assistance, please refer to esp8266.com or stackoverflow.
Closing.

closed 2017-09-09 by devyte