esp8266/Arduino #2937: EspSoftwareSerial and ESP8266HTTPClient broken

wuwx, 2017-02-06:
Please fill the info fields, it helps to get you faster support ;)

### Basic Infos

#### Hardware
Hardware:		NodeMCU
Core Version:      	e9dea9af999c324c55fe2f4ecfb16fa2f9e867c5 (also tested at 2.3.0)

### Description

when i use EspSoftwareSerial and ESP8266HTTPClient Frequent call，The WiFi Connection be broken

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		4MB
CPU Frequency:		80Mhz
Upload Using:	Serial

### NodeMCU Sketch

```cpp
#include <Arduino.h>
#include <SoftwareSerial.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

SoftwareSerial Serial2(12, 14, false, 2048);

void setup()
{
  Serial2.setTimeout(50);
  Serial2.begin(9600);
  Serial.begin(115200);
  WiFi.mode(WIFI_AP_STA);
  WiFi.begin("SSID", "password");
}

void loop()
{
  if (Serial2.available()) {
    Serial.printf("WiFi.getMode(): %d\r\n", WiFi.getMode());
    Serial.printf("IP address: %s\r\n", WiFi.localIP().toString().c_str());
    Serial.printf("[B] ESP FreeHeap: %d\r\n", ESP.getFreeHeap());
    String payload = Serial2.readString();
    HTTPClient client;
    client.begin("http://www.wuweixin.com/iot");
    client.GET();
    client.end();
    Serial.printf("[A] ESP FreeHeap: %d\r\n", ESP.getFreeHeap());
  }
  
}
```

### Debug Messages

Console Log

```
IP address: 0.0.0.0
[B] ESP FreeHeap: 39840
[A] ESP FreeHeap: 38544
WiFi.getMode(): 3
IP address: 192.168.38.149
[B] ESP FreeHeap: 38904
[A] ESP FreeHeap: 38120
WiFi.getMode(): 3
IP address: 192.168.38.149
[B] ESP FreeHeap: 38856
[... 400 lines cut ...]
Request timeout for icmp_seq 165
Request timeout for icmp_seq 166
Request timeout for icmp_seq 167
Request timeout for icmp_seq 168
Request timeout for icmp_seq 169
Request timeout for icmp_seq 170
```

Another Arduino Uno Sketch

```cpp
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(9600);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  for (int i = 0; i < 64; i++) {
    Serial.write(0x0A);
  }
  digitalWrite(LED_BUILTIN, LOW);
  delay(500);
}
```

wuwx, 2017-02-06:
the wifi connection lost is not immediately
need waiting some minutes
just like unstable, how can i fix it ? thanks

wuwx, 2017-02-06:
if only call Serial2.readString()
the nodemcuv2 WiFi also disconnect after about 30minutes

```cpp
#include <Arduino.h>
#include <SoftwareSerial.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

SoftwareSerial Serial2(12, 14, false, 2048);

void setup()
{
  Serial2.setTimeout(50);
  Serial2.begin(9600);
  Serial.begin(115200);
  WiFi.mode(WIFI_AP_STA);
  WiFi.begin("SSID", "password");
}

void loop()
{
  if (Serial2.available()) {
    Serial.printf("WiFi.getMode(): %d\r\n", WiFi.getMode());
    Serial.printf("IP address: %s\r\n", WiFi.localIP().toString().c_str());
    Serial.printf("[B] ESP FreeHeap: %d\r\n", ESP.getFreeHeap());
    String payload = Serial2.readString();
    Serial.printf("[A] ESP FreeHeap: %d\r\n", ESP.getFreeHeap());
  }
  
}
```

ping log
```
64 bytes from 192.168.38.149: icmp_seq=2000 ttl=128 time=3.109 ms
64 bytes from 192.168.38.149: icmp_seq=2001 ttl=128 time=1.281 ms
64 bytes from 192.168.38.149: icmp_seq=2002 ttl=128 time=4.404 ms
64 bytes from 192.168.38.149: icmp_seq=2003 ttl=128 time=1.326 ms
64 bytes from 192.168.38.149: icmp_seq=2004 ttl=128 time=1.365 ms
64 bytes from 192.168.38.149: icmp_seq=2005 ttl=128 time=3.335 ms
64 bytes from 192.168.38.149: icmp_seq=2006 ttl=128 time=1.325 ms
64 bytes from 192.168.38.149: icmp_seq=2007 ttl=128 time=1.383 ms
64 bytes from 192.168.38.149: icmp_seq=2008 ttl=128 time=1.437 ms
64 bytes from 192.168.38.149: icmp_seq=2009 ttl=128 time=26.724 ms
64 bytes from 192.168.38.149: icmp_seq=2010 ttl=128 time=2.393 ms
64 bytes from 192.168.38.149: icmp_seq=2011 ttl=128 time=2.183 ms
64 bytes from 192.168.38.149: icmp_seq=2012 ttl=128 time=2.496 ms
64 bytes from 192.168.38.149: icmp_seq=2013 ttl=128 time=2.271 ms
64 bytes from 192.168.38.149: icmp_seq=2014 ttl=128 time=1.422 ms
64 bytes from 192.168.38.149: icmp_seq=2015 ttl=128 time=2.294 ms
64 bytes from 192.168.38.149: icmp_seq=2016 ttl=128 time=1.539 ms
64 bytes from 192.168.38.149: icmp_seq=2017 ttl=128 time=12.928 ms
Request timeout for icmp_seq 2018
Request timeout for icmp_seq 2019
Request timeout for icmp_seq 2020
Request timeout for icmp_seq 2021
Request timeout for icmp_seq 2022
Request timeout for icmp_seq 2023
Request timeout for icmp_seq 2024
Request timeout for icmp_seq 2025
Request timeout for icmp_seq 2026
Request timeout for icmp_seq 2027
Request timeout for icmp_seq 2028
Request timeout for icmp_seq 2029
Request timeout for icmp_seq 2030
Request timeout for icmp_seq 2031
Request timeout for icmp_seq 2032
Request timeout for icmp_seq 2033
Request timeout for icmp_seq 2034
Request timeout for icmp_seq 2035
Request timeout for icmp_seq 2036
Request timeout for icmp_seq 2037
Request timeout for icmp_seq 2038
Re
```

devyte, 2017-09-09:
@wuwx is this issue still valid with latest git?

drorgl, 2017-09-10:
I'm not the bug owner but I thought I could pitch in.

I did not verify it on the latest git, but as far as I can tell, nothing that could affect this problem has changed, I think I've traced the source to interrupt time, the SoftwareSerial uses interrupts to capture incoming bytes, the first bit triggers the interrupt and then it adds delays between the bits according to the baud rate, for 115200 the wifi subsystem freezes less but it still happens, for 9600 it freezes almost immediately. I think its a design flaw in the way wifi works in esp8266, could be a buffer overrun due to the lack of interrupt processing when SoftwareSerial does its work inside the interrupt.

In my case I've attempted to use a GPS receiver, so the amount of interrupts generated were relatively large, in any case, it would miss an interrupt every once in a while and I would miss a byte or two from the stream. 

I couldn't find the source code for the wifi, only precompiled libraries, does anyone know where to find it?

IMHO if you need a reliable serial connection, you should use a different board with multiple hardware serials, like the esp32.

devyte, 2017-10-01:
@drorgl the wifi functionlity is in the SDK. The core code is a wrapper around the SDK api to make it behave Arduino-like.
If I had to take a WAG, I'd say the problem may be due to interrupts being disabled for the serial ops. Due to the slow comm speed, it may take too long, and as a result cause problems with the wifi stack.

devyte, 2017-12-05:
@wuwx @drorgl Quite a bit has changed recently, including the possibility to use lwip2, which fixes a lot of socket instabilities. Could you please try latest git with lwip2?

earlephilhower, 2020-01-01:
This is nearly 3 years old and no feedback in >2yrs.  We've updated SWSerial, the SDK, and the core so much this seems to be no longer valid.  If someone has a similar issue with GIT head/latest release, please file a new report and fill out the info/MCVE so we can look into it.

closed 2020-01-01 by earlephilhower