esp8266/Arduino #2938: Error Message On Arduino Uno Wifi

FelixB91, 2017-02-06:
Hey everyone,

I recently bought an Arduino Uno Wifi which has an ESP8266 chip on board. I want to use it to collect a bunch of data from different sensors and write it into a MySQL database, but I'm struggling with it at the moment. I use the [MySQL_Connector_Arduino](https://github.com/ChuckBell/MySQL_Connector_Arduino) library and want to combine it with the ESP8266WiFi library.
Down below is my code and the error message it throws. I put the ESP8266WiFi folder in my [libraries](http://i.imgur.com/yh5UGdd.png) folder. I also use the Arduino IDE v1.8.1.
Any ideas why this is happening?

### Sketch

```cpp

#include <ESP8266WiFi.h>
#include <MySQL_Connection.h>
#include <MySQL_Cursor.h>

IPAddress server_addr(88, 198, 61, 231); // IP of the MySQL *server* here
char user[] = "USERxxxxx";              // MySQL user login username
char password[] = "xxxxxx";        // MySQL user login password

// Sample query
char INSERT_SQL[] = "INSERT INTO `rasp_pi_weather`(`Humidity`, `Temperature`, `DateTime`) VALUES (60, 23, '2017-02-02 20:34:20')";

WifiData client;
MySQL_Connection conn((Client *)&client);

void setup() {
  Serial.begin(9600);
  while (!Serial); // wait for serial port to connect
  
  Wifi.begin("wifi", "password");
  
  Serial.println("Connecting");
  
  if (conn.connect(server_addr, 3306, user, password)) {
    delay(5000);
    Serial.println("Connection successful");
  } else {
    Serial.println("Connection failed!");
  }   
}

void loop() {
  delay(2000);

  Serial.println("Recording data.");

  // Initiate the query class instance
  MySQL_Cursor *cur_mem = new MySQL_Cursor(&conn);
  // Execute the query
  cur_mem->execute(INSERT_SQL);
  // Note: since there are no results, we do not need to read any data
  // Deleting the cursor also frees up memory used
  delete cur_mem;
}
```

### Debug Messages

```

In file included from /Users/Felix/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFi.h:33:0,
                 from /Users/Felix/Documents/Arduino/SQL Test/basic_insert_3/basic_insert_3/basic_insert_3.ino:5:
/Users/Felix/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFiType.h:26:19: fatal error: queue.h: No such file or directory
 #include <queue.h>

                   ^
compilation terminated.
exit status 1
```

wuwx, 2017-02-06:
you need use global esp8266/Arduino project as a new board~~
not only a library

FelixB91, 2017-02-06:
okay, I downloaded the whole "ESP8266 core for Arduino" and extracted it, but don't know where to put it beside the libraries folder (where it does not show up). And I also don't understand how I can use a project as a new board.

igrr, 2017-02-06:
Please take your time and read installation instructions in the main Readme of this project.

wuwx, 2017-02-06:
Start Arduino and open Preferences window.
Enter http://arduino.esp8266.com/stable/package_esp8266com_index.json into Additional Board Manager URLs field. You can add multiple URLs, separating them with commas.
Open Boards Manager from Tools > Board menu and install esp8266 platform (and don't forget to select your ESP8266 board from Tools > Board menu after installation).

FelixB91, 2017-02-06:
Sorry @igrr for not doing this in the first place, I kind of just thought I knew how it works. And also thank you @wuwx for taking the time to explain it to me!

I did install everything like in the instruction and chose "Generic ESP8266 Module" from the board list. Trying a basic example like WiFiScan results in an error message again
```
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed
```
I tried some different settings (like a different board, upload speed, etc.) but all resulted in the same error. Since I'm only using an Arduino Uno WiFi with a build in ESP module I wonder if that could be the reason to this error?

wuwx, 2017-02-06:
try change board to: WeMos D1

FelixB91, 2017-02-06:
I tried to set the board to both "WeMos D1 R2 & mini" and also to "WeMos D1 (retired)" but all produce the same error message unfortunately.

lrmoreno007, 2017-02-06:
This project is not intended for your board, you should select "Arduino Uno Wifi" directly from the board selector.

http://www.arduino.org/learning/getting-started/getting-started-with-arduino-uno-wifi

FelixB91, 2017-02-06:
@lrmoreno007 yes, that's pretty much my problem. I know that I normally have to use the "Arduino Uno Wifi" board, but in this case I cannot use the ESP8266WiFi library (even though the Uno Wifi includes an ESP chip), which I (think I) need in order to get my sketch from my initial post working. If there's is any other solution to this problem I'll gladly take it, but from what I figured out so far no one can tell me how to insert anything into an MySQL Database from an Arduino Uno Wifi, which does not use an Wifi or Ethernet Shield.

lrmoreno007, 2017-02-07:
Making a mix between this project and an arduino one wifi is not the right way.

ESP8266 / Arduino uses its own CPU, its own I / O, communications, etc.

Arduino UNO Wifi uses ATMega328 for CPU, I / O, communications, etc. and connects via UART / I2C with the ESP8266 only to make the WIFI connection (and web server), ESP8266 in the Arduino UNO Wifi has her own firmware developed by Arduino Team for this specific task. This is the reason why you can't program ESP8266 directly or making a mix with libraries (although in reality, there is a way but it would make your Arduino ONE Wifi unusable).

In my opinion, you must learn to use the Arduino ONE WIFI libraries using the examples that the IDE brings you, by asking at http://www.arduino.org/forums, looking for other examples in the www.

Examples / WIFI / WifiWebClient can be a good starting point to stablish internet connection. Once connected, you just have to learn how to insert SQL from a client, it's a good starting point https://www.tweaking4all.com/hardware/arduino/arduino-ethernet-data-push/  Although the example is via ethernet, it's enought for you and don't need libraries for SQL if you don't want.

Please close this issue.

Regards

P.D.: Excuse my english.

closed 2017-02-07 by FelixB91

5ran6, 2018-01-24:
Finally on the matter, I had exactly the same issue which you're having and just like Irmono007 explained, that's the reason yours isn't working. So I found a sample sketch that worked from me. WifiWebServer.ino by Arduino core. Try it

Puhaz, 2018-01-29:
Hi FelixB91,
Have you solved your problem? i have a same problem with my project. Hope your inputs may help me.

5ran6, 2018-03-07:
I jut concluded the board is bad. Right now it isn't uploading codes at all
again. I just got a Node MCU board instead. Works like a charm. Good luck