esp8266/Arduino #2940: Issue installing 2.3.0 via Boards Manager

southr1, 2017-02-07:
Issue installing 2.3.0 via Boards Manager from:
http://arduino.esp8266.com/stable/package_esp8266com_index.json

Hardware:			NODEMCU V3
Core Version:      	attempting to load 2.3.0

### Description

Problem installing 2.3.0 from:
http://arduino.esp8266.com/stable/package_esp8266com_index.json
On the Boards Manager Pop up I get 
Error Downloading https://github.com/esp8266/arduino/releases/download/2.3.0/esp8266-2.3.0.zip
in red at the bottom.

If I try to download manually from this address it reaches 2.2MB and stops.  Saying Failed - Network error.

I have downloaded the file from another site but cannot see how to install it any other way.

### Settings in IDE

Module:  			ESP8266 Module on NODEMCU V3
Flash Size: 		4MB
CPU Frequency:	?
Flash Mode:		?
Flash Frequency:	?
Upload Using:		?
Reset Method:		nodemcu?

### Sketch

n/a

### Debug Messages

n/a
messages here
```

Tensa225, 2017-02-14:
check the readme file included in the core .zip

devyte, 2017-09-09:
There is currently no issue with installation under Windows and Linux. Closing.

closed 2017-09-09 by devyte