esp8266/Arduino #2941: ISR in C++ class?

MarcFinns, 2017-02-07:
A real pity it is not possible to create fully encapsulated functionality, as apparently Interrupt handlers cannot be class methods. Wouldn't this be a nice improvement to the platform?

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

me-no-dev, 2017-02-07:
it can not be used in C if done so.

MarcFinns, 2017-02-08:
Not strictly true. Overloading the function and having one that takes a handler with the correct signature of a class method is very possible!
Some other platforms have it already. 
Anyone that pics it up? (I don't have the tool chain to do it myself)

igrr, 2017-02-08:
https://github.com/esp8266/Arduino/pull/2745 comes pretty close to what you are probably looking for. It needs a bit if rework to avoid having to include an extra header file from the sketch.

Also, I would like this more if such generic enhancements would be taken up to arduino/Arduino repo (or at least to the official arm and wifi101 cores). Introducing these features here will not help with compatibility, if same features are not present in the official arduino cores.

me-no-dev, 2017-02-08:
can be included in CPP only if it's not going to clash with other code possibly running in C.
Say you have two libs, one C and one C++ and that both use attachInterrupt. Both should be able to do so, if not, the above could not do.

igrr, 2017-02-08:
The idea was to include above header under ifdef cplusplus block inside arduino.h.

me-no-dev, 2017-02-08:
exactly what I meant by my first sentence :)

hreintke, 2017-02-08:
I tried to put in Arduino.h but did not succeed in that.
Putting the `void attachInterrupt(uint8_t pin, std::function<void(void)> intRoutine, int mode);` gives error that std::function is not defined. Including `<functional>` additionally gives other errors I could not solve. 

If you give me a hint/link on how to solve I will try again.

igrr, 2017-02-08:
Yep, including STL headers from Arduino.h is a pain, in part due to the fact that Arduino.h defines min and max as macros. It's probably time to figure out how to replace these macros with definitions from the STL (for the ifdef cplusplus case).

devyte, 2017-09-09:
Functional interrups has been merged. For Tickers, there is something equivalent, if not as pretty.
Closing.

devyte, 2017-09-09:
@hreintke by the way, I think your FunctionalInterrupt code has a small and non-critical mem leak when you detach. This results from the call to new, which has no delete.
It's non-critical because it is _very_ unlikely that someone will attach-detach often within the same uptime. Of course, I could be wrong, which is why I'm pointing it out.
At first glance, I don't know how to fix it.

closed 2017-09-09 by devyte