esp8266/Arduino #2942: Information about the WatchDog Timers.

tttapa, 2017-02-08:
Could someone please give some more information about the WDT reset, or at least point me to a link?  
From my experiments, it seems that some functions, like Serial.flush() automatically leave some time for the WiFi code, and reset the WDT.  
If you don't use yield() or delay(0), the soft WDT seems to reset after exactly 3 seconds, and when disabled, the hardware WDT resets after ~8375933µs (>8 seconds). This seems to be independent of the clock speed (80/160MHz).

This is the code I used:

```cpp

void setup() {
  Serial.begin(115200);
  // ESP.wdtDisable();
}

void loop() {
  unsigned long startMicros = micros();
  while(1) {
    Serial.print(ESP.getCycleCount());
    Serial.print('\t');
    Serial.println(micros() - startMicros);
  }
}
```

What are the safe limits? What is the absolute maximum time without yield()'ing? Will the WiFi crash if I wait too long, or will it just be paused?  
What is the advantage of having a software WDT as well as a hardware WDT?

Thanks!   
Pieter

igrr, 2017-02-08:
Your observations are correct. There is some info in this PR: https://github.com/esp8266/Arduino/pull/2533/files.
It's not a good practice to spend more than a few hundred milliseconds without yielding to the wifi stack. The time that the TCP stack can tolerate without yielding depends on the amount of data being received, so it's hard to give exact numbers.

Regarding having two kinds of watchdogs, that's the feature of the underlying SDK. Soft WDT results in a more informative stack trace, while hardware wdt resets the chip if all else fails. So each has its own benefits.

tttapa, 2017-02-08:
Ok, thanks a lot!

closed 2017-02-08 by tttapa