esp8266/Arduino #2944: Updater (OTA) was not working realiable

Torxgewinde, 2017-02-08:
#### Hardware
Hardware:	LYT8266, An LED E27 Retrofit Lightbulb with ESP
Core Version:  2.3.0

### Description
Not really an issue anymore since I found a solution now, but I am still struggling with GIT to contribute in the intended way:

Uploading Firmware via OTA ESP8266HTTPUpdateServer was not reliable. The CURL script or HTTP Upload via Browser stalled or stopped. Initially upgrading the Firmware by HTTP Uploading it worked, then it stopped working for no apparent reason. Turns out that somehow my FLASH is perhaps a bit slower than initially or takes longer for erasing pages. Anyway, after several evenings of frustration I dug into the files and after changing "Updater.cpp" it is now working slow but reliable. The changes in Updater.cpp result in the following code. The changes are the three **delay(10)** functions that allow the FLASH to somehow do its thing. Just the yield()-function-calls alone seem to be not long enough.

```
      bool UpdaterClass::_writeBuffer(){

      if(!_async) yield();
      delay(10);
      bool result = ESP.flashEraseSector(_currentAddress/FLASH_SECTOR_SIZE);
      if(!_async) yield();
      delay(10);
      if (result) {
         result = ESP.flashWrite(_currentAddress, (uint32_t*) _buffer, _bufferLen);
      }
      if(!_async) yield();
      delay(10);

      if (!result) {
```

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB
CPU Frequency:	80Mhz
Flash Mode:		DIO
Flash Frequency:	40Mhz
Upload Using:		OTA and SERIAL
Reset Method:		ck

### Sketch
```
/*******************************************************************************
#                                                                              #
#      LYT-SCRIPTED, an alternative firmware for LYT8266 RGBW LED lamps        #
#                                                                              #
#                                                                              #
#      Copyright (C) 2017 Tom Stöveken                                         #
#                                                                              #
# This program is free software; you can redistribute it and/or modify         #
# it under the terms of the GNU General Public License as published by         #
# the Free Software Foundation; version 2 of the License.                      #
#                                                                              #
# This program is distributed in the hope that it will be useful,              #
# but WITHOUT ANY WARRANTY; without even the implied warranty of               #
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the                #
# GNU General Public License for more details.                                 #
#                                                                              #
# You should have received a copy of the GNU General Public License            #
# along with this program; if not, write to the Free Software                  #
# Foundation, Inc., 59 Temple Place, Suite 330, Boston, MA  02111-1307  USA    #
#                                                                              #
*******************************************************************************/

#include <ESP8266WebServer.h>
#include <ESP8266HTTPUpdateServer.h>
#include <FS.h>

ESP8266WebServer server(80);
File fsUploadFile;

ESP8266HTTPUpdateServer httpUpdater;

/******************************************************************************
[... 260 lines cut ...]
    
    server.send(200, "text/plain", response);
    });
  
  server.on("/edit", HTTP_POST, [](){ 
    if ( !enableUpdates ) {
      server.send(200, "text/plain", "locked");
      return;
    }
    
    server.send(200, "text/plain", "");
    }, handleFileUpload);

  server.begin();
}

/******************************************************************************
Description.: handle HTTP-clients
Input Value.: -
Return Value: -
******************************************************************************/
void loop_webserver() {
  server.handleClient();  
}
```

robertgregor, 2017-02-14:
Hello nice, please read my 2938 - I think, it can be related to it. I have discovered, that the delay(10) is needed for the ESP FW to do socket maintenance tasks, if it is not done in some heavy operation, the socket is closed by the FW.

devyte, 2017-09-09:
@Torxgewinde is this issue still valid with latest git?

Torxgewinde, 2017-09-09:
Yes, I manually ensure the delays are added to the file. My issue only showed up after I did many many reflashes, as if the flash gets slower to accept new data. Initially it was working fine without the modifications

devyte, 2017-09-09:
@Torxgewinde core 2.3.0 stable is very old. The latest git uses SDK2.0.0. There is also a branch with SDK 2.1.0, but I don't think that has anything relevant to this issue (I could be wrong).

> only showed up after I did many many reflashes

How many is "many many"? dozens? hundreds? thousands?
It is normal for flash to slow down over time with use. I somehow doubt these chinese flash chips are superduper topoftheline high quality with more writecycles than can be used in a human life time, so the max write cycles is probably rather low.
In addition, there are several cases where spcific sectors of the flash get overwritten again and again, such as the SDK config area, or the EEPROM emulation. Those sectors can wear out much faster than the rest (and eventually fail making the module useless). If you then change SDK, the layout could change, moving those areas to other less-worn sectos. I'm not saying that is a certainty, I would have to delve into the differences between the 2.3.0 and latest git flash layouts, but I do know of things that have changed in the past.
The consequence of the above is that it _might_ be better with latest git, or conversely, your issue _might_ be consequence of a previous different layout, where a worn sector is now in use by the sketch image.
There are currently no known ways around what I just explained (except for those bad habits, like using the EEPROM or direct access to the flash, in which case that should be replaced with storing data on the SPIFFS). You just have to accept that the ESP flash write time may become slower over time, and that the module may fail earlier than expected because of specific wear.
The only thing you can do is replace the module with a new one from a "trustworthy" vendor.

As to your changes, I think that your having found a workaround is pretty awesome. However, merging it would make OTA flashing slower for other users who don't have a slowed-down flash (yet). In addition, the if(!_async) is there for a reason: it skips yield and delay calls altogether, to allow use from async callbacks (e.g.: the async webserver). Your delays, the way you have proposed them, don't include that logic, and would just crash when called from async callbacks. Finally, even with your workaround, your proposal wouldn't work from async callbacks, because the delays would be skipped.

So what to do...

Alright, here is a suggestion:
- Add a private member uint8_t delayTime to the updater class
- In the current constructor, init delayTime(0)
- Add a setter method, e.g.: setFlashDelay, that sets the above member
- Fix your delay logic: rather than add delays, replace the yield()s with delay(delayTime). 

The above doesn't fix all use cases, but it's an improvement over the current situation, it maintains current behavior for normal users, it doesn't really add significant complexity, and allows your use case for OTA, i.e.: others having slow flash chips with sync OTA can be directed to make use of it.

Please make a PR with the above to discuss further.

Torxgewinde, 2017-09-15:
Ok, but it will take some time before I'll make a PR.

Torxgewinde, 2017-11-18:
Hello,
I finally found the time to test the lastest ESP8266-Arduino trunk and now HTTP-updates just work without my patch.

I am not sure what changed, but now it works. I tested again with the LYT8266 Lightbulb that was having the issues.

I think this issue can be closed then. Thanks for the hints and ideas!

closed 2017-11-18 by Torxgewinde