esp8266/Arduino #2947: HTTPClientAuthorization Gmail Atom: connection refused

tttapa, 2017-02-09:
### Basic Infos
Hi, 
I was wondering if it's possible to get data from https://mail.google.com/mail/feed/atom using the HTTPClient on the ESP8266. (To get things like unread emails, and most recent emails.)
#### Hardware
Hardware:			ESP-07
Core Version:      	2.3.0

### Description

Every time I send a GET request to the above URL, I get `connection refused`.
This is the sketch I'm using (pretty much the same as the ESP8266HTTPClient/Authorization example):

```cpp

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>

#define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti;

void setup() {
    USE_SERIAL.begin(115200);
    USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

    for(uint8_t t = 4; t > 0; t--) {
        USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
        USE_SERIAL.flush();
        delay(1000);
    }

    WiFiMulti.addAP("WiFi SSID", "WiFi password");
}

void loop() {
    // wait for WiFi connection
    if((WiFiMulti.run() == WL_CONNECTED)) {

        HTTPClient http;
        USE_SERIAL.print("[HTTP] begin...\n");
        // configure traged server and url
        
        //http.begin("https://raw.githubusercontent.com/tttapa/MIDI_controller/master/keywords.txt","21 99 13 84 63 72 17 13 B9 ED 0E 8F 00 A5 9B 73 0D D0 56 58"); // WORKS
        //http.begin("http://172.217.17.133/mail/feed/atom/"); // DOESN'T WORK
        //http.begin("https://www.mail.google.com/mail/feed/atom",80,"/", "47 81 80 0C 61 B7 39 18 64 CC 9B 3A 8E 12 69 0E AD EB 84 B7"); // DOESN'T WORK
        http.begin("https://mail.google.com",443,"/mail/feed/atom/", "47 81 80 0C 61 B7 39 18 64 CC 9B 3A 8E 12 69 0E AD EB 84 B7"); // DOESN'T WORK
        //http.begin("https://mail.google.com",80,"/mail/feed/atom/", "47 81 80 0C 61 B7 39 18 64 CC 9B 3A 8E 12 69 0E AD EB 84 B7"); // DOESN'T WORK
        //http.begin("https://mail.google.com/mail/feed/atom",80,"/", "47 81 80 0C 61 B7 39 18 64 CC 9B 3A 8E 12 69 0E AD EB 84 B7"); // DOESN'T WORK
        //http.addHeader("Authorization", "Basic xAZ21haWwuYqsdfmlkjflmqksdjfml9tOkqsdfjIwMTc="); // echo -n 'email.address@gmail.com:password' | base64
        http.setAuthorization("email.address@gmail.com","password");
        http.addHeader("Host", "mail.google.com");
        http.addHeader("User-Agent", "Mozilla/4.0 (compatible;)");

        USE_SERIAL.print("[HTTP] GET...\n");
        // start connection and send HTTP header
        int httpCode = http.GET();

        // httpCode will be negative on error
        if(httpCode > 0) {
            // HTTP header has been send and Server response header has been handled
            USE_SERIAL.printf("[HTTP] GET... code: %d\n", httpCode);

            // file found at server
            if(httpCode == HTTP_CODE_OK) {
                String payload = http.getString();
                USE_SERIAL.println(payload);
            }
        } else {
            USE_SERIAL.printf("[HTTP] GET... failed, error: %d %s\n", httpCode, http.errorToString(httpCode).c_str());
            String payload = http.getString();
            USE_SERIAL.println(payload);
        }

        http.end();
    }

    delay(10000);
}
```

### Debug Messages

```
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with SSID, channel 1
dhcp client start...
[SETUP] WAIT 3...
[SETUP] WAIT 2...
[SETUP] WAIT 1...
state: 5 -> 0 (0)
rm 0
f r0, scandone
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with SSID, channel 1
dhcp client start...
ip:192.168.1.24,mask:255.255.255.0,gw:192.168.1.1
[HTTP] begin...
[HTTP] GET...
pm open,type:2 0
[HTTP] GET... failed, error: -1 connection refused
```

Is this possible? I've seen [this blog](http://www.areresearch.net/2016/07/gmail-notifier-with-esp8266-nodemcu.html) doing it using Lua, but I'm not able to do the same using the Arduino Core.
What am I doing wrong? Or could this be a bug?

Thanks!  
Pieter

tttapa, 2017-02-11:
I figured it out, I had to use WiFiClientSecure instead of HTTPClient. I don't understand what the difference between the two is though, could someone please explain this?
Thanks.

This is the final sketch I used. You'll have to allow access to "less secure apps" in your [Gmail settings](https://www.google.com/settings/security/lesssecureapps), and it doesn't work if you have two-factor authentication turned on. Untill we have support for OAuth2 on ESP8266, I don't think there's a better way ...
```cpp
#include <WiFiClientSecure.h>   // Include the HTTPS library
#include <ESP8266WiFi.h>        // Include the Wi-Fi library
#include <ESP8266WiFiMulti.h>   // Include the Wi-Fi-Multi library

ESP8266WiFiMulti wifiMulti;     // Create an instance of the ESP8266WiFiMulti class, called 'wifiMulti'

const char* host = "mail.google.com"; // the Gmail server
const int httpsPort = 443;            // the port to connect to the email server

                                      // The SHA-1 fingerprint of the SSL certificate for the Gmail server (see below)
const char* fingerprint = "3C 30 52 70 0C B8 BA AF 0E 16 9A FA 50 01 79 0A 0F 74 35 81";

                                      // The Base64 encoded version of your Gmail login credentials (see below)
const char* credentials = "aVeryLongStringOfBase64EncodedCharacters=";

void setup() {
  Serial.begin(115200);         // Start the Serial communication to send messages to the computer
  delay(10);
  Serial.println('\n');

  wifiMulti.addAP("ssid_from_AP_1", "your_password_for_AP_1");   // add Wi-Fi networks you want to connect to
  wifiMulti.addAP("ssid_from_AP_2", "your_password_for_AP_2");
  wifiMulti.addAP("ssid_from_AP_3", "your_password_for_AP_3");

  Serial.println("Connecting ...");
  int i = 0;
  while (wifiMulti.run() != WL_CONNECTED) { // Wait for the Wi-Fi to connect: scan for Wi-Fi networks, and connect to the strongest of the networks above
    delay(1000);
    Serial.print(++i); Serial.print(' ');
  }
  Serial.println('\n');
  Serial.print("Connected to ");
  Serial.println(WiFi.SSID());              // Tell us what network we're connected to
  Serial.print("IP address:\t");
  Serial.println(WiFi.localIP());           // Send the IP address of the ESP8266 to the computer
  Serial.println('\n');
}

void loop() {
  int unread = getUnread();
  if (unread == 0)
    Serial.println("\r\nYou've got no unread emails");
  else if (unread > 0)
    Serial.printf("\r\nYou've got %d new messages\r\n", unread);
  else
    Serial.println("Could not get unread mails");
  Serial.println('\n');
  delay(5000);
}

int getUnread() {    // a function to get the number of unread emails in your Gmail inbox
  WiFiClientSecure client; // Use WiFiClientSecure class to create TLS (HTTPS) connection
  Serial.printf("Connecting to %s:%d ... \r\n", host, httpsPort);
  if (!client.connect(host, httpsPort)) {
    Serial.println("Connection failed");
    return -1;
  }

  if (client.verify(fingerprint, host)) {   // Check the SHA-1 fingerprint of the SSL certificate
    Serial.println("Certificate matches");
  } else {                                  // if it doesn't match, it's not safe to continue
    Serial.println("Certificate doesn't match");
    return -1;
  }

  String url = "/mail/feed/atom";
  Serial.print("Requesting URL: ");
  Serial.println(url);

  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "Authorization: Basic " + credentials + "\r\n" +
               "User-Agent: ESP8266\r\n" +
               "Connection: close\r\n\r\n");

  Serial.println("Request sent");

  int unread = -1;

  while (client.connected()) {                          // the response is in XML format
    client.readStringUntil('<');                        // read until the first XML tag
    String tagname = client.readStringUntil('>');       // read until the end of this tag to get the tag name
    if (tagname == "fullcount") {                       // if the tag is <fullcount>, the next string will be the nubmer of unread emails
      String unreadStr = client.readStringUntil('<');   // read until the closing tag (</fullcount>)
      unread = unreadStr.toInt();                       // convert from String to int
      break;                                            // stop reading
    }                                                   // if the tag is not <fullcount>, repeat and read the next tag
  }
  Serial.println("Connection closed");

  return unread;
}

/*
 * # How to get the SHA-1 fingerprint for the Gmail server?
 * 
 *   Use the following command in a terminal window:
 * 
 *      openssl s_client -connect mail.google.com:443 < /dev/null 2>/dev/null | openssl x509 -fingerprint -noout -in /dev/stdin | sed 's/:/ /g'
 *      
 * # How to encode your credentials to Base64?
 * 
 *   Use the following command in a terminal window:
 * 
 *      echo -n "email.address@gmail.com:password" | base64
 */
```

Pieter

devyte, 2017-09-09:
@tttapa HTTPClient represents an unencrypted HTTP client connection. It (probably) uses WiFiClient under the hood, which represents an unencrypted socket.
WiFiClientSecure represents an encryped socket.
Thank you for posting the working sketch at the end.
Having said that, this is not the right place to ask "is this possible" type questions. This is an issue tracker, meant for tracking issues in the core. Please ask such questions at esp8266.com or stackoverflow.
Closing.

closed 2017-09-09 by devyte