esp8266/Arduino #2949: ESP module crashes when using HttpClient in .NET/C# program to upload OTA firmware

huming2207, 2017-02-10:
Hello guys,

I've got a NodeMCU 1.0 dev board  and I would like to upload my firmware from my own C# client program. But somehow the board crashes or return HTTP 504 status code when I start to upload the firmware.

**I guess there are some memory allocation issues when the board receiving HTTP POST.  Anyone who have any idea about this issue? Please help me soon! Thanks!**

**I have also tried to simulate cURL or Google Chrome web browser's HTTP headers and HTTP POST form data information. But it also causes the board crashes or return 504 status code.**

I have two version of my C# program, here is the first one, **which the board WILL NOT crash but return a 504 Gateway timeout status code:**

```csharp
using System;
using System.Collections.Generic;
using System.Linq;
using System.Text;
using System.Threading;
using System.Threading.Tasks;
using System.Net.Http;
using System.IO;

namespace ConsoleApp2
{
    class Program
    {
        static void Main(string[] args)
        {
            HttpClient client = new HttpClient();
            client.BaseAddress = new Uri("http://esp8266-ip-address");
        
            byte[] firmwareContent = File.ReadAllBytes(@"C:\Users\username\Desktop\firmware-esp.bin");
            HttpContent httpContent = new ByteArrayContent(firmwareContent);
            httpContent.Headers.Add("Content-Type", "application/octet-stream");
            MultipartFormDataContent formContent = new MultipartFormDataContent("-------test");
            formContent.Add(httpContent, "update", "firmware.bin");
            var result = client.PostAsync("/update", formContent).Result;
            Console.WriteLine(result.RequestMessage.ToString());
            Console.ReadKey();
        }
    }
}
```

Here is another program, which will cause the board crashes and print the stack trace information (mentioned below after this code):

```csharp
using System;
using System.Collections.Generic;
using System.Linq;
using System.Text;
using System.Threading;
using System.Threading.Tasks;
using System.Net.Http;
using System.IO;

namespace ConsoleApp2
{
    class Program
    {
        static void Main(string[] args)
        {
            HttpClient client = new HttpClient();
            client.BaseAddress = new Uri("http://192.168.1.104");
        
            byte[] firmwareContent = File.ReadAllBytes(@"C:\Users\username\Desktop\firmware-esp.bin");
            HttpContent httpContent = new ByteArrayContent(firmwareContent);
            var result = client.PostAsync("/update", httpContent).Result;
            Console.WriteLine(result.RequestMessage.ToString());
            Console.ReadKey();
        }
    }
}
```

Here is some details about the crash and serial printed debug information:

### Hardware
Hardware:			NodeMCU 1.0
Core Version:      	2.3.0

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		OTA and SERIAL
Reset Method:		nodemcu

### Sketch

Using default **WebUpdater.ino** sample from **ESP8266HTTPUpdateServer** library

### Debug Messages

**Serial output:**

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld

Booting Sketch...
wifi evt: 0
wifi evt: 3
HTTPUpdateServer ready! Open http://esp8266-webupdate.local/update in your browser
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
:urn 85
:urd 28, 85, 13
:urch 85, 137
:urch 137, 49
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
WS:ac
:ref 1
WS:av
:ref 2
:ur 2
:ref 2
:ur 2
:rn 255
:c0 1, 255
:rn 1460
:rch 1460, 1460
:rch 2920, 1460
:c 1, 1460, 4380
:c 1, 1460, 2920
:c0 1, 1460
:rn 1460
:c0 1, 1460
:rn 1460
:rch 1460, 1460
:rch 2920, 1460
:c 1, 1460, 4380
:c 1, 1460, 2920
:c0 1, 1460
:rn 1460
:c0 1, 1460
:rn 1460
:rch 1460, 1460
:rch 2920, 1460
:c 1, 1460, 4380
:c 1, 1460, 2920
:c0 1, 1460
:rn 1460
:c0 1, 1460
:rn 1460
:rch 1460, 1460
:rch 2920, 1460
:c 1, 1460, 4380
:c 1, 1460, 2920
:c0 1, 1460
:rn 1460
check_poison_block is called for free block 0x3fff2f00
Fatal exception 3(LoadStoreErrorCause):
epc1=0x4010011d, epc2=0x00000000, epc3=0x00000000, excvaddr=0x400384a0, depc=0x00000000

Exception (3):
epc1=0x4010011d epc2=0x00000000 epc3=0x00000000 excvaddr=0x400384a0 depc=0x00000000

ctx: cont 
sp: 3fff0810 end: 3fff0af0 offset: 01a0

>>>stack>>>
3fff09b0:  00000000 00005b4d 00000000 401003e9  
3fff09c0:  3ffef9dc 0000037e 0000037e 4010020c  
3fff09d0:  3fff2f0c 3ffef008 3fff2f04 401008e0  
3fff09e0:  3fff2f0c 3ffef008 3ffeefe4 402057f0  
3fff09f0:  00000000 00000000 00000000 3fff2eac  
3fff0a00:  0000004f 0000000a 3fff2dcc 0000001f  
3fff0a10:  0000000a 3fff2d8c 0000000f 00000000  
3fff0a20:  3fff2d2c 0000000f 00000000 3fff2d0c  
3fff0a30:  0000000f 00000000 3fff2cec 0000000f  
3fff0a40:  00000007 3fff2ccc 0000000f 00000004  
3fff0a50:  3fff2e2c 0000006f 00000000 3ffefad0  
3fff0a60:  00000000 0000558c 0003e7a0 00000000  
3fff0a70:  00000000 000005b4 0000000a 3ffefad0  
3fff0a80:  00000000 00000000 3ffeefe4 3ffefac0  
3fff0a90:  000000ff 3ffef008 3ffeefe4 4020472f  
3fff0aa0:  3ffe87b8 00000000 000003e8 3ffeefe4  
3fff0ab0:  00000000 3fff2d54 3ffefa04 40202282  
3fff0ac0:  3fffdad0 00000000 3ffefab8 402022a0  
3fff0ad0:  3fffdad0 00000000 3ffefab8 40208794  
3fff0ae0:  feefeffe feefeffe 3ffefad0 40100970  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
```

**Decoded exception stack trace:**

```
0x401003e9: check_poison_block at /Users/Jackson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc.c line 835
0x4010020c: _umm_free at /Users/Jackson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc.c line 1287
0x401008e0: free at /Users/Jackson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc.c line 1733
0x402057f0: readBytesWithTimeout at /Users/Jackson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/Parsing.cpp line 55
:  (inlined by) ESP8266WebServer::_parseRequest(WiFiClient&) at /Users/Jackson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/Parsing.cpp line 173
0x4020472f: ESP8266WebServer::handleClient() at /Users/Jackson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/detail/RequestHandlersImpl.h line 53
0x40202282: setup at /Users/Jackson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266HTTPUpdateServer/examples/WebUpdater/WebUpdater.ino line 38
0x402022a0: loop at /Users/Jackson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266HTTPUpdateServer/examples/WebUpdater/WebUpdater.ino line 42
0x40208794: loop_wrapper at /Users/Jackson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_main.cpp line 56
0x40100970: cont_norm at /Users/Jackson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/cont.S line 109
```

Torxgewinde, 2017-02-10:
That looks very similar to the issues i had after flashing several times...

Can you try and modify Updater.cpp as described here?

https://github.com/esp8266/Arduino/issues/2944

huming2207, 2017-02-10:
@Torxgewinde Thanks for your reply. I've tried it and it still remains the same as it was before. Maybe there is somewhere wrong with my C# program, but I've searched around and it seems that it  is the right way to use C# to process POST request.

umartechboy, 2017-02-19:
Happened all the time with me untill I switched to Arduino IDE with visual micro. Node MCU is not so efficient with the memory. You can easily overflow its limit and it will crash. Just use ESP in arduino instead, gives you much better control.

devyte, 2017-09-09:
If still valid, I suggest using e.g.: wireshark to figure out the diffence between the POST of your tool and a POST from a browser.

devyte, 2017-11-11:
Closing due to no feedback in over 2 months.

closed 2017-11-11 by devyte

huming2207, 2017-11-11:
sorry mate got too busy these days...
I'll try again later in the next a few weeks and re-open this issue if necessary

devyte, 2017-11-11:
I'll reopen if it makes sense once you add more info.
Consider latest git, with prebuilt lwip or lwip2.