esp8266/Arduino #2950: BUG: [APdisconnect] set_config failed!

tablatronix, 2017-02-10:
### Basic Infos

#### Hardware
Hardware:			nodemcu v1
Core Version:      	        2.3

### Description

When I call softap or I get debug level errors, on config `ret` variable failures
WiFi.softAPdisconnect();  

### Settings in IDE

Module:  			nodemcu v1 amica
Flash Size: 	       4MB/1MB
CPU Frequency:      80Mhz
Flash Mode:	        qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu?

### Sketch
Cannot reproduce outside of larger sketch

What in the world would cause this ?

tablatronix, 2017-02-11:
my board also get in a persistent mode AP+STA, and no matter what i do i a sketch to remove the AP, doesn't work. 

set_config failed!
always , I am guessing this is the problem.

```
    	WiFi.persistent(true);
    	WiFi.mode(WIFI_STA);    	
    	WiFi.softAPdisconnect(true);
```
reboot, printDiag still mode `STA+AP`
driving me insane.

I guess Ill throw master on here and see if its fixed

tablatronix, 2017-02-12:
looks like the only way I got it to work so far was to make it AP only, the disconnect still returns false, and says failed but it works, 

```
    ETS_UART_INTR_DISABLE();
    ret =  wifi_set_opmode_current(WIFI_AP);
    ETS_UART_INTR_ENABLE();
    WiFi.mode(WIFI_STA); // turns sta back on
```

devyte, 2017-09-09:
@tablatronix is this still an issue with latest git? To be honest, I don't fully understand your problem description.

tablatronix, 2017-09-09:
Ill check again, I have been waiting for rc1

devyte, 2017-10-27:
@tablatronix rc2 is out. Have you tested if this is still valid?

tablatronix, 2017-10-27:
Sorry I have not, let me install master real quick

tablatronix, 2017-11-03:
I have some example sketches and a bug with apdisconnect failing also now in rc2 I will post them tomorrow.

tablatronix, 2017-11-04:
created, #3793 
This issues was confusing and all over the place.

closed 2017-11-04 by tablatronix