esp8266/Arduino #2951: BUG : stringify macro with spaces in string has faulty result

gerardwr, 2017-02-11:
### Basic Infos

Arduino 1.6.9 on Mac

#### Hardware

Hardware:		Wemos D1 Mini ESP-12
Core Version:      	2.3.0

### Description

I re-use an Arduino sketch that contains an awkward "stringify" construct, that I must admit do not fully grasp.

It composes a filename from a path and a filename.

The path contains spaces!

Running the sketch on a Wemos D1 this stringify gets a faulty result.

When running the same sketch on an Arduino Nano the result is as expected.

When I remove the spaces from the pathname the Wemos also delivers the correct result.

My conclusion : there is a bug in the "stringify" for the ESP, and the original Arduino code is "bugfree".

I can use a workaround, but because I think this is a bug, I report it here.

Please tell me when the fault is mine ;-)

Thx. 

### Sketch

```cpp
#define SKETCH_PATH /Users/gerard/Documents/My Data/ESP8266 Arduino/RFLink

// to include Config_xx.c files:
#define stringify(x) #x
#define CONFIGFILE2(a, b) stringify(a/Config/b)
#define CONFIGFILE(a, b) CONFIGFILE2(a, b)
#define CONFIG_FILE Config_01.c
//#include CONFIGFILE(SKETCH_PATH,CONFIG_FILE)
#define even CONFIGFILE(SKETCH_PATH,CONFIG_FILE)

void setup() {
  Serial.begin(57600);                                                           // Initialise the serial port
  Serial.println(even);
}

void loop() {}

```

Compile & Run sketch on Arduino Nano (as expected) leads to:
```
/Users/gerard/Documents/My Data/ESP8266 Arduino/RFLink/Config/Config_01.c
```
Compile & Run the same sketch on ESP8266 leads to (serious error in composed name):
```
/Users/gerard/Documents/My Data/1 Arduino/RFLink/Config/Config_01.c
```
Compile & Run the same sketch on ESP8266, but removing the spaces in SKETCH_PATH (same as on Nano, so OK):
```
/Users/gerard/Documents/MyData/ESP8266Arduino/RFLink/Config/Config_01.c
```

igrr, 2017-02-11:
This is because 'ESP8266' token is defined to 1 when compiling for the ESP8266. It gets replaced with 1 by the preprocessor.

This define is added on the command line and is the equivalent of adding #define ESP8266 1 in the program itself.

When you remove the space, token becomes 'ESP8266Arduino' and no replacement takes place.

gerardwr, 2017-02-11:
Hi Ivan, long time no see! I read that you're extremely busy, so I'm pleasantly surprised to here from you.

Thanks for your explanation, this issue was driving me nuts!

So it's not a bug it's a feature ;-)

Will change the path-level from "ESP8266 Arduino" to "ESP Arduino" or so.

Issue closed, thanks.

closed 2017-02-11 by gerardwr