esp8266/Arduino #2952: No information about PWM frequency

Vitorbnc, 2017-02-11:
Hi,
I'd like to know what is the maximum pwm frequency for esp-01. I found the function `void analogWriteFreq(uint32_t freq);`, but no information about the max value that could be set.
I needed pwm in the MHz range if possible, could it be done?

tttapa, 2017-02-13:
According to the [datasheet](http://espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf):

> PWM frequency range is
adjustable from 1000 μs to 10000 μs, i.e., between 100 Hz and 1 kHz.

Why do you need such a high frequency? I'm not aware of any normal microcontroller that can reach PWM frequencies in the MHz range.

Pieter

devyte, 2017-09-09:
@Vitorbnc I believe I have seen somewhere the ESP being used with PWM up to 40KHx. No idea how or where right now, but I do remember it was on google. No way can it go to MHz range.
In any case, this is not the right place to ask such questions. This is an issue tracker, meant for tracking issues in the core. Please refer to esp8266.com or stackoverflow for discussion.
Closing.

closed 2017-09-09 by devyte