esp8266/Arduino #2955: Possible to make ESP8266WebServer#_handleRequest virtual?

sidoh, 2017-02-12:
(I'm omitting a bunch of the requested info because it didn't seem relevant. Please forgive me if I'm mistaken)

### Basic Infos

I'd like to write an extension for `ESP8266WebServer` that makes enabling authentication a little more ergonomical. The interface I'm using now is:

```c++
// to enable authentication for all routes
server->requireAuthentication("username", "password");

// to disable authentication for all routes
server->disableAuthentication();
```

I considered overriding `on` to wrap all handlers internally, but this would significantly complicate the auth toggle logic since they'd have to wrap/unwrap existing handlers. `on` is also not virtual, so I guess there could be some funky behavior due to that.

A very low-cost way of supporting this is to override `_handleRequest`:

```c++
void WebServer::_handleRequest() {
  if (this->authEnabled 
    && !this->authenticate(this->username.c_str(), this->password.c_str())) {
    this->requestAuthentication();
  } else {
    ESP8266WebServer::_handleRequest();
  }
}
```

Maybe it's sort of awful to override an internal method like this, but this is actually a pretty nice place to inject custom logic that wraps each request. I managed to hack around the non-virtualness by copy/pasting the source for `handleClient`. It works really well, and it feels like a nice interface.

How horrible would it be to make `_handleRequest` virtual? Alternatively could maybe have a virtual method that defaults to NOOP which is called by `_handleRequest`?

Disclaimer: I'm not super comfortable with C++, so it's entirely possible there's a sane way to do what I want without changing the library code.

devyte, 2017-09-09:
@sidoh why do you need it to be virtual? the only case where that would make sense is with implementaion of a derived class, and polymorphism with manipulation through a pointer to the base class. I can't see that making sense in any scenario within this context.
While your question aims at a change in the core libs, you are really asking how to solve your particular problem. This is not the right place to ask such questions. This is an issue tracker, meant for tracking issues in the core. Please refer to esp8266.com or stackoverflow for discussion.
Closing.

closed 2017-09-09 by devyte

sidoh, 2017-09-09:
@devyte - I understand that you feel this is an inappropriate place to ask for programming guidance.  It was not my intention to ask for programming help.  I agree this is not the appropriate place for that.  

I probably got a little too far into the weeds of how I hacked together a solution.  I was more curious to discuss with people familiar with ESP8266WebServer how best to support this feature, because it seems basic enough to be worthwhile. 

If I could try again:

I'm pretty sure the only way to achieve HTTP auth given how ESP8266WebServer is currently structured is to add logic to every request handler, which seems unnecessarily burdensome.

I would propose an interface similar to the one I pasted above:

```
// to enable authentication for all routes
server.requireAuthentication("username", "password");

// to disable authentication for all routes
server.disableAuthentication();
```

Probably a more complete interface would allow for multiple usernames/passwords.

Perhaps this is better structured as an issue?

sidoh, 2017-09-09:
And obviously a better title would be something like:

> Global HTTP authentication for ESP8266WebServer

I can create a new issue if that's better. If you still think this is inappropriate for the issue board, that's obviously fine too. :)