esp8266/Arduino #2957: Handling binary requests crashes

guatedude2, 2017-02-13:
### Basic Infos

#### Hardware
Hardware:			HiLetgo New Version NodeMCU/ESP8266
Core Version:      	?2.1.0-rc2?

### Description

Running the AdvancedWebServer demo, found a bug that crashes any sketch with the webserver.

When using Postman to make a POST call with a binary file to any uri of the device's ip causes it to crash:

Here's an example. (No headers have been added)
[image]

### Settings in IDE

Module:  			NodeMCU ESP-12E
Flash Size: 		4M
CPU Frequency:		160MHz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		Arduino IDE (esptool.py)
Reset Method:		?ck / nodemcu?

### Sketch

```cpp
/*
 * Copyright (c) 2015, Majenko Technologies
 * All rights reserved.
 *
 * Redistribution and use in source and binary forms, with or without modification,
 * are permitted provided that the following conditions are met:
 *
 * * Redistributions of source code must retain the above copyright notice, this
 *   list of conditions and the following disclaimer.
 *
 * * Redistributions in binary form must reproduce the above copyright notice, this
 *   list of conditions and the following disclaimer in the documentation and/or
 *   other materials provided with the distribution.
 *
 * * Neither the name of Majenko Technologies nor the names of its
 *   contributors may be used to endorse or promote products derived from
 *   this software without specific prior written permission.
 *
 * THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS" AND
 * ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED
 * WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
 * DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE FOR
 * ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES
 * (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES;
 * LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON
 * ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
 * (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE OF THIS
 * SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
 */

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

const char *ssid = "House on the left";
const char *password = "reddoor81";

ESP8266WebServer server ( 80 );

const int led = 13;

[... 149 lines cut ...]
v09f0c112
~ld
```

### Stack Trace

```
Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
Decoding 15 results
0x4010011d: umm_assimilate_up at /Users/Alex/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc.c line 1163
0x40100668: _umm_realloc at /Users/Alex/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc.c line 1629
:  (inlined by) realloc at /Users/Alex/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc.c line 1709
0x4010020c: _umm_free at /Users/Alex/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc.c line 1287
0x4010068c: free at /Users/Alex/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc.c line 1733
0x40205488: readBytesWithTimeout at /Users/Alex/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/Parsing.cpp line 55
:  (inlined by) ESP8266WebServer::_parseRequest(WiFiClient&) at /Users/Alex/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/Parsing.cpp line 173
0x4020307c: WiFiClient::~WiFiClient() at /Users/Alex/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/WiFiClient.cpp line 149
0x402043c7: ESP8266WebServer::handleClient() at /Users/Alex/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/detail/RequestHandlersImpl.h line 53
0x40202424: setup at /var/folders/wt/53wjtlfj2p91g40lh1wc79rc0000gn/T/arduino_modified_sketch_458405/AdvancedWebServer.ino line 123
0x402078a0: std::_Function_base::_Base_manager ::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at /Users/Alex/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 1934
0x4020788c: std::_Function_handler ::_M_invoke(std::_Any_data const&) at /Users/Alex/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2069
0x40202440: loop at /var/folders/wt/53wjtlfj2p91g40lh1wc79rc0000gn/T/arduino_modified_sketch_458405/AdvancedWebServer.ino line 127
0x402076c4: loop_wrapper at /Users/Alex/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at /Users/Alex/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/cont.S line 109
```

guatedude2, 2017-02-14:
Might be related to https://github.com/nodemcu/nodemcu-firmware/issues/1546

devyte, 2017-09-09:
@guatedude2 What is your core version? Please fill out the relevant fields left blank.
Is this still valid with latest git?
I don't think it's related to the nodemcu one. In their case it's the http client, while here it's the server.

devyte, 2017-10-11:
Old issue, and no reply in over a month => closing.

closed 2017-10-11 by devyte

ElectroSoft, 2018-06-13:
I have the same problem with the svg files.
I'm using a generic esp8266