esp8266/Arduino #2959: aws implementation @ ESP8266

remesh123, 2017-02-13:
I am using ESP8266 huzzah board, using arduino IDE 1.8.1.0.
i wanted write a program to publish some data to AWS  i am attaching code here .
i have used https://github.com/daniele-salvagni/aws-sdk-esp8266
error :::::::::::::::::::::::::
2.	WiFi connected
3.	IP address: 
4.	192.168.15.105
5.	A2N.   .amazonaws.com
6.	443
7.	POST /things/4_7-Thing/shadow HTTP/1.1
8.	Content-Type: application/json
9.	Connection: close
10.	Content-Length: 98
11.	Host: A26.iot.us-east-1.amazonaws.com
12.	x-amz-content-sha256: f02ceea3888d05c51fbe521503429a1690e82644c5392cb5ec30a6478f2f06a1
13.	x-amz-date: 20170213T093950Z
14.	Authorization: AWS4-HMAC-SHA256 Credential=AKIAJL2KQ/20170213/us-north-1/iotdata/aws4_request,SignedHeaders=content-type;host;x-amz-content-sha256;x-amz-date,Signature=23ernrwewre0b564d3c8e9e1220d
15.	
16.	{state :{reported : {welcome: bar,UnitState {gateway_connected : True,device_connected : True} }}}

17.	Exception (3):
18.	epc1=0x4010011d epc2=0x00000000 epc3=0x00000000 excvaddr=0x4000ac30 depc=0x00000000
19.	
20.	ctx: cont 
21.	sp: 3fff0910 end: 3fff1180 offset: 01a0
22.	
23.	>>>stack>>>
24.	3fff0ab0:  3fff30e4 3fff1150 3fff30e4 401004d8  
25.	3fff0ac0:  3fff0108 0000020a 0000020a 4010020c  
26.	3fff0ad0:  3fff30e4 3fff1150 3ffeff74 4010068c  
27.	3fff0ae0:  5e971a6c 3f380128 3512fba9 40205ee0  
28.	3fff0af0:  3fff0108 000002c6 000002c6 40203a4d  
29.	3fff0b00:  3fff30e4 3fff1150 3ffeff4c 40203ab5  
30.	3fff0b10:  3fff1150 00000062 3ffeff4c 40205ecc  
31.	3fff0b20:  8b1e475c 37525ce9 3ffeff4c 402038ee  
32.	3fff0b30:  00826d5f 389df80e 44d94294 54e55798  
33.	3fff0b40:  52575d5c 5d751658 5c5c5c5c 0000000c  
34.	3fff0b50:  3fff0fa4 3fff4d14 3fff4ddc 40220698  
35.	3fff0b60:  e18041c5 0246163c eb708d9d ad1b931d  
36.	3fff0b70:  c08bdc52 6620746e 73696e69 ee646568  
37.	3fff0b80:  11d00a16 51c1e432 8858e18b 0e3fdc6f  
38.	3fff0b90:  a619b107 898ae240 5f1aa042 3fc75f32  
39.	3fff0ba0:  36d55538 36363636 36363636 36363636  
40.	3fff0bb0:  00000061 e56f3b44 36363636 36363636  
41.	3fff0bc0:  36363636 36363636 36363636 00000014  
42.	3fff0bd0:  3fff4df4 00000004 3fff0c44 40220635  
43.	3fff0be0:  060c6fcf 2f47b6f5 b1bf2b23 0af3b8a5  
44.	3fff0bf0:  b9099995 f132c06d f0097acf 44a36fb7  
45.	3fff0c00:  00000000 00000000 00000000 00000000  
46.	3fff0c10:  00000000 00000000 00000000 000068f0  
47.	3fff0c20:  6f6689d9 4b681ca0 cafabf26 3463668a  
48.	3fff0c30:  1e8ad020 2b2d7a3f 49181cf5 7c8471a0  
49.	3fff0c40:  8897e7cc 04e000aa 6cfd9d0d 79e9028d  
50.	3fff0c50:  060c6fcf 2f47b6f5 b1bf2b23 0af3b8a5  
51.	3fff0c60:  d4c21c40 7fa48239 90ef68f0 20c81400  
52.	3fff0c70:  000c4632 fa766cff 0a69638e 11c38000  
53.	3fff0c80:  00000000 00000000 00000000 00006970  
54.	3fff0c90:  6f7e05bd bf84c55f de287b3a 17d57e42  
55.	3fff0ca0:  f75363df 0289da07 0e6ca101 5bdae4a0  
56.	3fff0cb0:  cf25f40c b4bb1142 1f37f829 9266149d  
57.	3fff0cc0:  87d0e53a 3b7c96bc e6e96043 c71d5b6a  
58.	3fff0cd0:  8a1e386e f4079439 e2a3774d 1a49116c  
59.	3fff0ce0:  15d167bb 0119bead 7bd24844 7b5b9266  
60.	3fff0cf0:  cb634940 2677e866 544213d8 a2e1ac18  
61.	3fff0d00:  00bdf47d 76726573 66207265 73696e69  
62.	3fff0d10:  85646568 29a44c96 c00e1361 b508d6e2  
63.	3fff0d20:  e59dd6b2 8b1e475c 37525ce9 de6b1429  
64.	3fff0d30:  b726ddb7 fb826d5f 8f790b8a bae1923e  
65.	3fff0d40:  9476ba1e 0dab90a6 161a25c4 f3d5c4be  
66.	3fff0d50:  61cd8c08 caa14e6c 147234e6 b5e6c215  
67.	3fff0d60:  7ef420fd 80000000 00000000 00000000  
68.	3fff0d70:  00000000 00000000 00000000 00000000  
69.	3fff0d80:  00000000 00000000 00000000 000002a0  
70.	3fff0d90:  eb7f71dc fe8f18f2 d50c2d76 bd336793  
71.	3fff0da0:  00f6701f aa185aec 7a66cf27 01ece57e  
72.	3fff0db0:  82ce5660 09d3afab a9c19011 7ffa63e6  
73.	3fff0dc0:  124bbf68 07b390ba ddc7ba3b da508089  
74.	<<<stack<<<

How can i solve this problem

remesh123, 2017-02-14:
please help me

sabas1080, 2017-02-15:
Please put your sketch here to better understand the error

devyte, 2017-09-09:
Closing due to lack of requested sketch and lack of feedback in several months.

closed 2017-09-09 by devyte