esp8266/Arduino #2966: How to troubleshoot such an Exception?

Didix57, 2017-02-14:
Hi there

I am quite new in this world and I'm picking up quite good so far.
A lot of programs work fine so far but now I'm stuck with an application that throws an exception.

Module: NodeMCU 1.0 (ESP-12E); 80MHz; Flash Size: 4M (3M SPIFFS)

I am running a webserver and the ESP throws the exception after playing tone() and before printing 
"after delay".

partial Sketch:
```
    for (i=0; i<power; i++) {
      ffreq = multiBtn * multiHttp * freqbase * pow(scale, random(10));
      k++;
      Serial.print(k);
      Serial.print(".  frequency = ");
      Serial.print(ffreq);
      ifreq = (int)ffreq;
      Serial.print("   ifreq = ");
      Serial.print(ifreq);
      rnd = random(1, 11);
      Serial.print("   rnd = ");
      Serial.print(rnd);
      duration = durationbase * rnd;  // 80..800ms
      Serial.print("   Duration = ");
      Serial.print(duration);
      tone(piezoPinD5, ifreq, duration);
      Serial.print("  after tone()");
      delay(duration);
      Serial.println("  after delay()");
    }
```
This loop produces random tones() in a frequency range of about 25'000..60'000Hz and a length of 80..800ms.

Serial Output with the two last printed lines and the Exception output:
```
2257.  frequency = 44289.03   ifreq = 44289   rnd = 4   Duration = 320  after tone()  after delay()
2258.  frequency = 33275.00   ifreq = 33275   rnd = 2   Duration = 160  after tone()

Exception (0):
epc1=0x40201a58 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc40 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffde0:  40107440 3fffc200 00000022 00000030  
3ffffdf0:  3fffc200 40107440 00000020 40107488  
3ffffe00:  00000001 00000000 3fffc230 4000050c  
3ffffe10:  4000437d 00000030 00000016 ffffffff  
3ffffe20:  60000200 00000008 05011b13 80000000  
3ffffe30:  20000000 3fff22a8 80000000 203fc0c0  
3ffffe40:  80000000 00000000 00000001 3fff22ac  
3ffffe50:  00000234 003fc0c0 60000600 00000030  
3ffffe60:  00000009 00000000 0000002f 0000028c  
3ffffe70:  0000008c 00000000 00000001 3ffed930  
3ffffe80:  3ffee890 3cd9da66 60000600 00000030  
3ffffe90:  3ffee890 3cd9da66 60000600 00000030  
3ffffea0:  ffffffff 3fffc6fc 3ffee868 3ffef7b4  
3ffffeb0:  3ffee890 3cd9da66 60000600 00000030  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  00000011 40201010 3ffee868 00000001  
3ffffee0:  ffffffff 3ffe94c4 3ffee868 3fffdab0  
3ffffef0:  00000000 3fffdcb0 3ffee8a0 00000030  
3fffff00:  00000000 400042db 3ffe97e1 60000600  
3fffff10:  40004b31 3fff21ec 000002f4 003fc000  
3fffff20:  40105bc6 3ffee890 3ffed930 40107598  
3fffff30:  4020d009 3ffed930 3ffee890 3cd9da66  
3fffff40:  3fff21ec 00001000 4020d49e 00000008  
3fffff50:  00000000 00000000 4020d54b 3ffed9e4  
3fffff60:  3ffee890 1303f1c2 3ffee890 40201018  
3fffff70:  4021ec25 3ffef7b4 3ffee890 4021ec25  
3fffff80:  4021ec6a 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffee8a8 00000000 40000f65 3fffdab0  
3fffffa0:  40000f49 0000413f 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
ø
```

And the interpretation of 'Exception Decoder':
```
Exception 0: Illegal instruction
Decoding 23 results
0x40201a58: timer1_disable at C:\Users\schrotty\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_timer.c line 70
0x40107440: timer1_isr_handler at C:\Users\schrotty\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_timer.c line 33
0x40107440: timer1_isr_handler at C:\Users\schrotty\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_timer.c line 33
0x40107488: timer1_isr_handler at C:\Users\schrotty\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_timer.c line 40
0x40201010: delay_end at C:\Users\schrotty\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring.c line 38
0x40105bc6: spi_flash_read at ?? line ?
0x40107598: pvPortZalloc at C:\Users\schrotty\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/heap.c line 33
0x4020d009: pm_set_sleep_time at ?? line ?
0x4020d49e: pm_get_sleep_type at ?? line ?
0x4020d54b: pm_get_sleep_type at ?? line ?
0x40201018: delay_end at C:\Users\schrotty\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring.c line 40
0x4021ec25: ets_timer_handler_isr at ?? line ?
0x4021ec25: ets_timer_handler_isr at ?? line ?
0x4021ec6a: ets_timer_handler_isr at ?? line ?
```

Three questions arise to me:
1. any clue what the problem might be?
2. how to I have to interpret the Exception message?
   (I have read https://github.com/esp8266/Arduino/blob/master/doc/faq/a02-my-esp-crashes.md - but I'm stuck).
   Ok, I guess it is a timer problem. But what can I do from here?
3. what would you propose how such problems should be troubleshooted 
   (by bloody fresh nerds like me :-)?

Thanx for helping!

Didix57, 2017-02-15:
I have now isolated the routine to a simple sketch.
It still throws the same Exception after some time (sometimes after 200 or so lines, sometime after 2000).
But always the same Exception at the same place.

```
/*  *******************************************************************************
 *  tone-Random-iso.ino
 *  
 *  isolated tone-generation routine to test, whether Exceptions are thrown when
 *  using it isolated outside the webserver-software
 * 
 */
 
#define btnPin 2 // GPIO 2     D4 
#define ledPin 13  // GPIO 13    D7 
#define piezoPinD5 14 // GPIO 14    D5

bool buzzerOn = false;
bool interrupted = false;
const float freqbase = 25000.0;
const float scale = 1.1;  // percentage to raise frequency
const int power = 10;    // how many frequency steps
float multiBtn = 1.0; // multiply frequency if button pressed
float multiHttp = 1.0; // multiply frequency by WebBrowser

float ffreq;
int ifreq;
const int durationbase = 80;    // shortest duration to play a sound
int duration;

int k = 0;  // counter for number of sounds
// =============================================================================
void setup() {
  Serial.begin(115200);
  Serial.println("\n\n*******************************************************");
  Serial.println("*** tone-Random-iso  ***");
  
  pinMode(btnPin, INPUT_PULLUP); 
  pinMode(ledPin, OUTPUT); 
  pinMode(piezoPinD5, OUTPUT); 
}

// =============================================================================
void loop() {
  while (not interrupted) {
    int i;
    int rnd; // random number
    buzzerOn = true;
    
#ifdef DEBUG  
    for (i=0; i<3; i++){
      digitalWrite(ledPin, HIGH);  
      delay(100);                      
      digitalWrite(ledPin, LOW);  
      delay(100);                     
    }
  
    Serial.println(" ................................... test-beep ...");
    tone(piezoPinD5, 2500, 100);
    delay(100);
    tone(piezoPinD5, 3000, 100);
    delay(100);
  
    Serial.print("PushButton = ");
    Serial.println( digitalRead(btnPin) );
#endif
  
    Serial.println();
    ( digitalRead(btnPin) ? multiBtn = 1.0  : multiBtn = 0.1 );  // btn=0 -> ON
    
    for (i=0; i<power; i++) {
      ffreq = multiBtn * multiHttp * freqbase * pow(scale, random(10));
      k++;
      Serial.print(k);
      Serial.print(".  frequency = ");
      Serial.print(ffreq);
      ifreq = (int)ffreq;
      Serial.print("   ifreq = ");
      Serial.print(ifreq);
      rnd = random(1, 11);
      Serial.print("   rnd = ");
      Serial.print(rnd);
      duration = durationbase * rnd;  // 80..880ms
      Serial.print("   Duration = ");
      Serial.print(duration);
      tone(piezoPinD5, ifreq, duration);
      Serial.print("  after tone()");
      delay(duration + 10);
      Serial.println("  after delay()");      
    }
  }
}
```

JamesGKent, 2017-02-16:
can you try the following modification of your code:
```
/*  *******************************************************************************
 *  tone-Random-iso.ino
 *  
 *  isolated tone-generation routine to test, whether Exceptions are thrown when
 *  using it isolated outside the webserver-software
 * 
 */
 
#define btnPin 2 // GPIO 2     D4 
#define ledPin 13  // GPIO 13    D7 
#define piezoPinD5 14 // GPIO 14    D5

bool buzzerOn = false;
bool interrupted = false;
const float freqbase = 25000.0;
const float scale = 1.1;  // percentage to raise frequency
const int power = 10;    // how many frequency steps
float multiBtn = 1.0; // multiply frequency if button pressed
float multiHttp = 1.0; // multiply frequency by WebBrowser

float ffreq;
int ifreq;
const int durationbase = 80;    // shortest duration to play a sound
int duration;

int k = 0;  // counter for number of sounds
// =============================================================================
void setup() {
  Serial.begin(115200);
  Serial.println("\n\n*******************************************************");
  Serial.println("*** tone-Random-iso  ***");
  
  pinMode(btnPin, INPUT_PULLUP); 
  pinMode(ledPin, OUTPUT); 
  pinMode(piezoPinD5, OUTPUT); 
}

// =============================================================================
int i;
int rnd; // random number

void loop() {
	if (i==0){
		buzzerOn = true;
		Serial.println();
		( digitalRead(btnPin) ? multiBtn = 1.0  : multiBtn = 0.1 );  // btn=0 -> ON
	}
	ffreq = multiBtn * multiHttp * freqbase * pow(scale, random(10));
	k++;
	Serial.print(k);
	Serial.print(".  frequency = ");
	Serial.print(ffreq);
	ifreq = (int)ffreq;
	Serial.print("   ifreq = ");
	Serial.print(ifreq);
	rnd = random(1, 11);
	Serial.print("   rnd = ");
	Serial.print(rnd);
	duration = durationbase * rnd;  // 80..880ms
	Serial.print("   Duration = ");
	Serial.print(duration);
	tone(piezoPinD5, ifreq, duration);
	Serial.print("  after tone()");
	delay(duration + 10);
	Serial.println("  after delay()");      
	i++;
	if (i >= power)
		i == 0;
}
```

the change above means that instead of running the entire `for loop` inside the `loop` function, only one interation is done inside the loop allowing the background tasks to be carried out between each successive call to `loop` this should hopefully avoid any watchdog timeout issues, i don't have my ESP with me to be able to test it, but i believe that should fix your issue.

Didix57, 2017-02-16:
Thanx for responding, James!

copy-and-paste of your proposal compiles perfect, but it hangs with the same cause.

```
146.  frequency = 40262.75   ifreq = 40262   rnd = 10   Duration = 800  after tone()  after delay()
147.  frequency = 27500.00   ifreq = 27500   rnd = 3   Duration = 240  after tone()  after delay()
148.  frequency = 27500.00   ifreq = 27500   rnd = 4   Duration = 320  after tone()

Exception (0):
epc1=0x402017b8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc40 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffde0:  40107340 3fffc200 00000022 00000030  
3ffffdf0:  3fffc200 40107340 00000020 40107388  
3ffffe00:  3fffc200 40107340 3fffc230 4000050c  
3ffffe10:  40004308 00000030 00000016 ffffffff  
3ffffe20:  60000200 00000008 0001f500 80000000  
3ffffe30:  20000000 80000000 143fc2e0 80000000  
3ffffe40:  00000000 3fffc6fc 3ffe93c4 3fff095c  
3ffffe50:  00000014 003fc2e0 60000600 00000030  
3ffffe60:  000002f4 04024103 60000600 00000030  
3ffffe70:  0000000c 00002bc0 00000000 bfffffff  
3ffffe80:  ffffffff 3fffc6fc 3ffe8f3c 3ffe93c0  
3ffffe90:  3fffc200 40107340 00000020 40107388  
3ffffea0:  4000050c 3ffe8f3c 3fffc230 4000050c  
3ffffeb0:  40000f68 00000030 00000011 ffffffff  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  00000007 4021ce68 3ffe93c4 bfffffff  
3ffffee0:  ffffffff 3ffeff04 3ffe93c4 3fffdab0  
3ffffef0:  00000000 3fffdcb0 3ffe8ae8 00000030  
3fffff00:  00000000 400042db 00000064 60000600  
3fffff10:  40004b31 3fff067c 000002f4 003fc000  
3fffff20:  401024ce 3ffe8ad0 3ffe9270 4010758c  
3fffff30:  4021c935 3ffe9270 3ffe8ad0 04024103  
3fffff40:  3fff067c 00001000 4021cdce 00000008  
3fffff50:  4021ed50 3ffe93c4 4021ce7b 3ffe9324  
3fffff60:  3ffe8ad0 01409303 3ffe8ad0 60000600  
3fffff70:  40205431 3ffe9324 3ffe8ad0 0402201f  
3fffff80:  40205476 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffe8af0 00000000 40000f65 3fffdab0  
3fffffa0:  40000f49 00017fe6 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
þ
```
and after Exception Decoder 
```
Exception 0: Illegal instruction
Decoding 28 results
0x402017b8: timer1_disable at C:\Users\schrotty\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_timer.c line 70
0x40107340: timer1_isr_handler at C:\Users\schrotty\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_timer.c line 33
0x40107340: timer1_isr_handler at C:\Users\schrotty\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_timer.c line 33
0x40107388: timer1_isr_handler at C:\Users\schrotty\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_timer.c line 40
0x40107340: timer1_isr_handler at C:\Users\schrotty\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_timer.c line 33
0x40107340: timer1_isr_handler at C:\Users\schrotty\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_timer.c line 33
0x40107388: timer1_isr_handler at C:\Users\schrotty\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_timer.c line 40
0x4021ce68: pm_get_sleep_type at ?? line ?
0x401024ce: spi_flash_read at ?? line ?
0x4010758c: pvPortZalloc at C:\Users\schrotty\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/heap.c line 33
0x4021c935: pm_set_sleep_time at ?? line ?
0x4021cdce: pm_get_sleep_type at ?? line ?
0x4021ed50: pp_noise_test at ?? line ?
0x4021ce7b: pm_get_sleep_type at ?? line ?
0x40205431: ets_timer_handler_isr at ?? line ?
0x40205476: ets_timer_handler_isr at ?? line ?
```

igrr, 2017-02-16:
`timer1_disable` has been moved into IRAM in git version; are you running latest git version or one of the releases?

Didix57, 2017-02-16:
Hi igrr

Thanx for the hint!

I am a bloody beginner with Arduino & Co and have just downloaded (three weeks ago) the standard Arduino IDE 1.8.1.
Plus the ESP8266 addon from http://arduino.esp8266.com/stable/package_esp8266com_index.json
On my HDD this is 
C:\Users\schrotty\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0

On https://github.com/esp8266/Arduino/releases/tag/2.3.0 it says "latest release"

Is there a newer one and is there somewhere an instruction where (and how) how I get those "latest git version? 
I guess downloading and just copying over the old stuff?

Didix57, 2017-02-16:
I have to download it from here: https://github.com/esp8266/Arduino
Right?
And copy it over the old stuff?

Didix57, 2017-02-16:
Or does it replace
C:\Users\schrotty\AppData\Local\Arduino15\staging\packages\esp8266-2.3.0.zip

vicnevicne, 2017-02-16:
Please follow [these instructions](https://github.com/esp8266/Arduino#using-git-version) to install from git (you will first need to [install git](https://git-scm.com/downloads) and [python 2.7](https://www.python.org/ftp/python/2.7.13/python-2.7.13.msi) on your PC).

Didix57, 2017-02-16:
Thank you vicnevicne!

I have seen that, but I hoped that I can avoid installing git and python (two more programs on my overloaded windows machine)..
I hoped, it might be possible to download the zip and then extract the files to a specific directory.
However, I do not find something like ..hardware/esp8266com/.. on my computer.

Even if I would proceed as described on https://github.com/esp8266/Arduino#using-git-version by creating a directory
hardware/esp8266com/esp8266 
I fear that I would get a mess on my disk, since there is already ESP8266 stuff installed:
C:\Users\schrotty\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
C:\Users\schrotty\AppData\Local\Arduino15\staging\packages\esp8266-2.3.0.zip

I am stuck at this point ..

vicnevicne, 2017-02-16:
Yeah, installing from git can be a hassle for newcomers. My system is also a mess with several ESP versions at different places, but I have no idea what is the "right way" to do, particularly if you want to keep both the latest stable release and the git one. I am also longing for a new official release because the 2.3.0 is mostly out-of-date, but unfortunately no developer has time to prepare a 2.4.0 package for the time being...

Didix57, 2017-02-17:
veni, vidi, copi ...

Ok, I copied now the content of the zip file from https://github.com/esp8266/Arduino#using-git-version
to the place where I had found the exact same structure:
C:\Users\schrotty\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0

Now the compiler of IDE seems to work, but when uploading, it says
```
Sketch uses 248985 bytes (23%) of program storage space. Maximum is 1044464 bytes.
Global variables use 33524 bytes (40%) of dynamic memory, leaving 48396 bytes for local variables. Maximum is 81920 bytes.

java.io.IOException: Cannot run program "C:\Users\schrotty\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/esptool/esptool.exe": CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:207)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:709)
	at processing.app.SketchController.exportApplet(SketchController.java:682)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2190)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 8 more
An error occurred while uploading the sketch
```
However, the only instance of esptool.exe is at 
C:\Users\schrotty\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9\esptool.exe

So this is not working that easy ...
Reversing everything (YES: veni vidi BACKUP FIRST :-) and everything works fine again.
Besides the fact, that the program still throws Exceptions ...

Didix57, 2017-02-17:
We shall never surrender ...

Everything again as described above.
Now I copied the existing esptool.exe to the expected path.
I actually doubted that it will work because of the mix of slashes and backslashes.
But somewhere I remembered, that many years ago I did something similar that worked.

Well then, it compiles, uploads and is running now. 
Let's hope it doesn't throw anymore Exceptions ...

Didix57, 2017-02-17:
Looks good!
No Exception after 12'000 lines yet!!

Didix57, 2017-02-17:
80'000 lines and 11 hours of continuos running -- Problem solved!

Thank you guys for supporting !!!

closed 2017-09-09 by devyte