esp8266/Arduino #2967: Heap is running out

OrakelKSL, 2017-02-14:
### Basic Infos

#### Hardware
Hardware:		ESP-12 (NodeMCU V2)
Core Version:      	2.3.0

### Description

As you can see in serial monitor: If you are tilting your mobile heap is running out quickly.

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:			?qio?
Flash Frequency:	40Mhz
Upload Using:		OTA / SERIAL
Reset Method:		ck / nodemcu

### Sketch
<pre>
<font color="#95a5a6">/*--------------------------------------------------------------------------------*</font>
<font color="#95a5a6"> &nbsp;&nbsp;Arduino WiFi-Car</font>
<font color="#95a5a6"> &nbsp;&nbsp;&nbsp;Manfred Detterbeck February 2017</font>
<font color="#95a5a6"></font>
<font color="#95a5a6"> &nbsp;&nbsp;&nbsp;Control of WiFi-Car with your smartphone via browser by</font>
<font color="#95a5a6"> &nbsp;&nbsp;&nbsp;using the orientation of the phone.</font>
<font color="#95a5a6"> &nbsp;&nbsp;&nbsp;Spec: </font><u><font color="#95a5a6">http://w3c.github.io/deviceorientation/spec-source-orientation.html</font></u><font color="#95a5a6"></font>
<font color="#95a5a6"> &nbsp;&nbsp;&nbsp;Based on: </font><u><font color="#95a5a6">https://github.com/squix78/esp8266-projects/tree/master/arduino-ide/wifi-car</font></u><font color="#95a5a6"></font>
<font color="#95a5a6"> &nbsp;&nbsp;&nbsp;Java-script now inline!</font>
<font color="#95a5a6"> &nbsp;&nbsp;&nbsp;</font>
<font color="#95a5a6"> &nbsp;&nbsp;&nbsp;Hardware: NodeMVU V2 (caution! V3 doesn't fit on most Motor-shields)</font>
<font color="#95a5a6"> &nbsp;&nbsp;&nbsp;L293D Motor shield</font>
<font color="#95a5a6"> &nbsp;&nbsp;&nbsp;Together: </font><u><font color="#95a5a6">https://de.aliexpress.com/wholesale?&SearchText=motor+nodemcu+kit</font></u><font color="#95a5a6"></font>
<font color="#95a5a6"> &nbsp;&nbsp;&nbsp;2 DC Gear-Motor (1 Left, 2 Right)</font>
<font color="#95a5a6"> &nbsp;--------------------------------------------------------------------------------</font>
<font color="#95a5a6"> &nbsp;&nbsp;"THE BEER-WARE LICENSE" (Revision 42):</font>
<font color="#95a5a6"> &nbsp;&nbsp;&nbsp;&lt;Manfred.Detterbeck5@freenet.de&gt; wrote this file. As long as you retain this</font>
<font color="#95a5a6"> &nbsp;&nbsp;&nbsp;notice you can do whatever you want with this stuff. If we meet some day, and</font>
<font color="#95a5a6"> &nbsp;&nbsp;&nbsp;you think this stuff is worth it, you can buy me a beer in return</font>
<font color="#95a5a6"> &nbsp;&nbsp;&nbsp;Manfred Detterbeck</font>
<font color="#95a5a6"> &nbsp;--------------------------------------------------------------------------------</font>
<font color="#95a5a6"> &nbsp;&nbsp;You need he following libraries:</font>
<font color="#95a5a6"></font>
<font color="#95a5a6"> &nbsp;&nbsp;WiFiManager ESP8266 WiFi Connection manager with fallback web configuration</font>
<font color="#95a5a6"> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;WiFiManager by tzapu Version 0.12.0</font>
<font color="#95a5a6"> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;</font><u><font color="#95a5a6">https://github.com/tzapu/WiFiManager</font></u><font color="#95a5a6"></font>
<font color="#95a5a6"></font>
<font color="#95a5a6"> &nbsp;&nbsp;-------------------------------------------------------------------------------*/</font>

<font color="#434f54">// IO **** ESP8266 ****</font>
<font color="#434f54">// &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;IO-Number ESP8266-12E &nbsp;Aufdruck &nbsp;NodeMCU</font>
<font color="#434f54">// &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;1 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Rst &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;RST &nbsp;&nbsp;&nbsp;&nbsp;Reset-SW: momentary switch NO to GND &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;/ Pullup 10k to 3V3</font>
<font color="#434f54">// &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;2 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;ADC &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;AD0 &nbsp;&nbsp;&nbsp;&nbsp;0..1V originqal, over Divider of Nodemcu 0..3,2V</font>
<font color="#434f54">// &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;3 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;EN &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;EN &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Enable. &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;/ Normally connect to 3V3.</font>
<font color="#5e6d03">#define</font> <font color="#000000">GPIO16</font> &nbsp;&nbsp;&nbsp;<font color="#000000">16</font> &nbsp;<font color="#434f54">// &nbsp;&nbsp;&nbsp;16 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;4 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;GPIO16 &nbsp;&nbsp;&nbsp;D0 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;(LED on some Node-MCU. Connect to RST for deep-sleep)</font>
<font color="#5e6d03">#define</font> <font color="#000000">GPIO14</font> &nbsp;&nbsp;&nbsp;<font color="#000000">14</font> &nbsp;<font color="#434f54">// &nbsp;&nbsp;&nbsp;14 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;5 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;GPIO14 &nbsp;&nbsp;&nbsp;D5 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;SPI SCK</font>
<font color="#5e6d03">#define</font> <font color="#000000">GPIO12</font> &nbsp;&nbsp;&nbsp;<font color="#000000">12</font> &nbsp;<font color="#434f54">// &nbsp;&nbsp;&nbsp;12 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;6 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;GPIO12 &nbsp;&nbsp;&nbsp;D6 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;SPI MISO &nbsp;/ (witty cloud: LED green)</font>
<font color="#5e6d03">#define</font> <font color="#000000">GPIO13</font> &nbsp;&nbsp;&nbsp;<font color="#000000">13</font> &nbsp;<font color="#434f54">// &nbsp;&nbsp;&nbsp;13 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;7 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;GPIO13 &nbsp;&nbsp;&nbsp;D7 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;SPI MOSI &nbsp;/ (witty cloud: LED blue)</font>
<font color="#434f54">// &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;8 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;VCC &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;3V3 &nbsp;&nbsp;&nbsp;&nbsp;3,3V regulated source, ca. 200mA</font>
<font color="#434f54">// &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;9 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;CS0 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;CMD &nbsp;&nbsp;&nbsp;&nbsp;Not recommended for user. Caution! Internal connection to Flash. Not available for user on all boards</font>
<font color="#434f54">// &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;10 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;MISO &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;SD0 &nbsp;&nbsp;&nbsp;&nbsp;Not recommended for user. Caution! Internal connection to Flash. Not available for user on all boards</font>
<font color="#5e6d03">#define</font> <font color="#000000">GPIO09</font> &nbsp;&nbsp;&nbsp;<font color="#000000">9</font> &nbsp;&nbsp;<font color="#434f54">// &nbsp;&nbsp;&nbsp;9 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;11 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;GPIO09 &nbsp;&nbsp;&nbsp;SD2 &nbsp;&nbsp;&nbsp;&nbsp;Caution! Only with modification of internal Flash-connections! Not recommended for user. Not available for user on all boards</font>
<font color="#5e6d03">#define</font> <font color="#000000">GPIO10</font> &nbsp;&nbsp;&nbsp;<font color="#000000">10</font> &nbsp;<font color="#434f54">// &nbsp;&nbsp;&nbsp;10 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;12 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;GPIO10 &nbsp;&nbsp;&nbsp [... cut ...] (</font><font color="#000000">)</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

</pre>

### Debug Messages

Heap started with 40192 and runs down to:

[2JFree Heap Size:	272 Bytes
GET /engines/285,371 HTTP/1.1
[285][371]
Zykluszeit: 216

Exception (28):
epc1=0x4020498f epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000008 depc=0x00000000

ctx: sys 
sp: 3ffff990 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffb30:  0000000c 00000000 3fff2194 40204985

OrakelKSL, 2017-02-15:
I made some investigations:
Every time client receives a request, the heap goes down and does NOT recover!
You can easily test it: Connect with any browser and keep F5 pressed: Heap goes rapidly down to a few bytes and does never recover.

Check the following very simple sketch:

<pre>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#d35400">ESP8266WiFi</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#00979c">const</font> <font color="#00979c">char</font><font color="#434f54">*</font> <font color="#000000">ssid</font> <font color="#434f54">=</font> <font color="#000000">S_SSID</font><font color="#000000">;</font>
<font color="#00979c">const</font> <font color="#00979c">char</font><font color="#434f54">*</font> <font color="#000000">password</font> <font color="#434f54">=</font> <font color="#000000">S_PASSWORD</font><font color="#000000">;</font>

<font color="#434f54">// Create an instance of the server</font>
<font color="#434f54">// specify the port to listen on as an argument</font>
<font color="#d35400">WiFiServer</font> <font color="#000000">server</font><font color="#000000">(</font><font color="#000000">80</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#00979c">void</font> <font color="#5e6d03">setup</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">begin</font><font color="#000000">(</font><font color="#000000">115200</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;
 &nbsp;<font color="#434f54">// Connect to WiFi network</font>
 &nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">print</font><font color="#000000">(</font><font color="#005c5f">"Connecting to "</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#000000">ssid</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;
 &nbsp;<b><font color="#d35400">WiFi</font></b><font color="#434f54">.</font><font color="#d35400">begin</font><font color="#000000">(</font><font color="#000000">ssid</font><font color="#434f54">,</font> <font color="#000000">password</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;
 &nbsp;<font color="#5e6d03">while</font> <font color="#000000">(</font><b><font color="#d35400">WiFi</font></b><font color="#434f54">.</font><font color="#d35400">status</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#434f54">!=</font> <font color="#000000">WL_CONNECTED</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;&nbsp;&nbsp;<font color="#d35400">delay</font><font color="#000000">(</font><font color="#000000">500</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">print</font><font color="#000000">(</font><font color="#005c5f">"."</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">}</font>
 &nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#005c5f">""</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#005c5f">"WiFi connected"</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;
 &nbsp;<font color="#434f54">// Start the server</font>
 &nbsp;<font color="#000000">server</font><font color="#434f54">.</font><font color="#d35400">begin</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#005c5f">"Server started"</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">// Print the IP address</font>
 &nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><b><font color="#d35400">WiFi</font></b><font color="#434f54">.</font><font color="#d35400">localIP</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">loop</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;&nbsp;&nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">write</font><font color="#000000">(</font><font color="#000000">27</font><font color="#000000">)</font><font color="#000000">;</font> &nbsp;&nb [... cut ...] t color="#000000">;</font>
 &nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#000000">req</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#d35400">client</font><font color="#434f54">.</font><font color="#d35400">flush</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

</pre>

OrakelKSL, 2017-02-15:
...ir recovers, but it takes MINUTES!

OrakelKSL, 2017-02-15:
... and if you keep f5 pressed exception 28 or 29 occours. 
Has anybody suggestions to force "free the heap" at end of loop?
:-(

„Nihil est in ESP, quod non fuerit in intellectu" - Nichts ist im ESP, was zuvor nicht im Verstand war.

devyte, 2017-09-09:
@OrakelKSL is this issu still valid? 
The fact heap would recover hints at sockets being kept in TIME_WAIT state, probably for a the typical 2 minutes. That is per RFC (I forget which), although for tiny embedded systems implementations usually don't hold to it.
Still, with latest git I don't currently see the issue when refreshing a webpage.

devyte, 2017-10-10:
Old issue and no feedback in over a month => closing.

closed 2017-10-10 by devyte

efeuentertainment, 2017-11-04:
there is a workaround for running out of heap due to sockets kept in TIME_WAIT.
[issue #1923](https://github.com/esp8266/Arduino/issues/1923)