esp8266/Arduino #2973: Missed UART Rx symbols while transmitting TCP

akouz, 2017-02-16:
### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	esp8266 by ESP8266 Community ver 2.3.0
	
### Description

While ESP sends TCP messages, its UART misses Rx chars. 

An external microcontroller sends a sesries of messages to ESP8266 via UART.  Each message covered by CRC. ESP8266 checks CRC and then sends the valid messages to PC via LAN or Internet by TCP. Occasionally messages received by ESP are corrupted. It looks like it is related to TCP connection, Messages are never corrupted when TCP is not used. 

Sorry, my sketch is very big. Also other software is involved.  It might take a while until I can prepare something to reproduce the issue. For time being I hope that description above is sufficient to locate the problem.  I think UART Rx interrupt is disabled while TCP messages are being sent.

vlast3k, 2017-02-16:
have you tried sending the messages in small chunks, e.g. 50 bytes, waiting for some milliseconds and then sending the next chunk?

akouz, 2017-02-16:
I have messages of variable length, from 8 to 300 bytes. The rate of messages varies too. 

I have implemented an ackowledge/retry mechanism. Each message I sent to ESP via UART must be confirmed by a short acknowledge from ESP.  If there is no acknowledge then after time-out the same message is repeated again and again, up to 5 times. Time-out starts from about 100 ms and then increased after each retry.  

Sometimes it takes almost one second to push a message to ESP.

akouz, 2017-05-10:
I ended up using a workaround. One of ESP8266 pins is used as RTS output signal to microcontroller's CTS input of UART. My sketch sets RTS=1 before writing anything to TCP, causing microcontroller UART to freeze and do not send any data to ESP8266. When TCP procedure finished, sketch resumes communication by releasing RTS=0. 

For example, to publish a message to a MQTT server:

          digitalWrite(RTS, 1);                                 
          Cloud_client.publish(topic.c_str(), rply->buf, rply->len, false); 
          digitalWrite(RTS, 0);                  

Sometimes I saw RTS pulses up to 1 sec long. However, this fix the problem, now my sketch receives UART messages with no corruption.

devyte, 2017-09-09:
@akouz is this still valid? If so, please post a reduced sketch. Without it, this will be closed.
Also, your described symptoms are typical of a weak power supply. The ESP spikes power usage during wifi transmission, which could cause corruption in serial comm. I have seen serial corruption due to this. It is one of the reasons that I use a 5V 20A power supply for development. Any corruption or resets during developent are assured to not be due to a weak power supply.

akouz, 2017-09-10:
@devyte, yes, it is valid. It might take a while to make a sketch to demonstrate it because an external microcontroller involved. I think I should make a dedicated sketch and  a  PC program to show the problem. 

Device is powered from USB by a 0.7A regulator XC6222-33. There is 1000 uF cap on the 3.3V rail.

devyte, 2017-10-27:
@akouz please don't forget to make a dedicated sketch that shows the issue. Also, please make sure to test with latest git, or at least rc2.

devyte, 2018-02-28:
@akouz Does latest git + pr #4327 fix this?

devyte, 2018-07-12:
lwip glue and Serial has undergone major changes in the meantime.
No feedback in several months.
Closing

closed 2018-07-12 by devyte