esp8266/Arduino #2975: crach on websocket & UpdateServer service

mkeyno, 2017-02-17:
### Basic Infos

#### Hardware
Hardware:			?ESP-12F
Core Version:      	git version

### Description

when I test all examples of [universal arduino telegram bot](https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/tree/master/examples/ESP8266) it work fine even without time syncing at setup , I add  WebServer  service and still its work fine but when I add @Links2004  websocket service (just create instance ) & UpdateServer its falling to crack as soon as it tried to get message
following is the crash decode  and  sketch 

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

 /*******************************************************************
 *  An example of how to use a custom reply keyboard markup.       *
 *                                                                 *
 *                                                                 *
 *  written by Brian Lough       
botname   smart_iot
bot username  @mkeyno_bot                                  *
 *******************************************************************/
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <UniversalTelegramBot.h>
#include <FS.h>
#include <Hash.h>
#include <ESP8266WiFi.h>
#include <time.h>
#include <ESP8266mDNS.h>
#include <ArduinoOTA.h>
#include <StreamString.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
          ESP8266WebServer server(80);
 #include <WebSocketsServer.h> 
         //  WebSocketsServer webSocket = WebSocketsServer(81);
/*     
if uncommented it'll crashed 
*/
#include <EEPROM.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266HTTPUpdateServer.h>
           ESP8266HTTPUpdateServer httpUpdater;
#include <ESP8266httpUpdate.h>
                  
// Initialize Wifi connection to the router
char ssid[] = "zzzz";     // your network SSID (name)
char password[] = "zzzzzzzzz"; // your network key

// Initialize Telegram BOT
#define BOTtoken "298704009:AAHnCd7AAdmzDG6RpwVaAB-VgLe5iqftW0s"  // your Bot Token (Get from Botfather)

WiFiClientSecure client;
UniversalTelegramBot bot(BOTtoken, client);

int Bot_mtbs = 1000; //mean time between scan messages
long Bot_lasttime;   //last time messages' scan has been done

[... 145 lines cut ...]
0x4020e184: __yield at D:\arduino-1.6.9\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 56
0x4023afa6: do_handshake at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1545
:  (inlined by) basic_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1421
0x40201795: _malloc_r at D:\arduino-1.6.9\hardware\esp8266com\esp8266\cores\esp8266/heap.c line 14
0x4023b2ef: do_client_connect at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 171
0x4023b10c: ssl_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 271
0x40208874: SSLContext::connect(ClientContext*, char const*, unsigned int) at D:\arduino-1.6.9\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 587
:  (inlined by) WiFiClientSecure::_connectSSL(char const*) at D:\arduino-1.6.9\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 317
0x40207c2a: WiFiClient::connect(IPAddress, unsigned short) at D:\arduino-1.6.9\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/WiFiClient.cpp line 349
0x40208a35: WiFiClientSecure::connect(char const*, unsigned short) at D:\arduino-1.6.9\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 306
0x402096a5: UniversalTelegramBot::sendPostToTelegram(String, ArduinoJson::JsonObject&) at D:\arduino-1.6.9\hardware\esp8266com\esp8266\cores\esp8266/WString.h line 47
0x4020d987: String::reserve(unsigned int) at D:\arduino-1.6.9\hardware\esp8266com\esp8266\cores\esp8266/WString.cpp line 720
0x4020d9b9: String::copy(char const*, unsigned int) at D:\arduino-1.6.9\hardware\esp8266com\esp8266\cores\esp8266/WString.cpp line 720
0x4020dac0: String::operator=(String const&) at D:\arduino-1.6.9\hardware\esp8266com\esp8266\cores\esp8266/WString.cpp line 720
0x40209948: UniversalTelegramBot::sendPostMessage(ArduinoJson::JsonObject&) at D:\arduino-1.6.9\hardware\esp8266com\esp8266\cores\esp8266/WString.h line 47
0x4020f268: ArduinoJson::Internals::JsonBufferAllocated::operator new(unsigned int, ArduinoJson::JsonBuffer*) at D:\arduino-1.6.9\libraries\ArduinoJson/include/ArduinoJson/Internals/JsonBufferAllocated.hpp line 20
0x40209b9d: UniversalTelegramBot::sendMessage(String, String, String) at D:\arduino-1.6.9\hardware\esp8266com\esp8266\cores\esp8266/WString.h line 47
0x4020d9b9: String::copy(char const*, unsigned int) at D:\arduino-1.6.9\hardware\esp8266com\esp8266\cores\esp8266/WString.cpp line 720
0x4020da06: String::String(char const*) at D:\arduino-1.6.9\hardware\esp8266com\esp8266\cores\esp8266/WString.cpp line 720
0x40206f68: handleNewMessages(int) at C:\Users\Private\AppData\Local\Temp\arduino_modified_sketch_695968/ReplyKeyboardMarkup.ino line 101
0x40207175: loop at C:\Users\Private\AppData\Local\Temp\arduino_modified_sketch_695968/ReplyKeyboardMarkup.ino line 153
0x4020e158: loop_wrapper at D:\arduino-1.6.9\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at D:\arduino-1.6.9\hardware\esp8266com\esp8266\cores\esp8266/cont.S line 109

```

devyte, 2017-09-09:
@mkeyno is this still valid with latest git?

devyte, 2017-10-10:
No reply in over a month, closing.

closed 2017-10-10 by devyte