esp8266/Arduino #2977: flashing prepared filesystem with esptool.py

kommancs96, 2017-02-17:
I have the following problem:
Given a larger (~100kb) png image that i want to serve with DNSServer. It is too large to embed in html as base64, so I have to use SPIFFS, to store the image. I've run into these problems:

1. How to prepare a file system, that contains my image on my laptop (MacBook)?
2. How to flash the filesystem next to the sketch code, to use it, using esptool.py?
3. Do I need to flash 2 different hex files to different addresses?

NodeMCU board, 4M flash size

vicnevicne, 2017-02-18:
I guess all those questions are answered [here :-)](https://github.com/esp8266/Arduino/blob/master/doc/filesystem.md#uploading-files-to-file-system)

kommancs96, 2017-02-18:
@vicnevicne But how do I find out the correct address to flash the filesystem?
After I make the filesystem with `mkspiffs -c fs fs.espfs -s 262144` , I use esptool to flash it to my board: 
`esptool.py --baud 115200 --port /dev/cu.SLAB_USBtoUART write_flash 0xeb000 ./fs.espfs`
How do i know that `0xeb000` is correct? It seems that it isn't, because `SPIFFS.begin()` returns `false`.

igrr, 2017-02-19:
You can find the values of page, block, and filesystem size in boards.txt. Look for the NodeMCU board, and choose the same filesystem size as you are selecting in the IDE (either 1M or 3M).
Address to upload the filesystem will also be listed there:

```
nodemcuv2.menu.FlashSize.4M3M=4M (3M SPIFFS)
nodemcuv2.menu.FlashSize.4M3M.build.flash_size=4M
nodemcuv2.menu.FlashSize.4M3M.build.flash_ld=eagle.flash.4m.ld
nodemcuv2.menu.FlashSize.4M3M.build.spiffs_start=0x100000
nodemcuv2.menu.FlashSize.4M3M.build.spiffs_end=0x3FB000
nodemcuv2.menu.FlashSize.4M3M.build.spiffs_blocksize=8192
nodemcuv2.menu.FlashSize.4M3M.build.spiffs_pagesize=256

nodemcuv2.menu.FlashSize.4M1M=4M (1M SPIFFS)
nodemcuv2.menu.FlashSize.4M1M.build.flash_size=4M
nodemcuv2.menu.FlashSize.4M1M.build.flash_ld=eagle.flash.4m1m.ld
nodemcuv2.menu.FlashSize.4M1M.build.spiffs_start=0x300000
nodemcuv2.menu.FlashSize.4M1M.build.spiffs_end=0x3FB000
nodemcuv2.menu.FlashSize.4M1M.build.spiffs_blocksize=8192
nodemcuv2.menu.FlashSize.4M1M.build.spiffs_pagesize=256
```
The size is equal to `spiffs_end - spiffs_start`, upload address is `spiffs_start`, and block size and page size are as given.

kommancs96, 2017-02-19:
So now I make the filesystem like this:
 `rm -rf ./fs.espfs && mkspiffs -c fs fs.espfs -b 8192 -p 256 -s 0x2fb000`
And then I try to flash it to may board using: 
`esptool.py --baud 115200 --port /dev/cu.SLAB_USBtoUART write_flash 0x300000 ./fs.espfs`

But I get the following output:
```
Connecting...
Erasing flash...
Traceback (most recent call last):
  File "/usr/local/bin/esptool.py", line 5, in <module>
    pkg_resources.run_script('esptool==0.1.0', 'esptool.py')
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 742, in r
un_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 1674, in
run_script
    exec(script_code, namespace, namespace)
  File "/Library/Python/2.7/site-packages/esptool-0.1.0-py2.7.egg/EGG-INFO/scripts/esptool.py", line 586, in <module>

  File "/Library/Python/2.7/site-packages/esptool-0.1.0-py2.7.egg/EGG-INFO/scripts/esptool.py", line 214, in flash_begin

Exception: Failed to enter Flash download mode
```

kommancs96, 2017-02-19:
Also tried an 1MB fs like this:
`rm -rf ./fs.espfs && mkspiffs -c fs fs.espfs -b 8192 -p 256 -s 0xfb000`
`esptool.py --baud 115200 --port /dev/cu.SLAB_USBtoUART write_flash 0x100000 ./fs.espfs`
This resulted no error while flashing but then `SPIFFS.begin()` is still returning `false`.
I have [this board](http://www.ebay.com/itm/NodeMcu-Lua-WIFI-Internet-Things-development-board-based-ESP8266-CP2102-module-/262136737828) , ESP12-E or ESP12-F.

igrr, 2017-05-15:
1MB or 3MB choice has to match the FS size selected when programming the sketch in Arduino IDE.

Also, you seem to have a very old version of esptool.py. Please get one of the more recent releases at https://github.com/espressif/esptool.

devyte, 2017-09-09:
@kommancs96 There is a tool available (i.e.: plugin) for the Arduino IDE to upload "sketch data" into the ESP. It takes the contents of a subdir named "data", makes a SPIFFS image out of it based on selected params, and flashes it to the ESP by calling esptool under the hood.
I can vouch for it functioning correctly under windows and linux in the latest git, I use it in my development.
Closing this as no longer relevant.

closed 2017-09-09 by devyte

CzokNorris, 2022-08-25:
It is still relevant, because some of us want to use this in automated flashing, where using an IDE is not feasible. Is there a pure command line way of doing this?

FxS19, 2023-04-13:
I had a similar Problem with littleFS. After some research there are some steps how to get the right parameters for the selected filesystem type:

1. Create the desired filesystem with the ESP itself (My way of doing this was creating a file and appending some lines)
    ```cpp
    void appendFile(){
      File file = LittleFS.open("test.txt", "a");
      if(!file){
        Serial.println("Failed to open file for appending");
        return;
      }
      file.write("Test Data\n");
  
      file.close();
    }

    void readFile(){
      File file = LittleFS.open("test.txt", "r");
      if(!file){
        Serial.println("Failed to open file for reading");
        return;
      }
  
      Serial.println("File Content:");
      while(file.available()){
        Serial.write(file.read());
      }
      file.close();
   }
   ```
   After that ist was obvious, that the filesystem was working (every reboot a new line containing "Test Data" is added)
2. Try to download the filesystem with the size you assume it is
    ```bash
    esptool.py --baud 921600 read_flash 0x200000  2097152 out.little
    ```
4. Try to Read its contents (mkspiffs has similar functionality)
    ```bash
    mklittlefs -p 256 -b 4096 -l out.little
    ```
    This resulted in an error: `Invalid block count (253 != 256)`  
    So the assumption that the "partition"  is  2097152 bytes is wrong by 3 blocks (253 - 256).  
5. Alter the size to `2072576`
    ```bash
    esptool.py --baud 921600 read_flash 0x200000  2072576 out.little
    ```
6. Try to read the partition again succeded
    ```bash
    mklittlefs -p 256 -b 4096 -l out.little
    ```
    50      /test.txt       Thu Jan  1 00:00:00 1970
    Creation time:  Thu Jan  1 00:00:00 1970
    ```
7. Create the filesystem with the size of 2072576