esp8266/Arduino #2978: WiFiClientSecure verify can't match Cert Name (FIX provided)

J6B, 2017-02-17:
During my HTTPS client tests, I found a bug into verify (now into _verifyDN) function of WiFiClientSecure.cpp

I have a webserver with a certificate with a CN that contains capital letters like : sub.TOTO.com
(from a private PKI it's easy to get one like this)

The problem is that domain_name is convert into String and then lowered.
But common_name got from certificate is not lowered so they will never match.

FIX : 

bool WiFiClientSecure::_verifyDN(const char* domain_name)
{
    DEBUGV("domain name: '%s'\r\n", (domain_name)?domain_name:"(null)");
    String domain_name_str(domain_name);
    domain_name_str.toLowerCase();

    const char* san = NULL;
    int i = 0;
    while ((san = ssl_get_cert_subject_alt_dnsname(*_ssl, i)) != NULL) {
        if (matchName(String(san), domain_name_str)) {
            return true;
        }
        DEBUGV("SAN %d: '%s', no match\r\n", i, san);
        ++i;
    }
    const char* common_name = ssl_get_cert_dn(*_ssl, SSL_X509_CERT_COMMON_NAME);
    String common_name_str(common_name);
    common_name_str.toLowerCase();
    if (common_name && matchName(common_name_str, domain_name_str)) {
        return true;
    }
    DEBUGV("CN: '%s', no match\r\n", (common_name)?common_name:"(null)");

    return false;
}

dandandandandandandan, 2017-02-21:
I tested this fix with previously working servers and it still works with them. (Since most of the SSL ciphers are not supported by the ESP8266 it is not that easy to find a server for testing, i could connect to twitter.com, api.github.com and paypal.com but not much else)

igrr, 2017-05-22:
PR #3271 includes this fix.

commit 2fcc1b32 (2017-05-21): WiFiClientSecure: match CN and SANs ignoring case

Some websites have certificates with uppercase letters in CN. This change
makes _verifyDN function accept such certificates by converting all names
to lower case before comparing them.

Resolves #2978
commit f6d232f1 (2017-05-22): WiFiClientSecure: match CN and SANs ignoring case

Some websites have certificates with uppercase letters in CN. This change
makes _verifyDN function accept such certificates by converting all names
to lower case before comparing them.

Resolves #2978
closed 2018-01-02 by igrr