esp8266/Arduino #2979: HTTP Post Request - ESP8266HTTPClient generates wrong http headers

schlaubstar, 2017-02-18:
Hi when using this simple code:

    HTTPClient http;
    string message="Hello from ESP8266";
    http.begin("http://192.168.178.21:8000");
    http.addHeader("Content-Type", "text/plain");
    int httpCode = http.POST(message);
    if(httpCode > 0) {
      Serial.printf("[HTTP] POST... code: %d\n", httpCode);
      if(httpCode == HTTP_CODE_OK) {
                String payload = http.getString();
                Serial.println(payload);
            }
        } else {
            Serial.printf("[HTTP] POST... failed, error: %s\n", http.errorToString(httpCode).c_str());
        }
    http.end();

I always received the message: "error uploading payload" (like many others considering google and this issue list). So I did setup a server on my computer, to check about whats going on.

The matching message looks like this:
POST  HTTP/1.1
Host: 192.168.178.34:8000
User-Agent: ESP8266HTTPClient
Connection: close
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Content-Type: text/plain
Content-Length: 20

Hello from ESP8266!

The problem is caused by the wrong post header: If you manually generate the post message then it is working and everything looks the same except for the first line:
POST / HTTP/1.1

I assume, that HttpClient.Post(message) does generate a wrong header lacking one slash.
Maybe someone wants to fix this.

igrr, 2017-02-18:
It looks like the URL you specify is lacking the trailing slash (after the port number). I suppose the library could be smart enough to handle this case and automatically add the slash, but it isn't :)

schlaubstar, 2017-02-18:
Damn ! You're a genious ! I'm so sorry > That issue can be closed. Obviously I just needed to add a trailing "/" to the host .... But maybe it is of help to others.

closed 2017-02-18 by schlaubstar