esp8266/Arduino #2984: Issue with running sketch

MrAda, 2017-02-19:
### Basic Infos

#### Hardware
Hardware:		NODEMCU LoLin V3 ESP8266
Core Version:      	?

### Description

I have noticed that before my program would work and now it seems to do nothing, like it crashes but it doesn't say that it has crashed in the serial monitor.  Here is the sizing information.  Maybe I have too  much of a program.  I hope not as I need to have every bit of what I am using.

Sketch uses 230485 bytes (22%) of program storage space. Maximum is 1044464 bytes.
Global variables use 32604 bytes (39%) of dynamic memory, leaving 49316 bytes for local variables. Maximum is 81920 bytes.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:	40Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

igrr, 2017-02-19:
Hi Chris, the memory usage seems to be fine.

Is there any other information you can provide? It's pretty hard to help you without knowing the sketch, the steps to reproduce, etc.

MrAda, 2017-02-19:
What I had done was incorporate an NTP UDP socket program into my WIFI/Relay based app.  I went back to the NTP demo I was playing with and my ESP works again (connects to WIFI and gets the NTP time).  I will try and back off those changes to my original app to see if I can get at least the WIFI talking. What I was seeing in the Monitor was the gibberish and that was all.  With the NTP demo I am seeing the gibberish and then my "Connecting to SSID"

MrAda, 2017-02-19:
I figured out the problem!  I had this:
  Serial.begin(115200);

  while (!Serial.available()) {
    ;
  }

I saw this in another INO file and thought it was a good idea.  As soon as I removed it the serial started working and I guess I was blocking myself!  Doh!

closed 2017-02-19 by MrAda