esp8266/Arduino #2987: Local\Temp\arduino_build_92969/arduino.ar'; reason: File exists

ncisrobert, 2017-02-21:
I just got a new windows 1064 bit laptop and when trying to verify a blank esp sketch i get this error everytime 

C:\Users\Robert\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar: unable to rename 'C:\Users\Robert\AppData\Local\Temp\arduino_build_92969/arduino.ar'; reason: File exists

it only happens when i try and esp board any ideas

lrmoreno007, 2017-02-22:
Try running as administrator.

ncisrobert, 2017-02-22:
Iv tried running as an admin but still no luck

devyte, 2017-09-09:
@ncisrobert assuming your problem is still valid, do a full uninstall of Arduino, then delete the Arduino folder and Arduino15 (back them up first if you have any sketches in there!!!!), as well as any arduino folders under Temp. Make sure the folders and all contents are really gone, reboot and try again if not. Then, do a clean install of the IDE.
I can vouch for the correct installation of the Arduino IDE under linus and windows 10, as of 1.8.4. Therefore, this looks like a user installation error, or corruption. 
This is not the right place to ask for installation assistance. This is an issue tracker, meant for tracking issues in the core. If you still need help, please ask for assistance at esp8266.com or stackoverflow.
Closing.

closed 2017-09-09 by devyte