esp8266/Arduino #2988: WiFi.config(ip) not working

mistergreen, 2017-02-22:
Hi,
WiFi.config(ip);
Throws an error. It's expecting ip, gateway, subnet.

> bool config(IPAddress local_ip, IPAddress gateway, IPAddress subnet, IPAddress dns1 = (uint32_t)0x00000000, IPAddress dns2 = (uint32_t)0x00000000);

I have an arduino ethernet sever where if I include the gateway and subnet, it wouldn't POST data to my php script on a web server. I'm seeing the same problem on the WiFi esp8266.

If any body knows how to solve 1 or 2 of the issue, let me know.

jjssoftware, 2017-02-26:
you're right..you need to at least supply IP addresses for local_ip, gateway and subnet when calling WiFi.config. The WiFi.config function is used to assign a static IP address to your widget so if you want this behaviour, you're going to need to supply the required params to this function. If you don't want/need a static IP assigning to your widget and you're happy with DHCP, don't bother calling WiFi.config.
* https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.h#L42

devyte, 2017-09-09:
@mistergreen the previous comment is correct: all 3 parameters are required for a static IP configuration. For DHCP, either don't call it, or call it with 0 values.
Having said that, this is not the right place to ask for help. This is an issue tracker, meant for tracking issues in the core. Please ask at esp8266.com or stackoverflow for assistance.
Closing.

closed 2017-09-09 by devyte