esp8266/Arduino #2989: Is there a possibility to make HTTPClient accept all certificates (without checking fingerprint)?

Defozo, 2017-02-22:
Hey, AFAIK as for now in order to make SSL request using HTTPClient you need to provide SSL fingerprint.
I'm wondering how you guys handle it, because the fingerprint can change over time.
Is there a possibility to make HTTPClient accept all certificates (without checking fingerprint)?

zoopla, 2017-02-23:
I look into HTTPClient code (commit 04df3adb54a73f13960738fa10a0a0b680f68cf0) and it seems to be impossible, at least in this state. A WiFiSecureClient instance is only created when fingerprint is provided and then the certificate is checked in the connect() method. 

The only secure solution based on fingerprint is to recompile your code and reflash your esp8266 with the new fingerprint string each time the fingerprint is updated (certificate has changed).

Otherwise if your issue is browsing the same site that regularly updates its certificate, WiFiClientSecure now supports checking against root ca certificate (only one for now I think), you can look this way.

Defozo, 2017-02-24:
My ESP8266s are connecting to my server which updates the fingerprint once a year.
However, I'd like to save myself a problem to update the whole firmware every year.

Perhaps I could connect to another website via HTTP and get fingerprint for my server and then connect to my server via HTTPS using this fingerprint?

Could you point me in the direction where should I look for information about that WiFiClientSecure supporting root ca certificate? I can't find it in the documentation.

salqadri, 2017-05-03:
I have a solution to this in the following PR: #3176. But it does not accept 'all' certificates; you have to load the root certificate of every different service you wish to access, but the great thing is that you no longer need to use fingerprints.

Defozo, 2017-05-04:
Thank you very much @salqadri for your input.

However, I started to wonder if this approach is safe? Can a root certificate become invalid without notice? If the solution needs SNTP enabled, what if that would be compromised (the connection to sntp - how does it work)?

salqadri, 2017-05-16:
@Defozo Yea you have to be careful about that. Most Root CAs will last a while (e.g., I have one on my machine expiring as late as 2046) but I've noticed that some root CAs on my machine expire as early as 2018! Depending on your application and the expiry of the root certs of the services you want to use, you may want to consider developing an upgrade mechanism that can deliver a new cert to your device.