esp8266/Arduino #2992: ESP8266 errors

Vedant803, 2017-02-24:
I just bought Node MCU ESP8266 developement board from amazon.**(link is down here)**
https://www.amazon.com/HiLetgo-Version-NodeMCU-Internet-Development/dp/B010O1G1ES/ref=sr_1_3?ie=UTF8&qid=1487921045&sr=8-3&keywords=ESP8266
And whenever I try to upload script from Arduino it compiles the sketch.
but after compiling it is showing this errors,
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_open failed
now I don't know what to do, help me ASAP.

igrr, 2017-02-24:
Please follow the troubleshooting guide for this issue:
http://arduino-esp8266.readthedocs.io/en/docs_to_readthedocs/faq/a01-espcomm_sync-failed.html

Feel free to reopen this if you can't resolve it, in which case please provide information mentioned in the "I'm stuck" section.

closed 2017-02-24 by igrr

Vedant803, 2017-02-24:
didn't helped me,
still facing this error.

JamesGKent, 2017-02-24:
did you miss this part of the comment above:
`Feel free to reopen this if you can't resolve it, in which case please provide information mentioned in the "I'm stuck" section.`

Vedant803, 2017-02-24:
okay, Firstly I'm trying to do this,**(watch this video)**
https://www.youtube.com/watch?v=N5JVQ-m5Kd0&t=649s
My actual problem starts at **8:38(pause the video at it and have a look.)**
in his Video it is showing **(Updating XXX bytes n blah blah blah)**
I did Everything right just when I reached at this uploading part the problem accord.
I have this chip -> https://www.amazon.com/HiLetgo-Version-NodeMCU-Internet-Development/dp/B010O1G1ES/ref=sr_1_3?ie=UTF8&qid=1487921045&sr=8-3&keywords=ESP8266
2nd. I'm not a Electrician nor anything and I don't even have stuffs to do **Advance check.**
I explained My **exact problem,**
hope you guys will help...

Vedant803, 2017-02-24:
anyone there?help me pls

d-a-v, 2017-02-24:
Excellent :)

Vedant803, 2017-02-24:
do i have  to do anything with library?
reply me ASAP

vicnevicne, 2017-02-25:
Sentences like "didn't helped me,", "I explained My exact problem,", "anyone there?" or "reply me ASAP" guarantees **most developers will just skip your issue**. Most people are here on their free time and have more interesting things to do than receive such orders from someone who doesn't show much effort nor gratitude. And requesting that all developers watch a 13min youtube video of **something that works** is just useless. We know how it should work.
If you want a quick response from ESP8266 enthusiasts, I advise you to post this question in the user forums at esp8266.com . But no matter if it's here or on the forums, please at least show your efforts, explain what *you* have done to solve your problem.
If not, and you are patient and willing to learn, I'm just going to repeat what igrr said :
**"please provide information mentioned in [the "I'm stuck" section of the troubleshooting page](http://arduino-esp8266.readthedocs.io/en/docs_to_readthedocs/faq/a01-espcomm_sync-failed.html#i-m-stuck)"**
If you can't follow the *Advanced check*, then at least answer for the tests you have made.

RudyFiero, 2018-10-28:
> espcomm_send_command: wrong direction/command:

I got the above error when trying to program a new ESP-12F. I have never had this kind of error before. I tried a number of things and always the same. I had the module plugged into my most used development platform. I have been using it all morning. I just wanted to program a new module with the latest project code, then move it to the final hardware.  

I knew it wasn't a problem with the computer, the Arduino framework, the hardware board I had the ESP12F module plugged into. It has to be a problem with the ESP12F. It would start to program, get through about 10% of the code, then the error. 

I thought it might be a timing problem. So I reduced the baud rate from 115200 to 57600. It programmed to completion.  Not sure why, but it is worth a try is you get this error.