esp8266/Arduino #2993: Loading problem library

clecle16, 2017-02-24:
Hello,

I can't manage to load the library in the Arduino IDE 1.8.1. 
Every time I add .ZIP library, I have this error message : 

`Arduino: 1.8.1 (Windows XP), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

java.io.IOException: Zip doesn't contain a library
	at processing.app.Base.handleAddLibrary(Base.java:2246)
	at processing.app.Base$6.actionPerformed(Base.java:1067)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:833)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:877)
	at java.awt.Component.processMouseEvent(Component.java:6533)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6298)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Zip doesn't contain a library

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

How I can I fix that ? Did I miss something ? 
Thanks

jjssoftware, 2017-02-26:
_"Every time I add .ZIP library"_
You didn't mention what zip file so in the absence of that, have you considered that whatever zip file it is you're referring to, it might not contain a library as suggested by the output?

clecle16, 2017-02-27:
Sorry, I was referring to the esp8266 library

jjssoftware, 2017-02-27:
Right got it. You're trying to get the esp8266 arduino core installed into your arduino ide by downloading one of the zip files from github and using that as a direct library load zip. That's not the way to get this project installed correctly. Start here: https://github.com/esp8266/Arduino/blob/master/README.md#installing-with-boards-manager

jjssoftware, 2017-02-27:
No need for any other IDE. 

So you've followed the [setup guide](https://github.com/esp8266/Arduino/blob/master/README.md#installing-with-boards-manager), you've installed the esp8266 arduino core in the arduino IDE, you can see the esp8266 board support package in the arduino boards manager and it looks like this:

[image]

If so you should be able to open one of the available esp8266 example sketches from the File -> Examples -> Examples for generic esp8266 module section here:

[image]

..make sure you've got an esp8266 board selected in Tools -> Board here:

[image]

and then Sketch -> Verify/Compile.

If that doesn't work, you've not got it installed correctly.

clecle16, 2017-02-28:
Hi, I already did (many time) this installation..
I select Genereic ESP8266, open any exemple of esp8266 (WifiClient as your exemple) and I have this error message : 
`Arduino: 1.8.1 (Windows XP), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

In file included from C:\Documents and Settings\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.h:27:0,

                 from C:\Documents and Settings\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.h:28,

                 from C:\Documents and Settings\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:34,

                 from C:\Documents and Settings\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\examples\WiFiClient\WiFiClient.ino:9:

c:\documents and settings\local settings\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\functional:48:28: fatal error: bits/c++config.h: No such file or directory

 #include <bits/c++config.h>

                            ^

compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

JamesGKent, 2017-02-28:
there seems to be a bug with the arduino compiler setup on xp where it's unable to find some of the files that it has been told to include, if you follow the suggestion in this comment:
[https://github.com/esp8266/Arduino/issues/1631#issuecomment-192680550](https://github.com/esp8266/Arduino/issues/1631#issuecomment-192680550)
you should be able to get it working.

jjssoftware, 2017-02-28:
Yeah windows. Sorry not a clue. Perhaps some other windows user can help you out.

JamesGKent, 2017-02-28:
i would suggest trying that fix above, i have used this library on windows XP, 7 and on 10, for some reason only XP suffers from this issue, but following the suggestions in that comment fixed the issue for me, note that if you do it correctly it should not ask you about overwriting files as none of the files in the source or destination folders should share the same name.

clecle16, 2017-02-28:
So, I tried the solution you proposed me. 
When I move the files, it ask me to overwrite on the news files so I didn't do it
I take the problem in the order : I just move c++config.h because it's missing. 
When I moved it, I had no overwriting message. 
I do it again 2 time for to other file and now I'm at this file : stl_iterator_base_types.h
And the problem is that this file is already in the good folder, I don't understand why the code (stl_algobase.h here) don't detect it 

here me error code of arduino 

`
Arduino: 1.8.1 (Windows XP), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

In file included from c:\documents and settings\frenchstudent\local settings\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\memory:62:0,

                 from C:\Documents and Settings\frenchstudent\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\Esp.cpp:24:

c:\documents and settings\frenchstudent\local settings\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:65:42: fatal error: bits/stl_iterator_base_types.h: No such file or directory

 #include <bits/stl_iterator_base_types.h>

                                          ^

compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

JamesGKent, 2017-02-28:
if you copy the whole contents of the folder `4.8.2\xtensa-lx106-elf\bits` to `4.8.2\bits` you should not receive any overwriting messages, so only if you copy the folder itself (because there are two `bits` folders...
so in `4.8.2\xtensa-lx106-elf\bits` you should have:
- atomic_word.h
- basic_file.h
- c++allocator.h
- c++config.h
- c++io.h
- c++locale.h
- cpu_defines.h
- ctype_base.h
- ctype_inline.h
- cxxabi_tweaks.h
- error_constants.h
- extc++.h
- gthr-default.h
- gthr-posix.h
- gthr-single.h
- gthr.h
- messages_members.h
- opt_random.h
- os_defines.h
- stdc++.h
- stdtr1c++.h
- time_members.h

and in `4.8.2\bits` you need to have:
- algorithmfwd.h
- allocator.h
- alloc_traits.h
- atomic_base.h
- atomic_lockfree_defines.h
- atomic_word.h
- basic_file.h
- basic_ios.h
- basic_ios.tcc
- basic_string.h
- basic_string.tcc
- boost_concept_check.h
- c++0x_warning.h
- c++allocator.h
- c++config.h
- c++io.h
- c++locale.h
- char_traits.h
- codecvt.h
- concept_check.h
- cpp_type_traits.h
- cpu_defines.h
- ctype_base.h
- ctype_inline.h
- cxxabi_forced.h
- cxxabi_tweaks.h
- deque.tcc
- error_constants.h
- exception_defines.h
- exception_ptr.h
- extc++.h
- forward_list.h
- forward_list.tcc
- fstream.tcc
- functexcept.h
- functional_hash.h
- gslice.h
- gslice_array.h
- gthr-default.h
- gthr-posix.h
- gthr-single.h
- gthr.h
- hashtable.h
- hashtable_policy.h
- hash_bytes.h
- indirect_array.h
- ios_base.h
- istream.tcc
- list.tcc
- localefwd.h
- locale_classes.h
- locale_classes.tcc
- locale_facets.h
- locale_facets.tcc
- locale_facets_nonio.h
- locale_facets_nonio.tcc
- mask_array.h
- memoryfwd.h
- messages_members.h
- move.h
- nested_exception.h
- opt_random.h
- ostream.tcc
- ostream_insert.h
- os_defines.h
- postypes.h
- ptr_traits.h
- random.h
- random.tcc
- range_access.h
- regex.h
- regex_compiler.h
- regex_constants.h
- regex_cursor.h
- regex_error.h
- regex_grep_matcher.h
- regex_grep_matcher.tcc
- regex_nfa.h
- regex_nfa.tcc
- shared_ptr.h
- shared_ptr_base.h
- slice_array.h
- sstream.tcc
- stdc++.h
- stdtr1c++.h
- stl_algo.h
- stl_algobase.h
- stl_bvector.h
- stl_construct.h
- stl_deque.h
- stl_function.h
- stl_heap.h
- stl_iterator.h
- stl_iterator_base_funcs.h
- stl_iterator_base_types.h
- stl_list.h
- stl_map.h
- stl_multimap.h
- stl_multiset.h
- stl_numeric.h
- stl_pair.h
- stl_queue.h
- stl_raw_storage_iter.h
- stl_relops.h
- stl_set.h
- stl_stack.h
- stl_tempbuf.h
- stl_tree.h
- stl_uninitialized.h
- stl_vector.h
- streambuf.tcc
- streambuf_iterator.h
- stream_iterator.h
- stringfwd.h
- time_members.h
- unique_ptr.h
- unordered_map.h
- unordered_set.h
- uses_allocator.h
- valarray_after.h
- valarray_array.h
- valarray_array.tcc
- valarray_before.h
- vector.tcc

clecle16, 2017-02-28:
Yes now I have the same names files as you describe.
But when I run the exemple of esp8266 (Blink here), I still have the same problem, the same error message as previously with the Blink exemple.
If I test is with WifiClient I have this type of error message : 

```
Arduino: 1.8.1 (Windows XP), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

In file included from c:\documents and settings\frenchstudent\local settings\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\typeinfo:34:0,

                 from c:\documents and settings\frenchstudent\local settings\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\functional:53,

                 from C:\Documents and Settings\frenchstudent\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.h:27,

                 from C:\Documents and Settings\frenchstudent\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.h:28,

                 from C:\Documents and Settings\frenchstudent\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:34,

                 from C:\Documents and Settings\frenchstudent\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\examples\WiFiClient\WiFiClient.ino:9:

c:\documents and settings\frenchstudent\local settings\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\exception:38:42: fatal error: bits/atomic_lockfree_defines.h: No such file or directory

 #include <bits/atomic_lockfree_defines.h>

                                          ^

compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

JamesGKent, 2017-02-28:
you can't have the same list as i put above or you would not be getting the error message you just posted, please copy the entire contents of `4.8.2\xtensa-lx106-elf\bits` to `4.8.2\bits` the only messages about overwriting should be files you have previously copied, you need to copy all of those files to get around this issue.

clecle16, 2017-02-28:
Ok, I did it and now when I run the Blink exemple I still have this message : 
``` 
Arduino: 1.8.1 (Windows XP), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

In file included from c:\documents and settings\frenchstudent\local settings\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\memory:62:0,

                 from C:\Documents and Settings\frenchstudent\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\Esp.cpp:24:

c:\documents and settings\frenchstudent\local settings\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:65:42: fatal error: bits/stl_iterator_base_types.h: No such file or directory

 #include <bits/stl_iterator_base_types.h>

                                          ^

compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

this is thwo bits folders side by side, I can't see anything wrong...
[image]

JamesGKent, 2017-02-28:
i've looked through your list and your files match what i have, all i can say is that it works on my xp machine with arduino IDE 1.6.9.

have you tried restarting the arduino IDE?

clecle16, 2017-02-28:
I tried on an other PC (win10 same IDE version) and it's working, so I think the problem come from the XP PC ... 
Yes every time I change something I restart the arduino IDE
Thank's for the help :) , I will try to continue on another pc under win 10 ..

JamesGKent, 2017-02-28:
i know that this bug only happens on XP (but don't know why) i have used this library without having to do any changes on windows 7 and 10, but i managed to get mine working with minimal effort so gave up trying to work out why, not much point chasing an issue on a well out of support EOL OS...

closed 2017-05-15 by igrr