esp8266/Arduino #2994: ESP8266 I'm Stuck.

Vedant803, 2017-02-24:
okay, Firstly I'm trying to do this,(watch this video)
https://www.youtube.com/watch?v=N5JVQ-m5Kd0&t=649s
I did exact same things which he's doing. plus I'm windows 7 user.
My actual problem starts at 8:38(pause the video at it and have a look.)
in his Video it is showing (Updating XXX bytes n blah blah blah) at 8:38.
but in mine, this error is occurring,(the port 5 is choosen which is only i can select.)
Sketch uses 522510 bytes (50%) of program storage space. Maximum is 1044464 bytes.
Global variables use 51560 bytes (62%) of dynamic memory, leaving 30360 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_open failed
I did Everything right just when I reached at this uploading part the problem occurred.
I have this chip -> https://www.amazon.com/HiLetgo-Version-NodeMCU-Internet-Development/dp/B010O1G1ES/ref=sr_1_3?ie=UTF8&qid=1487921045&sr=8-3&keywords=ESP8266
2nd. I'm not a Electrician nor anything and I don't even have stuffs to do Advance check.
I explained My exact problem,
hope you guys will help...

vicnevicne, 2017-02-25:
You posted this problem also in issue #2992 and #466. Polluting every other issue with a copy/paste of your problem is just wasting developer's time and introducing confusion. It's not going to help you.
This system is meant to track bugs in the Arduino environment for ESP8266, but unless you prove otherwise, the problem is most probably on your side.
If you want quick response from ESP8266 enthusiasts, I advise you to post this question in the user forums at esp8266.com . If not, please at least remain focused on bug #2992 and expose what you have already tried to solve your problem...

jjssoftware, 2017-02-26:
you could try working through this: https://github.com/esp8266/Arduino/blob/master/doc/faq/a01-espcomm_sync-failed.md

devyte, 2017-09-09:
The cause for the error is documented. If that doesn't work, your board is likely dead.
This is not the right place for usage help. This is an issue tracker, meant for tracking issues in the core. Please refer to esp8266.com or stackoverflow.
Closing.

closed 2017-09-09 by devyte