esp8266/Arduino #2996: fatal error ESP8266WiFi.h no such file or directory

flirt, 2017-02-25:
I want to turn my esp8266 into access point so that i would able to connect directly to it.i use the following sketch but when i select GENERIC module ESP8266 from menu the two libraries which is ESP8266WiFi.h and #include ESP8266WebServer.h become available but at this point i was unable to upload sketch to arduino.Also when i select Board as arduino uno (Which i have really) and then what to upload then the above two files becomes unavailable and says fatal error no such file...no idea what to do next...Thanks in advance for kind attention.

jjssoftware, 2017-02-26:
_"i use the following sketch"_ That's nice..what sketch might that be?
Anyway it sounds like you have some sort of local install / setup problem. The best I can offer is you start again here: https://github.com/esp8266/Arduino/blob/master/README.md#installing-with-boards-manager

lmk1988, 2017-02-27:
Which Arduino version are you using? The latest that that worked for me was 1.6.9, any higher it generates compile errors. Official is on 1.6.8, perhaps try downgrading to it if you are using a higher Arduino Version

jjssoftware, 2017-02-27:
I have this installed and working in v1.8.1 of the Arduino IDE here. This is their latest version today and a test compile of one of the included sample sketches works without any build issues. This is running on Linux Mint 18.1.

I'm not a fan of the Arduino IDE and I don't actually use it..but that's another story.

lmk1988, 2017-02-28:
:( Sadly I just tried 1.8.1, it doesn't work for me. It still have compile errors but due to external .h files

jjssoftware, 2017-02-28:
take a look at the screenshots I posted in this other issue: https://github.com/esp8266/Arduino/issues/2993#issuecomment-282806813

if that doesn't help, I suggest you post your own screenshots

Ahmedabbas1993, 2017-06-15:
Hi dear 
I'm use the blynk library,stand alone program but there is an error when I verify the program ,
c/users/arduino /libraries............. no such file or directory

Roanb89, 2017-07-08:
Im having a similar error "fatal error: data.h: No such file or directory" then under, it says #include "data.h"

devyte, 2017-09-09:
There is no compilation issue using the Arduino IDE 1.8.4 and with latest git core.
This as either user error or installation error. For assistance, please refer to esp8266.com or stackoverflow.
Closing.

closed 2017-09-09 by devyte

crwhite57, 2021-05-23:
I love using Arduino IDE but I hate it when someone is getting an error, people like devyte just say that it is a user or installation error. We are asking for help with the error, not asking what it is not. I am trying to compile a sketch and I keep getting a "wl_definitions.h: No such file or directory" even though when I try to install the library that has the file, Arduino tells me that it is already installed. Normally if I get something like this I just find the library and install it and that fixes it, but not with this. Does anyone have any ideas about what is going on? I am using Arduino IDE 1.8.13 on Windows 10 and I have my board selected as Modemcu32s. It is an esp-wroom-32 developmental board. Maybe I have the wrong board selected but I don't know. It worked fine when I was programing it to work with alexa and some relays.

d-a-v, 2021-05-23:
Lots of user error indeed in your post:
- You are using an esp32 but you're in the esp8266 arduino core issue tracker.
- This issue tracker is meant to track the esp8266 arduino core issue and not meant to help users with installation issue,
  except when the issue is because of the arduino core installer - which is not the case here.
  The reason is that it'd be too much work for the people responsible for the repo.
  User errors are generally well managed in forum like stack overflow arduino section for example, or esp8266.com for esp8266 users.
- If there was a real issue with `wl_definitions.h`, nobody could use this core.
- This is a closed issue. If there's something wrong, please open a new issue.

We hope you'll be able to use your ESP32 with the right arduino core.
You need to select the right board in the Arduino IDE tools menu.
You seem to be using the esp8266 specific libraries with an esp32 installation. I personally don't know how this can happen.
I'd advise you to delete your arduino installation including the "arduino15" directory, and start over from scratch.

themindfactory, 2021-05-23:
> Lots of users error indeed in your post:

Either its a case of picking the right board definition or your system is
real messed up and its time to start from scratch

crwhite57, 2021-05-24:
Ok, thanks.

umtolva, 2021-11-02:
I had the same issue. It's fixed after I followed the installation guide that is written in github README.md
In my case, it's because I just copied ESP8266WiFi directory to the Arduino libraries directory directly.