esp8266/Arduino #2997: Socket status?

mistergreen, 2017-02-26:
With the arduino Wifi shield, you can show the socket status with this code.

```
 for(int x = 0; x < MAX_SOCK_NUM; x++) {
    Serial.print(x);
    Serial.print(" - ");
    Serial.print(WiFi._state[x]);    
    Serial.print("  ");
    Serial.println(WiFi._server_port[x]);    
  }
```

What is the equivalent for the esp8266?

jjssoftware, 2017-02-26:
have you tried using the ethernet library (Ethernet.h) included in this project? all of those symbols and properties in your code snip are in there..

* https://github.com/esp8266/Arduino/tree/4897e0006b5b0123a2fa31f67b14a3fff65ce561/libraries/Ethernet
* https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/libraries/Ethernet/src/Ethernet.h#L11
* https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/libraries/Ethernet/src/Ethernet.h#L18
* https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/libraries/Ethernet/src/Ethernet.h#L19
* https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/libraries/Ethernet/examples/WebClient/WebClient.ino

devyte, 2017-09-09:
@mistergreen did you ty the suggestion in the previous comment?

devyte, 2018-02-28:
Closing due to lack of feedback and age.

closed 2018-02-28 by devyte