esp8266/Arduino #2998: Linking Error on new SDK 2 with time.h

sticilface, 2017-02-26:
Compiling the basic NTPClient.ino example gives the following Linking Error since ae13809c8184300aab9e3f09ef23af23d936b7ee

```
Linking everything together...
"/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc" -g  -Os -nostdlib -Wl,--no-check-sections -u call_user_start -u _printf_float -u _scanf_float -Wl,-static "-L/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/lib" "-L/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/ld" "-L/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/lib" "-Teagle.flash.512k64.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read  -o "/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_147180/NTPClient.ino.elf" -Wl,--start-group "/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_147180/sketch/NTPClient.ino.cpp.o" "/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_147180/libraries/ESP8266WiFi/ESP8266WiFi.cpp.o" "/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_147180/libraries/ESP8266WiFi/ESP8266WiFiAP.cpp.o" "/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_147180/libraries/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o" "/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_147180/libraries/ESP8266WiFi/ESP8266WiFiMulti.cpp.o" "/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_147180/libraries/ESP8266WiFi/ESP8266WiFiSTA.cpp.o" "/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_147180/libraries/ESP8266WiFi/ESP8266WiFiScan.cpp.o" "/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_147180/libraries/ESP8266WiFi/WiFiClient.cpp.o" "/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_147180/libraries/ESP8266WiFi/WiFiClientSecure.cpp.o" "/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_147180/libraries/ESP8266WiFi/WiFiServer.cpp.o" "/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_147180/libraries/ESP8266WiFi/WiFiUdp.cpp.o" "/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_147180/arduino.ar" -lhal -lphy -lpp -lnet80211 -llwip -lwpa -lcrypto -lmain -lwps -laxtls -lespnow -lsmartconfig -lmesh -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group  "-L/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_147180"
/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/lib/libmain.a(time.o): In function `millis':
(.irom0.text+0x94): multiple definition of `millis'
/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_147180/arduino.ar(core_esp8266_wiring.c.o):/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring.c:64: first defined here
/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/lib/libmain.a(time.o): In function `micros':
(.irom0.text+0xd0): multiple definition of `micros'
/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_147180/arduino.ar(core_esp8266_wiring.c.o):/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring.c:70: first defined here
/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/lib/libmain.a(time.o): In function `configTime':
(.irom0.text+0x138): multiple definition of `configTime'
/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_147180/arduino.ar(time.c.o):/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/time.c:65: first defined here
/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/lib/libmain.a(time.o): In function `time':
(.irom0.text+0x1dc): multiple definition of `time'
/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_147180/arduino.ar(time.c.o):/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/time.c:87: first defined here
collect2: error: ld returned 1 exit status
```

igrr, 2017-02-27:
I'm was sure I had this fixed at some point, will check again.
Also not sure why CI has not caught this... maybe I have messed up building the tests directory?

sticilface, 2017-03-05:
OK, figured it out.  It only ocured when I used the core development version as the selected board. 

Further more, if i then select open source lwip, it goes away.  

So that is why it is occurring and how to fix it.

d-a-v, 2017-12-13:
@sticilface can this issue be closed ?

PR #3611 (2017-09-15): tools/sdk: remove conflicting time.o from libmain.a
Fixes #2998
closed 2018-01-02 by igrr