esp8266/Arduino #3000: How to send and receive data from TCP IP based ?

Joypacharya, 2017-02-27:
Hello.
Actually i am using ESP8266 - 12E module.I done some task from MQTT server.
But i want data Tx and Rx via TCP IP.So, its easily communicate via this coding. 
I searched so many links still it cant be worked. Manily thay used ESP-01 and data use via GET and POST (HTTP based).
So, i want to deliver data directly using this TCP IP even receive  from it.Help me for this issues. Thank you in advance.....

lmk1988, 2017-02-27:
Do you mean this?
[WiFiClient.ino](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiClient/WiFiClient.ino)

It's sort of the same

- **Connect to host/IP address after connecting to router**
`// Use WiFiClient class to create TCP connections`
  `WiFiClient client;`
  `const int httpPort = 80;`
  `if (!client.connect(host, httpPort)) {`
    `Serial.println("connection failed");`
    `return;`
`}`
_NOTE for WiFiClient Connect function, it accepts IPAddress too:_
`connect(IPAddress ip, uint16_t port);`

- **Tx**
`client.print("Your RAW text to send");`

- **Rx**
`while(client.available()){`
    `String reply = client.readStringUntil('\r');`
    `Serial.print(reply);`
`}`

Joypacharya, 2017-02-28:
I want work same like MQTT. Any time i will get information from this IP and port.
One more question ?
i do not need convert string to byte buffer.
likewise  <5F CF 10 FF EF ........... >

lmk1988, 2017-02-28:
I'm sorry if I do not quite understand the question but here are a few answers to the question I think you are asking.

- 1) **You want MQTT (least likely the question asked)**
Well you can actually google for libraries to do that, so I do not think this is the question asked

- 2) **You want something like MQTT but using TCP**
If you do not disconnect the WiFiClient, the TCP socket should still remain unless you have a TCP timeout implemented which you will either need to disable that or implement a interval TCP refresh packet

- 3) **You want something like MQTT but not MQTT but with notification**
Have you tried websockets library? I recommend [arduinoWebsockets](https://github.com/Links2004/arduinoWebSockets). I think this should be able to fulfill your requirements

I do not understand the convert string to byte buffer part though. If I am not mistaken you are asking if you need to convert String to a byte buffer to use in `client.print`? Well internally Strings are sort of stored as array if I do not understand wrongly. So you could actually just do this if it is not accepted

`String temp = "hello world";`
`temp.c_str(); //<-- this is the buffer` ([for reading only](https://www.arduino.cc/en/Reference/CStr))
`temp.length(); //<-- length of buffer excluding null terminator`

But then again because `WiFiClient` extends `Client` which extends `Stream` which extends `Print` which contains the function `size_t print(const String &);` It is fine to use String directly

devyte, 2017-09-09:
@Joypacharya the answer is WiFiClient and WiFiServer. They are essentially objects that represent sockets, as you need.
Having said that, this is not the right place to ask "how do I..." type questions. This is an issue tracker, meant for tracking issues in the core. Please refer to readthedocs, or ask at esp8266.com or stackoverflow.
Closing.

closed 2017-09-09 by devyte