esp8266/Arduino #3003: MD5 for SPIFFS file checking: Feature request

SensorsIot, 2017-02-28:
We implemented the MD5 mechanism to check version of the sketch file in our IOTappstory.com infrastructure and it works great.
Now, we also implemented the download of the SPIFFS OTA from the same site. Unfortunately, MD5 is not supported by the SPIFFS downloader. Would it be possible to write a patch for that, too?

devyte, 2017-10-01:
@SensorsIot does the proposed code address your request?

SensorsIot, 2017-11-27:
It works now with the 2.4.0 version.

closed 2017-11-27 by SensorsIot

mcmega, 2017-12-12:
I get the value is not true. Windows gives one value, but the function is different.
```cpp
String md5File (File & f) {
  if (f.seek(0, SeekSet)) {
    MD5Builder md5;
    md5.begin();
    md5.addStream(f, f.size());
    md5.calculate();
    return md5.toString();
  }
  return String();
}
```
If you download the file and count md5, then the values are different. What could be the problem?