esp8266/Arduino #3004: Arduino UNO WiFi's ESP8266 module won't boot. No response at all.

gvJaime, 2017-02-28:
### Basic Infos

#### Hardware
Hardware:			Arduino UNO WiFi's esp8266
Core Version:      	2.3.0

### Description

Arduino UNO WiFi's esp8266 won't do anything when loaded with a firmware made with the Arduino esp8266 core. No wifi, no serial, no GPIO response. Not even debug messages, or stack messages. The ESP8266 won't even heat.

#### What does work
It will however, boot with the [Arduino original firmware](https://github.com/arduino-org/Esp-Link) and the [original Jeelabs firmware](https://github.com/jeelabs/esp-link) this is based on.

Original source-code-examples will boot, but serial communication is difficult due to [the wiring of the UART0 to the USB port](http://download.arduino.org/products/UNOWIFI/Arduino-UNO-WiFi-V4_AF1.pdf).

#### What does not work

Using eboot to boot the original esp-link firmware won't work. Also, trying to boot Arduino Core made firmware with espressif's bootloader won't work either (I don't know if doing that even makes sense).

#### Hardware particularities
The module seems nothing special compared to any other board, except for the fact that it has an UART to I2C attached to the UART0 pins, and into the Arduino I2C pins. Serial communication relays in [a sketch that turns the ATMEGA328P into another I2C to Serial converter](https://github.com/arduino-org/arduino-library-arduino-wifi/blob/master/examples/Tools/EspRecovery/EspRecovery.ino), so it goes throught the I2C to Serial, the ATMEGA, and the Serial-to-USB at a max rate of 9600 baud.

This process can be skipped by grounding a testpoint in the board, so the UART0 is connected directly to the Serial-to-USB module, and faster baudrates are possible. No serial communication is detected this way either.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		80MHz
Flash Mode:			qio
Flash Frequency:	80MHz
Upload Using:		SERIAL 9600 baud
Reset Method:		nodemcu works.

This has also been tried with the PlatformIO IDE. With the same results. Using espressif's esptool, or igrr esptool-ck yields same results.

### Sketch

```main.cpp
#include <ESP8266WiFi.h>

void setup()
{
  Serial.begin(9600);
  Serial.println();
  Serial.println("Initializing...");
  Serial.println();
}

void loop(){
  delay(1000);
  Serial.println("Test");
}

```

Although I tried every example I found out there.

### Debug Messages

We get no response from the chip, so there is no Debug messages.

Although, when the board is freshly uploaded with a new sketch, the Serial Monitor always spits 3 characters, usually "00,". Upon restarting, no signal is received anymore.

When following the instructions, selecting 9600 baud, this code appeared:

```
00, c112
~ld
```

Again, upon restart, no messages are detected anymore.

gvJaime, 2017-03-06:
Hello @RudyFiero . Sorry for the late response, I'd sweared that I answered, but apparently got distracted and never clicked "Comment". 

The board I'm using is an [Arduino UNO WiFi](http://www.arduino.org/products/boards/arduino-uno-wifi). Which is apparently an Arduino UNO with an ESP8266 module integrated in it, with programmer and everything. 

As far as I can conclude from [the schematic](http://download.arduino.org/products/UNOWIFI/Arduino-UNO-WiFi-V4_AF1.pdf), the ESP8266 is installed the same way it would go in any other ESP8266 based board, so I expected that a sketch made with this library would work, but for some reason it doens't boot. It doesn't do anything at all.

I've read all the information there is on the wiki regarding this subject: Memory Map, flashing process and methods. The board has its particularities (weird bridge when it comes to serial comm) but aside of that, these are not features that would make an ESP8266 just sit there. I'm missing something.

Which board are you using? Is there any other information I can provide? 

Thanks and greetings

lrmoreno007, 2017-03-07:
You want to use the Arduino Uno Wifi in the opposite direction for which it was designed.

You want to use the CPU of ESP8266 side but the I / O are connected in the Arduino side. Assuming you can program the ESP8266, then you will have to communicate via UART with the Arduino side to handle the I / O. Of course this Arduino side must also have a firmware appropriate for this.

It's crazy what you want to do and make your life complicated.

You should use it in the way it was designed for, directly selecting "Arduino Uno Wifi" in the boards selector. You don't need this project ESP8266 / Arduino, you can uninstall it.

More info: http://www.arduino.org/learning/getting-started/getting-started-with-arduino-uno-wifi Select "First sketch" tab.

Good luck.

RudyFiero, 2017-03-07:
I had a look at the Arduino UNO WiFi schematic and I have to agree with lrmoreno007. You are not using the board as intended. The expectation is to use the ESP8266 as a communications module and not as the primary processor for the board. 

I primarily am working with the ESP8266 (ESP-12) and not using the AVR microcontrollers. But there are problems with this approach. I don't have full control of the processor. I have to be aware that the WiFi functions need to have time to work. And I don't always have precise control because of this. But for many things that isn't a problem.

The ESP8266 has only on A/D channel and there are accuracy issues with it. One way this, and the above issue, is to use a second processor to do the I/O and control functions and use the ESP8266 mainly for WiFi communications. That is what the Arduino UNO WiFi, and also the ESP-14 from Expressif, do.

The dual processor combination is a good approach but I want to use the ESP8266 for more than a WiFi communications device. It has a lot more program memory and ram available. It is also a lot faster. Using  it and another processor is more complicated though since you need to program both sides and handle the communications between them.

lrmoreno007, 2017-03-07:
http://www.arduino.org/learning/tutorials/advanced-guides/how-to-change-the-firmware-on-an-arduino-uno-wifi

But It isn't the best place to discuss this.

Please close it.

lrmoreno007, 2017-03-07:
http://www.arduino.org/learning/tutorials/advanced-guides/how-to-change-the-firmware-on-an-arduino-uno-wifi

Be careful with this, can make the board useless.

This issue isn't the best place to discuss It.

Please close it.

gvJaime, 2017-03-07:
@lrmoreno007 , thanks for the links. I had already read them thoroughly, but I should have said it on my first post.

## Just a bit of context

My project scenario similar to the one @RudyFiero describes in his last paragraph: ESP8266 as an principal, more powerful processor coordinating the process, and communicating the outer world, and the ATMega strictly as a mecanical controller. It's not that uncommon. I got the Arduino UNO WiFi because it seemed like an integrated solution, as opposed of having a bunch of boards hanging around. I cannot use it the way Arduino wants, because the UNO is already filled to the brim with [GRBL](https://github.com/gnea/grbl). (also I2C pins are occupied). 

I was aware that this configuration would need extra programming work, but I didn't expect the board to just ignore custom firmwares.

## Regarding the scope of the issue

I have an ESP8266 board that won't boot with a custom firmware, despite being installed pretty much the same way as in any other ESP module. Although the board is not explicitly supported by the framework, I think this is exactly the place to discuss this, because if we succeed, it will be of interest of the Arduino ESP8266 Core project to add one more board to the list.

So I think it's better to keep this issue open and running.

## Back at technical attempts

I've been fiddling with the Espressif SDK, and the source code examples:
* Fiddled with the [basic_example](https://github.com/esp8266/source-code-examples/tree/master/basic_example), trying to print a "Hello!" to UART0 every second. I just received gibberish every second, but I think it's because sync issues.
* I tried [the blinky example](https://github.com/esp8266/source-code-examples/tree/master/blinky), but for GPIO 14 (It is connected to the "WiFi" led). It didn't work, although the ESP8266 heated like when it's working.

There has to be something in [Esp-link firmware](https://github.com/jeelabs/esp-link) inicialization, that my ESP8266 misses in the custom firmware, and makes it halt.

Another theory is that the SC16IS750IBS I2C bridge connected to the ESP8266 could be for some reason powering it down. (It has it's 3rd GPIO connected to the ESP-OFF pin in the ESP8266)

Are there any other scenarios in which an ESP8266 refuses to boot? (Aside from the ones in the [FAQ](https://github.com/esp8266/Arduino/blob/master/doc/faq/readme.md))

RudyFiero, 2017-03-07:
What I would do is try and isolate the ESP8266. What I mean by that is make sure you have control over the key signal lines. Because the SC161S750 is connected to the pins of the ESP8266 I would try to disable them. I expect if you held the SC161S750 in reset that it would no longer be a concern. 

I would try and make the ESP8266 look like it is in it's own little electrical world, load the blink code, then see what happens. And if you had a good digital oscilloscope then see what happens. I don't see why the ESP part can't be run on it's own, unless there is special code in ROM that prevents normal operation. I doubt that is the case but it is possible.

gvJaime, 2017-03-07:
> unless there is special code in ROM that prevents normal operation. I doubt that is the case but it is possible.

@RudyFiero , I've been thinking of that for a while, because when I started reading about this, I remember something like an internal ROM that was a very few bytes long, inside the ESP8266. Until one day I read this on page 8 of [the ESP8266EX datasheet](http://espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf):

> There is no programmable ROM in the SoC, therefore, user program must be stored
in an external SPI flash.

Since then, I've been thinking that all the execution relevant info (Bootloader, Program, Variable Data...) is stored in the SPI Chip, but now I am confused.

I have a series of questions that may seem pretty basic, but I just want to make sure that I understand everything clearly:
* Has the ESP8266 some kind of ROM with code aside from the user program?
* In case of it existing, where is it stored, inside the ESP8266, or in the SPI Memory?
* And in case of being in the SPI Memory, do we have any chance to change it? (I have a SOIC-8 clamp if necessary...)

Thank you for your patience. Meanwhile, I'm testing crucial pin levels while using a working firmware, and while using a custom non-working firmware, with a tester and two needles. Will post my findings tomorrow, when I get them together.

RudyFiero, 2017-03-07:
> There is no programmable ROM in the SoC, therefore, user program must be stored
in an external SPI flash.

The is no programmable ROM, but it does not say there is no ROM, as in not programmable. Yes user programs must be stored in external flash. But what about a boot loader program in ROM? 

I have not got deep into the chip yet. I have been trying to avoid that. That's why I am using the Arduino framework.

davisonja, 2017-03-07:
The ESP8266 has ROM, which includes a bootloader.
Among other things the bootloader tends to output a header to the serial
port on startup, and is responsible for starting the (SPI-Flash-Based)
Arduino main code. I believe it is the ESP ROM Bootloader that deals with
serial-based-updates to the SPI Flash.

I don't believe that the ESP ROM is changable, and it provides a number of
functions (such as Wifi) used by the Arduino libraries.

J,

lrmoreno007, 2017-03-07:
Go ahead that I can´t try this!!! but reading... http://www.arduino.org/learning/tutorials/advanced-guides/how-to-change-the-firmware-on-an-arduino-uno-wifi

**This must be the process:**
Select board: Arduino Uno Wifi
Load example: Tool/ESPRecovery from: https://github.com/arduino-org/arduino-library-arduino-wifi
Upload to the board.
Unplug Arduino Uno Wifi
Plug Arduino Uno Wifi but pressing at same time: Bootloader button

Now you are programming directly the ESP8266 because Arduino side is working as a simple FTDI
Load your sketch for the ESP8266 side in Arduino IDE. Note that you must communicate through the Serial with Arduino. SerialEvent Example can be a good start point
Select Board: Generic ESP8266, DIO, Flash 4M, Reset ck, Speed 115200, COM (your number)
Upload the sketch

Unplug and plug the board
Select board: Arduino Uno Wifi
Load your sketch for the Arduino side, you can communicate in both side as in Tool/ESPRecovery and must receive orders as in SerialEvent Example.
Upload to the board.

Good luck.

gvJaime, 2017-03-08:
@RudyFiero @davisonja 

> The ESP8266 has ROM, which includes a bootloader.

Is it safe to assume that the code in that ROM is the same in every ESP8266? I'm wondering if a manufacturer can decide the contents of that internal ROM.

Speaking of bootloaders: I observed that both the original bootloader provided by the SDK and the "eboot" bootloader provided in the Arduino Core range from 0x0000 to 0x1000, so I tried loading the original esp-link with the eboot bootloader. It didn't work: cold ESP8266, no response at all. Just wanted to write it down here.

Yesterday I was measuring pin voltages in the ESP8266, while running these three programs:
* Arduino's Esp-Link, uploaded with esptool
* Simple Blink example with GPIO14, made with Arduino Core. (GPIO14 has a LED connected in this board) compiled and uploaded with esptool-ck from platformio
* [Blinky example](https://github.com/esp8266/source-code-examples/blob/master/blinky/user/user_main.c) modified for GPIO14, made with original SDK. Compiled with the SDK tools, and uploaded with esptool.

I still haven't put together all the info, but I found that in the two latter examples, some pins were outputting 1,67V (???) including the GPIO14.
I already checked if it was blinking too fast by using ridiculously longer blink times, and by simply outputting a 1 or a 0. In all those cases the GPIO14 was 1,67V. I don't know what can cause this. Maybe the GPIOs aren't initialized properly?

If have been busy with other tasks today, so I couldn't organize the info I gathered yesterday into a good post. I will post it today or tomorrow, depending on when can I get time to research this.

@lrmoreno007 , The link you provide is a guide to flash Arduino's original firmware, or any possible update that they may publish. That's not what I'm talking about in this issue. This issue is about the Arduino WiFi not booting with a firmware made with the ESP8266 Arduino framework.
When I started researching this, I read and comprehended all the content from that article, so the flashing process is not a problem.

This last thing makes me wonder if the title of the issue could be misleading. Maybe I should change it to something that specifies that I'm trying to make the Arduino UNO WiFi's ESP8266 work with this framework.

gvJaime, 2017-03-16:
# Pin voltage analysis

Well! it took a bit longer because my old tester died on me. I got to borrow another one yesterday.

## What I did

I measured the voltage in several pins on the ESP8266, when running either:
* [Arduino's Esp-Link](https://github.com/arduino-org/Esp-Link)
* [ESP8266 source code example Blinky, on GPIOs 2 and 14](https://github.com/esp8266/source-code-examples/tree/master/blinky)
* A blink example with the Arduino Framework, for GPIO2 and GPIO14.

The reason I chose those output pins for blinking, is that GPIO2 not used for anything else and has an accesible testpoint, and GPIO14 has a LED connected.

## Results

| Esp-Link (Arduino)  | Blinky (GPIO2)  | Blinky (GPIO14)  | Blink2 (ArdCore)  | Blink14 (ArdCore) | Pin  |
|---|---|---|---|---|---|
| 3,33V | 3,33V  | 3,33V  | 3,33V | 3,33V  |XPD-DCDC   |
| 3,33V  | 3,33V  | 3,33V  | 3,33V  | 3,33V  |CHIP-EN   |
| 3,33V  | 3,33V  | 3,33V  | 3,33V  | 3,33V  |GPIO0   |
| 3,33V  | Blinking 3,33V-0V  | 3,33V  | 1.76V (???) | 1.76V (???) |GPIO2  |
| 3,33V  | 3,33V  | 3,33V  | 3,33V  | 3,33V  |GPIO4 |
| 3,33V  | 3,33V  | 3,33V  | 3,33V  | 3,33V  |GPIO13 |
| 3,33V (The led twitches)| 1.76V (???) | 1.76V (???) | 1.76 (???) | 1.76V (???) |GPIO14 |
| 3,33V  | 3,33V  | 3,33V  | 3,33V  | 3,33V  |GPIO15 |
| 3,33V  | 3,33V  | 3,33V  | 3,33V  | 3,33V  |GPIO12 (MTDI)|

Esp-Link and the Blinky example on GPIO2 work fine, but Blinky on the GPIO14 and the two blink examples with the arduino framework didn't work.

I don't understand those halfway high voltages on GPIO2 and GPIO14. It seems like there is no way to make a blink on the 14, and the arduino core also loses control over GPIO2.

I remember reading something about some internal pullup resistance on some of the pins, and that may explain those marks, but doesn't explain why the ESP8266 does nothing.

It's not surprising that the firmwares made with the arduino framework leave some of the GPIOs in a undetermined state, given the fact that, apparently, it doesn't even boot.

I'll keep on looking into this. And posting results here.

# Regarding issue name change

Re-reading the title and first post, I can see that they fail to explain the thing I'm trying to do with the board, so they have been changed.

gvJaime, 2017-03-17:
This morning I tried to use this framework withoud any IDE, with [the make file recommended in the README](https://github.com/plerup/makeEspArduino). 

I used a blink example on GPIO14 and it worked!!!. So I figure that maybe it's an IDE configuration issue.

I'll look further into this, because I still haven't got WiFi to work, but this is the first signs of life this board gives me.

I'll take this occasion to thank everyone's attention. I'll keep posting my findings here, until I hopefully get to the bottom of this.

gvJaime, 2017-03-22:
After some parameter tweaking, I got the UNOWiFi to work with PlatformIO, with this framework. 

Serial works, and GPIO works.

**But anything WiFi related won't work**, no AP, no STA mode, nothing. And still, all the functions return a "true".

I'll post the UNOWiFi.json that got this to work for now, and close this issue, as it seems to be the IDE's fault, not the framework.

UNOWiFi.json
```
{
  "build": {
    "core": "esp8266",
    "extra_flags": "-DLWIP_OPEN_SRC   -DARDUINO=10605  -DARDUINO_ARCH_esp8266  -DESP8266",
    "f_cpu": "80000000L",
    "f_flash": "40000000L",
    "flash_mode": "qio",
    "ldscript": "esp8266.flash.4m.ld",
    "mcu": "esp8266",
    "variant": "nodemcu"
  },
  "frameworks": [
    "arduino"
  ],
  "name": "Arduino UNO WiFi's ESP8266",
  "upload": {
    "maximum_ram_size": 81920,
    "maximum_size": 1044464,
    "require_upload_port": true,
    "resetmethod": "nodemcu",
    "speed": 9600
  },
  "url": "http://www.arduino.org/products/boards/arduino-uno-wifi",
  "vendor": "Espressif"
}
```

closed 2017-03-22 by gvJaime

gvJaime, 2017-04-02:
It turns out that the Arduino UNO WiFi is one of the few esp8266 boards that mount a 40MHz crystal instead of 26MHz, so everything timing-related didn't work: Memory reading, Serial transmission, WiFi transmission.

Changing the line [48] = 1 to [48] = 0 in cores/esp8266/core_esp8266_phy.c solves the problem.

There is a [pull request](https://github.com/esp8266/Arduino/pull/3025) that turns this into a -D flag, but for now the Arduino framework doesn't support 40MHz crystals unless you modify it.