esp8266/Arduino #3005: WiFiEvents.ino example: 'class ESP8266WiFiClass' has no member named 'onSoftAPModeProbeRequestReceived'

Defozo, 2017-02-28:
### Basic Infos
Core Version: 2.3.0

There's a problem with an example code here: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiEvents/WiFiEvents.ino
It uses methods that are not available.

### Description

Problem description

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB (1M SPIFFS)
CPU Frequency:	80Mhz
Flash Mode:		dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

Code from https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiEvents/WiFiEvents.ino

```cpp
/*
    This sketch shows how to use WiFi event handlers.
    In this example, ESP8266 works in AP mode.
    Three event handlers are demonstrated:
    - station connects to the ESP8266 AP
    - station disconnects from the ESP8266 AP
    - ESP8266 AP receives a probe request from a station
    Written by Markus Sattler, 2015-12-29.
    Updated for new event handlers by Ivan Grokhotkov, 2017-02-02.
    This example is released into public domain,
    or, at your option, CC0 licensed.
*/

#include <ESP8266WiFi.h>
#include <stdio.h>

const char* ssid     = "ap-ssid";
const char* password = "ap-password";

WiFiEventHandler stationConnectedHandler;
WiFiEventHandler stationDisconnectedHandler;
WiFiEventHandler probeRequestPrintHandler;
WiFiEventHandler probeRequestBlinkHandler;

bool blinkFlag;

String macToString(const unsigned char* mac) {
  char buf[20];
  snprintf(buf, sizeof(buf), "%02x:%02x:%02x:%02x:%02x:%02x",
           mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
  return String(buf);
}

void onStationConnected(const WiFiEventSoftAPModeStationConnected& evt) {
  Serial.print("Station connected: ");
  Serial.println(macToString(evt.mac));
}

void onStationDisconnected(const WiFiEventSoftAPModeStationDisconnected& evt) {
  Serial.print("Station disconnected: ");
  Serial.println(macToString(evt.mac));
}

void onProbeRequestPrint(const WiFiEventSoftAPModeProbeRequestReceived& evt) {
  Serial.print("Probe request from: ");
  Serial.print(macToString(evt.mac));
  Serial.print(" RSSI: ");
  Serial.println(evt.rssi);
}

void onProbeRequestBlink(const WiFiEventSoftAPModeProbeRequestReceived&) {
  // We can't use "delay" or other blocking functions in the event handler.
  // Therefore we set a flag here and then check it inside "loop" function.
  blinkFlag = true;
}

void setup() {
  Serial.begin(115200);
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);

  // Don't save WiFi configuration in flash - optional
  WiFi.persistent(false);

  // Set up an access point
  WiFi.mode(WIFI_AP);
  WiFi.softAP(ssid, password);

  // Register event handlers.
  // Callback functions will be called as long as these handler objects exist.
  // Call "onStationConnected" each time a station connects
  stationConnectedHandler = WiFi.onSoftAPModeStationConnected(&onStationConnected);
  // Call "onStationDisconnected" each time a station disconnects
  stationDisconnectedHandler = WiFi.onSoftAPModeStationDisconnected(&onStationDisconnected);
  // Call "onProbeRequestPrint" and "onProbeRequestBlink" each time
  // a probe request is received.
  // Former will print MAC address of the station and RSSI to Serial,
  // latter will blink an LED.
  probeRequestPrintHandler = WiFi.onSoftAPModeProbeRequestReceived(&onProbeRequestPrint);
  probeRequestBlinkHandler = WiFi.onSoftAPModeProbeRequestReceived(&onProbeRequestBlink);
}

void loop() {
  if (millis() > 10000 && probeRequestPrintHandler) {
    // After 10 seconds, disable the probe request event handler which prints.
    // Other three event handlers remain active.
    Serial.println("Not printing probe requests any more (LED should still blink)");
    probeRequestPrintHandler = WiFiEventHandler();
  }
  if (blinkFlag) {
    blinkFlag = false;
    digitalWrite(LED_BUILTIN, LOW);
    delay(100);
    digitalWrite(LED_BUILTIN, HIGH);
  }
  delay(10);
}
```

### Debug Messages

```
In function 'void setup()':

wifieventtest:79: error: 'class ESP8266WiFiClass' has no member named 'onSoftAPModeProbeRequestReceived'

   probeRequestPrintHandler = WiFi.onSoftAPModeProbeRequestReceived(&onProbeRequestPrint);

                                   ^

wifieventtest:80: error: 'class ESP8266WiFiClass' has no member named 'onSoftAPModeProbeRequestReceived'

   probeRequestBlinkHandler = WiFi.onSoftAPModeProbeRequestReceived(&onProbeRequestBlink);

                                   ^

exit status 1
'class ESP8266WiFiClass' has no member named 'onSoftAPModeProbeRequestReceived'
```

igrr, 2017-02-28:
You have specified that you are using

> Core Version: 2.3.0

but then you are using an example from master branch. This example happens to use functions which are not available in 2.3.0 (which were added in the same commit as the new example IIRC).

Defozo, 2017-02-28:
Thank you for your answer.
Could you tell me, where I can find this library installed, so I could swap the code with the newest one?
I can't find it anywhere: Neither in libraries folder in Arduino installation dir nor in Arduino folder in Documents in Windows.

jjssoftware, 2017-03-02:
> "where can I find this library installed"
if you're using 2.3.0 as you say you are, I'm going to go with "you don't have it installed". If you want it, directly install the git version: https://github.com/esp8266/Arduino#using-git-version

closed 2017-05-15 by igrr

macjonesnz, 2017-09-13:
Add the staging version of the board and it will compile. (2.3.0 rc2)
http://arduino.esp8266.com/staging/package_esp8266com_index.json

fralomb, 2017-10-25:
How can i add the staging version that you linked @macjonesnz ? I'm using PlatformIO as IDE.. Thank you!

macjonesnz, 2017-10-26:
Sorry, I don't use platformIO. 

On Arduino IDE ...
1. Start Arduino and open Preferences window.
2. Enter JSON URL into Additional Board Manager URLs field

google the rest....

:-)