esp8266/Arduino #3006: Installation Problems

thahnen, 2017-02-28:
Hello dead Developer,

i tried to install the ESP8266 using your Installation Instruction using the Arduino Software AND Git. But there seems to be an errror in the json File you have to copy because there ist a Link to the esptool-sk (Link: https://github.com/igrr/esptool-ck) by a Download Link to the distributionspecific files. This Link ist unavailable/ broken/ dead. Even using the older Version of your Package in the Boardmanager of the Arduino IDE didn't work.
May i am an Idiot and it really didn't work or its broken.

Sincerly, thahnen

JamesGKent, 2017-03-01:
some more information such as the platform/OS you are using and the arduino IDE version you are using might be helpful

igrr, 2017-03-01:
Do you mean some of the releases in https://github.com/igrr/esptool-ck/releases are not available for download?

thahnen, 2017-03-01:
Yes you're right, yesterday some Download Links for the Releases of Version 0.4.9 were not reachable. Tried it using multiple OS and WiFi Networks. By now (2 PM MEZ) its working again.
Nevertheless thanks for your helb.

igrr, 2017-03-01:
GitHub Releases are hosted on AWS, so that's not surprising :)

closed 2017-03-01 by igrr