esp8266/Arduino #3010: WiFi Drops with SDK 2.0.0_16_08_09

9lash, 2017-03-02:
Hello All, 

I updated my Espressif SDK to 2.0.0 and I am using the Arduino stable package. I am getting WiFi connection drops when my ESP tries to send the sensor data to the data.sparkfun.com. I have attached the debug log. This is debug messages that I get in this board but there are some other ESP boards in which I get total disconnect (6). No connection at all. Please let me know if you have seen such problems. 

Possible areas of error: 
1. Hardware check before delving into software in detail. Any recommendations here? 

2. Next step on the software side, we are thinking of cleaning the whole Flash. Any recommendations on how to do that and will that help us solving the problem? 

### Basic Infos

#### Hardware
Hardware:	       Custom ESP8266  board
Core Version:      	Not sure about the core version. I followed Igrr instruction of branching the SDK 2.0. 0 from the git. 

### Description

Problem description

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512KB/64KB (Note: my flash is 4M Winbond W25x40cl but somehow if I choose 4M     in Arduino, it gets corrupt)
CPU Frequency:       80Mhz
Flash Mode:		DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck 

### Sketch

```cpp

#include <SPI.h>
#include <ESP8266WiFi.h>
#include <Wire.h>

//For VCC 
extern "C" {
#include "user_interface.h"
uint16 readvdd33(void);
}

//Time to connect to Wifi
long timer;
int connection =0; 

//Wifi Parameters: 
const char* ssid = "DGA_front";
const char* password = "password";

// Host
const char* host = "data.sparkfun.com";

// Temperature & Humidity Variables
double temperature;
double humidity;
unsigned int pollingInterval = 100;

void setup() {
 Serial.begin(115200);
 delay(10);

 // We start by connecting to a WiFi network
  Wire.begin(2, 14);
  Wire.setClock(100000);
  delay(10);

 
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.persistent(false);
  WiFi.mode(WIFI_OFF);   // this is a temporary line, to be removed after SDK update to 1.5.4
  WiFi.mode(WIFI_STA);
  WiFi.setOutputPower(0);

  WiFi.begin(ssid, password);  timer = millis();

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  timer = millis()-timer;
 
 Serial.println("WiTime: " + String(timer)); 
 Serial.println("");
 Serial.println("WiFi connected");  
 Serial.println("IP address: ");
 Serial.println(WiFi.localIP());

  if (!smeHumidity.begin()) { // Flash red light if there is an error initializing the sensor
    Serial.println("Sensor did not begin");
  }
  else {
    Serial.println("Sensor Begins!");
  }

}

 
void loop() {

/*
  //Read the Temperature from Humidity sensor
 i=10;
  while(i>0)
  {
    // The value is returned in Celsius
    temperature = smeHumidity.readTemperature();
    //Serial.println(temperature);
    delay(pollingInterval);
    i--;
  }
    Serial.print("Temperature: ");
    Serial.println(temperature);

*/

// Activate the Temperature sensor and read the temperature & humidity and then Deactivate  
  
    int i =0;
    while(i<10)
    { 
    temperature = smeHumidity.readTemperature();
    delay(pollingInterval);
    i++;
    }
    Serial.print("Temperature: ");
    Serial.println(temperature);
  //  Serial.println("I am in Hyderabad");  
    
    
    i=0;
    while(i<10)
    {
    humidity = smeHumidity.readHumidity();
    delay(pollingInterval);
    i++;
    }
    
    Serial.print("Humidity: ");
    Serial.println(humidity);

    Serial.print("MACess of ESP         : "); 
    Serial.println(WiFi.macAddress());
  
/*
  //  ADC_MODE(ADC_VCC);
    int Vcc = ESP.getVcc();
    Serial.print("VCC:");
    Serial.println(Vcc);
*/

   // Get VCC/ Battery life from ESP
   int Vcc = readvdd33();
   Serial.print("VCC:");
   Serial.println(Vcc);

   
  //**********************************
  Serial.print("Connecting to ");
  Serial.println(host);

  
  // Use WiFiClient class to create TCP connections
  WiFiClient client;
  const int httpPort = 80;
  if(WiFi.status()== WL_CONNECTED)
  {
  if(!client.connect(host, httpPort))
  {
    Serial.print("connection failed and Wifi Status now is : ");
    Serial.print(WiFi.status());
    Serial.print("  with RSSI:");
    Serial.println(WiFi.RSSI());
    Serial.print("And now, IP address: " );
    Serial.println(WiFi.localIP());
    connection =0;
    reconnect();
    delay(10000);
 // return;
  }
  else
  { Serial.print("Connected to Internet while making the TCP request and the WiFi status is now :  ");
    Serial.print(WiFi.status());
    Serial.print("  with RSSI:");
    Serial.println(WiFi.RSSI());
    Serial.print("And now, IP address: " );
    Serial.println(WiFi.localIP());
    connection =1; 
    client.print(String("GET /input/n1V8AM99KjTj0LEnQnnq?private_key=Mo95GwkkYZc2AnE5K55Y&conn=") + String(connection)+ "&humidity=" + String(humidity)+ "&temp=" + String(temperature) + "&vcc=" + String(Vcc) + "&witime="+String(timer)+ " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" + 
               "Connection: close\r\n\r\n");
    Serial.println();
    Serial.println("closing connection");
    delay(3000);
   }
  }
  else
  {
    reconnect();
    }
  //  Serial.print("Connection: ");
  //  Serial.println(connection);
  Serial.println();
    

  //Sleep for 60 seconds
              
  // ESP.deepSleep(5000000, WAKE_RF_DEFAULT);
    }

    void reconnect()
    {
        WiFi.begin(ssid, password);
        timer = millis();
        while (WiFi.status() != WL_CONNECTED) {
        delay(5000);
         Serial.print(".");
         }
         timer = millis()-timer;
 
         Serial.println("WiTime: " + String(timer)); 
         Serial.println("");
         Serial.println("WiFi reconnected");  
         Serial.println("IP address: ");
         Serial.println(WiFi.localIP());      
    }
    

```

### Debug Messages

```
...wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 4
.....wifi evt: 0
..wifi evt: 3
.WiTime: 13003

WiFi connected
IP address: 
192.168.1.190
Sensor Begins!
Temperature: 28.83
Humidity: 36.00
MACess of ESP         : 60:01:94:02:61:99
VCC:0
Connecting to data.sparkfun.com
[hostByName] request IP for: data.sparkfun.com
pm open,type:2 0
[hostByName] Host: data.sparkfun.com lookup error: -5!
connection failed and Wifi Status now is : 3  with RSSI:-69
And now, IP address: 192.168.1.190
sta config unchangedsl
scandone
usl
.WiTime: 5000

WiFi reconnected
IP address: 
192.168.1.190

Temperature: 28.90
Humidity: 36.00
MACess of ESP         : 60:01:94:02:61:99
VCC:15
Connecting to data.sparkfun.com
[hostByName] request IP for: data.sparkfun.com
[hostByName] Host: data.sparkfun.com lookup error: -5!
connection failed and Wifi Status now is : 3  with RSSI:-64
And now, IP address: 192.168.1.190
sta config unchangedWiTime: 0

WiFi reconnected
IP address: 
192.168.1.190
sl
scandone
usl

Temperature: 28.78
Humidity: 36.00
MACess of ESP         : 60:01:94:02:61:99
VCC:0
Connecting to data.sparkfun.com
[hostByName] request IP for: data.sparkfun.com
[hostByName] Host: data.sparkfun.com lookup error: -5!
connection failed and Wifi Status now is : 3  with RSSI:-68
And now, IP address: 192.168.1.190
sta config unchangedWiTime: 0

WiFi reconnected
IP address: 
192.168.1.190
sl
scandone
usl

Temperature: 28.83
Humidity: 36.00
MACess of ESP         : 60:01:94:02:61:99
VCC:0
Connecting to data.sparkfun.com
[hostByName] request IP for: data.sparkfun.com
[hostByName] Host: data.sparkfun.com lookup error: -5!
connection failed and Wifi Status now is : 3  with RSSI:-65
And now, IP address: 192.168.1.190
sta config unchangedWiTime: 0

WiFi reconnected
IP address: 
192.168.1.190
sl
scandone
usl

```

9lash, 2017-03-03:
Any ideas on resolving the issue will be really helpful. I am running out of test cases to check.

devyte, 2017-03-03:
Hi,

I'm a bit new around here, so sorry if the following is stoopid :P

There was a function added in a recent SDK release (not sure if it was
1.5.4.1), related to wifi radio calibration. As I understand it, the SDK
function must be called on bootup, and in some cases after waking up from
sleep. Failure to do that reduces wifi radio quality and wifi drops with
similar symptoms as described here.

Is this being done already? If not, that could be the cause.
The function name is user_rf_cal_something (I'd look it up for more
details, but I'm on the road atm).

9lash, 2017-03-04:
Thank you for your response @devyte . Appreciate it! 

I did not know that the SDK 2.0.0 needs RF calibration to be done. I read the documentation after you mentioned about it. It looks like the function you are referring to is:  user_rf_cal_set() but I cant find where this function is implemented in the SDK and therefore dont know how to use it in my program. But I found this function in one of the example project, IOT_demo which had its implementation in user_main.c. It looks like it sets the variable rf_cal_sec to the right value depending upon your Flash size. My flash size here is a 4Mbit (512KB ) therefore it should select rf_cal_sec = 128-5; .  Do you know what file to include in arduino to set this variable to the above value?

devyte, 2017-03-04:
Apologies, I don't know details about its use, only that it is mentioned in
the Espressif api reference, and that there was talk about it in the
Nodemcu github.

I would think that calling it should not be up to the user (it's not for
the nodemcu firmware), but rather by the core in the correct places.

devyte, 2017-09-09:
@9lash have you tried latest git? I have been using the libs for a looong while now without any wifi issues.

devyte, 2017-10-10:
No reply in over a month, closing.

closed 2017-10-10 by devyte