esp8266/Arduino #3012: An AP is created automatically, even without any code loaded in.

vgeraud, 2017-03-02:
Hi,

I'm using WeMos D1 mini with the simple following sketch : 

void setup() {
  Serial.begin(115200);
  Serial.println("test");
}

void loop() {}

I noticed that even though my sketch isn't involving any connection process, an access point is created by the D1 mini and that I can connect to it with any device. However there is nothing if I enter the board's IP in my browser...

It seems that the AP is automatically created on boot.
Any idea on how to disable this function ?

Thanks.

kaofishy, 2017-03-02:
See the WiFiClientBasic example for ESP8266 included in the Arduino board library.

101010b, 2017-04-06:
Hi, 
I encountered the same problem and don't see how the WiFiClientBasic example helps here. Any additional hints are highly appreciated.
Thanks.

themindfactory, 2017-04-06:
ESP8266 remembers its old settings, you do not need code to be an AP, the
module will do that all on its own....

If you want it not to be an AP then add WiFi.softAPdisconnect(false); and
WiFi.enableAP(false); to your code.

http://www.esp8266.com/viewtopic.php?p=56442

101010b, 2017-04-09:
Works! Thanks for the hint.

closed 2017-09-09 by devyte