esp8266/Arduino #3015: WIFI Dropouts with SDK 2.0 -Helloserver.ino

9lash, 2017-03-02:
### Basic Infos

#### Hardware
Hardware:			Custom ESP8266 based board
Core Version:      	        2.3.0

### Description

Problem description

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512KB/No SPIFFS
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		 SERIAL
Reset Method:		ck 

### Sketch

```cpp
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

 
// WiFi parameters
const char* ssid = "DGA_front";
const char* password = "password";

ESP8266WebServer server(80);

const int led = 13;

void handleRoot() {
  digitalWrite(led, 1);
  server.send(200, "text/plain", "hello from esp8266!");
  digitalWrite(led, 0);
}

void handleNotFound(){
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}

void setup(void){
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
  WiFi.disconnect();
  delay(100);
  
  WiFi.mode(WIFI_STA);
  delay(100);
  
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  Serial.print("PHY Mode: ");
  Serial.println(WiFi.getPhyMode());

  Serial.println(" PHY_MODE_11B = 1, PHY_MODE_11G = 2,PHY_MODE_11N = 3");
 
 /* 
  Serial.println("Setting it to 2");
  //Setting the Physical layer mode in 801
  WiFi.setPhyMode(WiFiPhyMode_t(1));
  Serial.print("New PHY Mode: ");
  Serial.println(WiFi.getPhyMode());
*/
  
  if (MDNS.begin("esp8266")) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);

  server.on("/inline", [](){
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
  
}

void loop(void){
  server.handleClient();
}

```

### Debug Messages

```

wifi evt: 1
STA disconnect: 2
wifi evt: 7
wifi evt: 7
wifi evt: 7
.wifi evt: 7
wifi evt: 7
wifi evt: 7
.wifi evt: 7
....wifi evt: 1
STA disconnect: 2
....wifi evt: 7
.wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
..wifi evt: 1
STA disconnect: 2
.....wifi evt: 7
.wifi evt: 1
STA disconnect: 4
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
.wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
.....wifi evt: 1
STA disconnect: 2
.....wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
.wifi evt: 1
STA disconnect: 2
.......wifi evt: 1
STA disconnect: 4
wifi evt: 7
.wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
.wifi evt: 7
....wifi evt: 1
STA disconnect: 2
.....wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
..wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
..wifi evt: 7
wifi evt: 7
wifi evt: 7
.wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
...wifi evt: 1
STA disconnect: 2
......wifi evt: 7
wifi evt: 7
wifi evt: 1
STA disconnect: 2
.......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
..wifi evt: 7
....wifi evt: 7
wifi evt: 1
STA disconnect: 2
....wifi evt: 7
..wifi evt: 1
STA disconnect: 2
.......wifi evt: 1
STA disconnect: 2
.....wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
.wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
.wifi evt: 7
....wifi evt: 7
wifi evt: 7
..wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
..wifi evt: 7
wifi evt: 7
...wifi evt: 7
.wifi evt: 1
STA disconnect: 2
....wifi evt: 7
..wifi evt: 1
STA disconnect: 2
.......wifi evt: 1
STA disconnect: 2
.....wifi evt: 7
.wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
.......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
......wifi evt: 7
wifi evt: 1
STA disconnect: 2
.......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
.....wifi evt: 7
.wifi evt: 7
wifi evt: 7
wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
.......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
...wifi evt: 7
wifi evt: 7
...wifi evt: 1
STA disconnect: 2
.....wifi evt: 7
wifi evt: 7
.wifi evt: 1
STA disconnect: 2
.......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
wifi evt: 7
wifi evt: 7
.wifi evt: 7
wifi evt: 7
.....wifi evt: 1
STA disconnect: 2
.......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
.....wifi evt: 7
wifi evt: 7
......wifi evt: 7
wifi evt: 7
wifi evt: 1
STA disconnect: 2
.......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
...wifi evt: 7
wifi evt: 7
...wifi evt: 1
STA disconnect: 2
..wifi evt: 7
wifi evt: 7
wifi evt: 7
....wifi evt: 1
STA disconnect: 2
..wifi evt: 7
...wifi evt: 7
wifi evt: 7
..wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
..wifi evt: 7
.wifi evt: 7
.wifi evt: 7
wifi evt: 7
..wifi evt: 1
STA disconnect: 2
......wifi evt: 7
.wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
......wifi evt: 7
wifi evt: 7
.wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
wifi evt: 7
wifi evt: 7
wifi evt: 7
......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
.......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
.wifi evt: 7
......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
....wifi evt: 7

...wifi evt: 7
wifi evt: 7
wifi evt: 7
...wifi evt: 1
STA disconnect: 2
.......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
.......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
....wifi evt: 7
wifi evt: 7
wifi evt: 1
STA disconnect: 2
.......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
.......wifi evt: 1
STA disconnect: 2
.wifi evt: 7
wifi evt: 7
.....wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
.......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
..wifi evt: 7
wifi evt: 7

..wifi evt: 7
wifi evt: 7
....wifi evt: 1
STA disconnect: 2
...wifi evt: 7
wifi evt: 7
...wifi evt: 1
STA disconnect: 2
.......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
......wifi evt: 1
STA disconnect: 2
......wifi evt: 7
wifi evt: 7
wifi evt: 1
STA disconnect: 2
.wifi evt: 7
......wifi evt: 1
STA disconnect: 2

...wifi evt: 7
...wifi evt: 1
STA disconnect: 2
..wifi evt: 7
.wifi evt: 7
.wifi evt: 0
..wifi evt: 3
.
Connected to DGA_front
IP address: 192.168.1.190
PHY Mode: 3
 PHY_MODE_11B = 1, PHY_MODE_11G = 2,PHY_MODE_11N = 3
MDNS responder started
HTTP server started
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
:urn 123
:urch 123, 220
:urch 220, 202
:urd 43, 202, 13
:urch 202, 202
:urd 43, 202, 13
:urch 202, 202
:urd 43, 202, 13
:urch 202, 220
:urch 220, 481
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
:urch 481, 40
:urd 11, 40, 13
:urd 4, 40, 25
:urd 5, 40, 30
:urch 40, 40
:urd 11, 40, 13
:urd 4, 40, 25
:urd 5, 40, 30
:urch 40, 40
:urd 11, 40, 13
:urd 4, 40, 25
:urd 5, 40, 30
:urch 40, 40
:urd 11, 40, 13
:urd 4, 40, 25
:urd 5, 40, 30
wifi evt: 7
:urch 40, 220
:urch 220, 202
:urd 43, 202, 13
:urch 202, 202
:urd 43, 202, 13
:urch 202, 202
:urd 43, 202, 13
:urch 202, 220
:urch 220, 323
:urch 323, 40
:urd 11, 40, 13
:urd 4, 40, 25
:urd 5, 40, 30
:urch 40, 40
:urd 11, 40, 13
:urd 4, 40, 25
:urd 5, 40, 30
:urch 40, 40
:urd 11, 40, 13
:urd 4, 40, 25
:urd 5, 40, 30
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7

```

9lash, 2017-03-02:
Please let me know if anybody experienced similar problems?

andig, 2017-03-20:
Double post with https://github.com/esp8266/Arduino/issues/3010?

devyte, 2017-09-09:
Closing as duplicate.

closed 2017-09-09 by devyte