esp8266/Arduino #3016: digitalWeite( led, HIGH ) doesn't work!

danxinzhujian, 2017-03-03:
I tried to let arduino work as a CoAP server., everything works well but "digitalWrite(led, HIGH)", which is in a .cpp file, has no reply. I also tried digitalRead(pin), the value is LOW all the time. Is there something wrong with Arduino library? I'm using General esp8266 module.

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:		ESP8266-01
Core Version:      	2.1.0

### Description

Problem description

### Settings in IDE

Module:  			      Generic ESP8266 Module
Flash Size: 		       512K
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	        40Mhz
Upload Using:		        SERIAL
Reset Method:		        ck

### Sketch

```cpp

#include <stdbool.h>
#include <string.h>
#include "coap.h"
#include "Arduino.h"

static char light = '0';

const uint16_t rsplen = 1500;
static char rsp[1500] = "";
void build_rsp(void);

static int led = 6;

static int handle_put_light(coap_rw_buffer_t *scratch, const coap_packet_t *inpkt, coap_packet_t *outpkt, uint8_t id_hi, uint8_t id_lo)
{
    if (inpkt->payload.len == 0)
        return coap_make_response(scratch, outpkt, NULL, 0, id_hi, id_lo, &inpkt->tok, COAP_RSPCODE_BAD_REQUEST, COAP_CONTENTTYPE_TEXT_PLAIN);
    if (inpkt->payload.p[0] == '1')
    {
        light = '1';
        digitalWrite(led, HIGH);
        return coap_make_response(scratch, outpkt, (const uint8_t *)&light, 1, id_hi, id_lo, &inpkt->tok, COAP_RSPCODE_CHANGED, COAP_CONTENTTYPE_TEXT_PLAIN);
    }
    else
    {
        light = '0';
     }
}

     
    ......

}

RudyFiero, 2017-03-03:
It is impossible to run your code without the included libraries.  It would help if you were to specify the COAP library that you are using.

danxinzhujian, 2017-03-04:
Thanks! The probled is solved. I have realized esp8266 module can't control Arduino, so I should use the GPIO pins of esp8266.

jjssoftware, 2017-03-06:
Close it

closed 2017-09-09 by devyte