esp8266/Arduino #3017: Cannot connect to remote server

Katniss394, 2017-03-03:
Hardware:			ESP-12F

Problem description
I'm trying to connect my ESP8266 12F microcontroller to a remote Apache server hosted on hostinger.com. But the client is unable to connect. What I need to do is write some readings to my mysql database on server. I've connected the microcontroller to my mobile hotspot but cannot connect to server. 

### Settings in IDE

Module:  			Generic ESP8266 Module?
Flash Size: 		        512k
CPU Frequency:		80Mhz
Flash Mode:			Dio
Flash Frequency:	40Mhz
Upload Using:	 SERIAL
Reset Method:	 ck

### Sketch

#include<ESP8266WiFi.h>

#define SS_PIN 4
#define RST_PIN 5
#define host http://smartlbus.esy.es/
MFRC522 mfrc522(SS_PIN, RST_PIN); // Create MFRC522 instance.

static const int RXPin = 0, TXPin = 16;
static const uint32_t GPSBaud = 9600;

// The TinyGPS++ object
TinyGPSPlus gps;

// The serial connection to the GPS device
SoftwareSerial ss(RXPin, TXPin);
void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600); // Initialize serial communications with the PC
  SPI.begin(); // Init SPI bus
  mfrc522.PCD_Init(); // Init MFRC522 card
  WiFi.mode(WIFI_STA);
  WiFi.begin("smartbus","qwerty123");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
  if(WiFi.status()==WL_CONNECTED)  Serial.println("Connected to wifi");

}

void loop() {

  // Use WiFiClient class to create TCP connections
  WiFiClient client;
  const int httpPort = 8000;
 if (client.connect("http://www.smartlbus.esy.es", httpPort)) {
  Serial.println("Connected to server");
  while (ss.available() > 0)
    if (gps.encode(ss.read()))
      {
        double lat,lng;
        float speed;
         if (gps.location.isValid())
              {
              lat=gps.location.lat();
              lng=gps.location.lng();
              speed=gps.speed.value();
              }

        String data = "lat1=" + (String)  lat + "&lng1=" + (String)lng + "&speed=" + (String)speed;

          client.println("POST /add.php HTTP/1.1"); 
          client.println("Host: www.smartlbus.esy.es");
          client.println("Content-Type: application/x-www-form-urlencoded");
          client.print("Content-Length: "); 
          client.println(data.length()); 
          client.println(); 
          client.print(data);
          delay(500);

      }
 }
  else{
    Serial.println("Cannot cannect to server..");
  }
 // Look for new cards

Please help me out.

jjssoftware, 2017-03-06:
take a look at the many examples - your client.connect stuff clearly isn't right

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiClient/WiFiClient.ino

devyte, 2017-09-09:
@Katniss394 WiFiClient is currently working on latest git. In addition, you probably shouldn't be trying to do your own POST, but rather use ESP8266HTTPClient class.
In any case, this isn't the right place to ask for help for your particular project. This is an issue tracker, meant for tracking issues in the core. If you still have a valid prblem with a core feature, please open a new issue, and add a sketch that is both complete and minimal.
Closing as user error.

closed 2017-09-09 by devyte