esp8266/Arduino #3020: Problem with platform.io and ArduinoOTA (Example BasicOTA.ino)

chessweb01, 2017-03-03:
The example BasicOTA.ino that illustrates over the air firmware updates of ESP8266 modules compiles with Arduino 1.6.8 but not with Atom/platform.io. 

With the latter I get "error: 'class ArduinoOTAClass' has no member named 'getCommand'". platform.io somehow incorporates an obsolete version of the ArduinoOTA class.

I'm positive that Atom, platform.io, and all packages are up to date on my computer.

devyte, 2017-09-07:
@chessweb01 is this still an issue? I know of others using platformio without problems.

gschintgen, 2017-09-26:
@devyte 
Hi, I'm not the OP but I'd say this is still an issue. I've just done some maintenance on a small project that I started some months ago and tried to remove my local copy of ArduinoOTA in order to reduce deviations from upstream, when I hit this "has no member named 'getCommand'" error. I'd say my Atom / PlatformIO setup is current as of today.

devyte, 2017-10-04:
Duplicate of #2833 , closing.

closed 2017-10-04 by devyte